dresden-elektronik/deconz-rest-plugin #1566: Xiaomi Sensors not refreshing

hgelpke, 2019-06-05:
I upgraded to 2.05.65 via the Hass.io addon and it's improved my Xiaomi sensors which were previously losing their connection entirely. Now it appears that they are not refreshing and the 2 that I have re-paired are showing as last refreshing 14 hours ago

thowi, 2019-06-15:
I have a similar problem that my Xiaomi temp sensor doesn't update anymore:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1514

manup, 2019-06-15:
If you're using the ConBee II please make sure that you have the latest firmware installed. For proper end-device handling at least version 0x26490700 is required. If the update isn't shown in the *Phoscon App > Settings > Gateway* you may update the firmware as described in:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

thowi, 2019-06-15:
In my case it's Conbee I, latest firmware.

tanzerm, 2019-06-17:
Same problem here with ConBee 1 and Xiaomi Multisensors, Version 2.05.65, Firmware 36330500

manup, 2019-06-17:
Can you please provide more details about your setup.

* To which router was the problematic sensor connected?
* What lights or power-plugs do you have?

Might be related to https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1595

thowi, 2019-06-17:
I believe I've seen the problems with the multi sensor directly connected to Conbee.
Have not tried with another router in-between.

I own these routers: Xiaomi socket, Osram SMART+ socket (Germany), Osram SMART+ bulb, Innr socket, Bitron socket, IKEA repeater.
I can try with any of those. But I'm pretty sure I tried it in isolation with no router at all, and directly at the controller, because I first wanted to try the range of the dongle/sensors without any router first.

manup, 2019-06-17:
Hmm strange at least with ConBee (1) and firmware 0x26330500 the connection to Xiaomi sensors should never get lost when the sensor is connected directly to the gateway.

To connect it again just start sensor search and reset the sensor by pressing the button until the LED flashes.

thowi, 2019-06-17:
In my case the connection isn't lost. I believe I see a green line in deCONZ. I just never seem to get any measurement updates from the sensor.

I'll try this tonight:
* Remove all router.
* Pair multi sensor with Conbee.
* Trigger updates on the sensor (warming it up, pushing the button). Verify that I (don't) get updates.
* See if I can execute any commands against the sensor.

Anything else I can try?

thowi, 2019-06-17:
So, I tried with only the Xiaomi temp sensor connected to the Conbee I controller:
[image]

I get a reading just after pairing and then no more updates, even when I press the button on the sensor or when I warm it up in my hand.
When I try to read the temperature cluster info it just times out and I see the left dot flashing red in deCONZ.

Here's what I see in the logs:

> 21:44:33:177 SensorNode 5: Temperature 5 added
> 21:44:33:178 don't create binding for attribute reporting of sensor Temperature 5
> [..]
> 21:44:33:183 Websocket 192.168.1.197:55747 send message: {"e":"changed","id":"5","r":"sensors","state":{"lastupdated":"2019-06-17T19:44:33","temperature":2733},"t":"event","uniqueid":"00:15:8d:00:02:32:0b:57-01-0402"} (ret = 160)
> [..]
> 21:44:34:670 Websocket 192.168.1.100:53804 send message: {"e":"changed","id":"5","name":"Multi Sensor","r":"sensors","t":"event","uniqueid":"00:15:8d:00:02:32:0b:57-01-0402"} (ret = 117)
> [..]
> 21:50:35:956 Node 0x00158D0002320B57 is known by 1 neighbors, last seen 4 s
> [..]
> 23:15:00:915 Node 0x00158D0002320B57 is known by 0 neighbors, last seen 4390 s
> [..]
> 23:14:56:147 ZCL read cluster: 0x0402, attribute: 0x0000
> 23:14:56:148 ZCL read cluster: 0x0402, attribute: 0x0001
> 23:14:56:148 ZCL read cluster: 0x0402, attribute: 0x0002
> 23:14:56:148 ZCL read cluster: 0x0402, attribute: 0x0003

I believe I'm using the latest firmware on Conbee I. At least it doesn't ask me to update.
[image]

My deCONZ version (using Ubuntu/APT installation):
[image]

I only own this one temp sensor, so I can't compare to another one.
I don't own a Xiaomi gateway and have never tried to upgrade the sensor firmware.
So it is possible that the sensor itself may be at fault.
I ordered more, but that will take some time to ship.
It says this on the back:
> WSDCGQ11LM
> CMIT ID: 2016DP6443

LMK if I can provide any more information.

Thanks!

hgelpke, 2019-06-17:
Looking at my own setup, it seems to have been distance related

I moved my conbee ii away from the back of my PC and that seems to have
solved the problem

thowi, 2019-06-18:
My controller was already on a USB cable, but only ~30 cm away from my NUC.
Now with ~1.5m distance I got some very sporadic updates, but again the last update is >2 hrs ago, even though I warmed up the sensor and pushed the button on the case.
Maybe my device is just faulty, I don't know. I don't have a Xiaomi gateway so can't try if it's better there. Also don't own a second sensor yet. Waiting for delivery.

In https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1514 there's a recommendation to try deCONZ 2.05.66. I'm waiting for the latest version to hit the APT repository.

manup, 2019-06-19:
@thowi I think the specific Xiaomi problem was fixed in 2.05.65, if you update to latest version 2.05.66 it should work again.

thowi, 2019-07-05:
I'm on 2.05.65 now (previously .64) and it seems to work much better!

thowi, 2019-07-05:
As @hgelpke and I are happy now, I believe we can close this issue.
I can open another issue should I see problems again.

Thanks so much for your help, @manup!

klada, 2019-09-10:
I am still having this issue with my ConBee II (FW 264A0700) and the latest gateway version (2.05.69 / 6.9.2019).

For testing purposes I have only paired one Xiaomi multisensor (round version, FW 20160516) with DeCONZ. It's visible in the UI and sends a few updates right after pairing. After a few hours the connection is lost and no more updates are sent, although the sensor is very close to the ConBee stick.

After rebooting the DeCONZ host the sensor is greyed out in the web UI. I have already tried resetting the network and switching ZigBee channels, without any luck.

thowi, 2019-09-10:
I also have 1 Xiaomi Multi Sensor that loses updates after a while. My other 3 sensors all work fine though. So maybe that sensor is just somehow broken?

ArkadiuszNiemiec, 2019-09-23:
I just bought ConBee II with v264A0700. I am testing it on temp, motion and door sensor. All of them goes offline after reboot but temperature sensors goes back after a temp or humidity change and both motion sensor and door sensor stays offline until I single press their button...

thowi, 2019-10-02:
Btw, @manup, I can mail you my misbehaving square Xiaomi Multi Sensor if you want to try this out in our own lab.
For me, it's just one out of 7 sensors. The other 6 work fine. So either it's a defective device, or it has some other special behavior that Conbee/deCONZ doesn't handle well.

Cheers

derheinrich2019, 2019-10-09:
Hi!
Same issue here.

[image]

My setup:
raspberry pi 1, Raspbian Buster Headless sd-card image, conbee2. It works for weeks without any problems. I'm using ca. 40 xiaomi sensors and two hue- and innr lights. 
Suddenly I've got no updates from all xiaomi sensors. Hue and innr lamps are still working.

Now I reset the configuration, used another deconz-image, change the raspberry, usb cable, distance ... 
Still the same. I can pair the sensors and after 20 minutes - no updates.

wuast94, 2019-10-23:
today i added 10 of the sonsors. after a few hours all looks fine exepts one sensor that stops updating after a few minutes. what looks strange to me is that this is the only sensor that have a yellow connection. the nearest repeater is a hue light bulb "Schlafzimmer" in the same room but it doesnt connect to it and stays on the conbee stick
[image]

Smanar, 2019-10-23:
You can try to cheat, re-pair it but realy close to the hue light, and from my tests, let it at this place somes hours. During firsts hours after pairing Xiaomi sensors have stranges reactions, but after never change their parent.

apmon, 2019-10-25:
I have a bunch of the square xiaomi temperature on my network (Conbee I) and generally they work decently well. However, at seemingly random after a couple of weeks or month they drop off of the network and don't work anymore. (quite possible they are at the edge of the range and fail for a bunch of updates). In the past I have usually been able to simply re-pair them again. More recently though I seem to be having issues re-pairing them and they don't connect anymore at all, even when right next to the conbee. I use the phoscon app to add a new sensor and then hold down the sensor button until it starts blinking, but it never connects to network anymore. Connecting a new sensor still works fine. So it looks like either the sensor doesn't reset correctly and still has old broken state information in it, or deConz still has old wrong information about the sensor and doesn't let the sensor back onto the network. I have tried deleting it through phoscon, but that still didn't help. However in the past deleting a sensor clearly didn't delete all information out of the internal databases of deconz, so I am not sure if the relevant information is still in deconz.

ebaauw, 2019-10-26:
Did you double-check the sensor’s battery?

klada, 2019-11-05:
@ebaauw I have replaced the battery and it shows as 100% in DECONZ - still the connection drops after a few minutes.

acoloss, 2019-12-04:
same issue here.. just bought a conbee II, latest firmware, temp sensor + door sensor.
Temp sensor seems to just "stop", have not had an update in 24 hours. shows same values.

Hydrocephalic, 2019-12-07:
I have the same issue with Conbee II (v2.50.71), temp sensor and door sensor.

thowi, 2019-12-13:
Recently my sensors are dying like flies.
Over the last days  3 out of 10 Xiaomi temp sensors and 2 out of 10 door sensors became unavailable.
Batteries reported 60 - 90% before they become unavailable.
Also, 2 out of 5 sockets became unavailable. (Bitron 902010/28 and Xiaomi Socket unavailable, another Xiaomi Socket, another Bitron, and a Paulmann relais are working).

I didn't change anything in my setup. No wiring or positioning change.
I'm particularly surprised about the sockets. They are wired (so battery cannot be the issue) and one of them is just ~4 m away from Conbee with no obstacles in between.
One of the sensors is also just 1.5 m away from Conbee.
I have a feeling that door sensors became less reliable, too, recently.
This feels like a somewhat steady decline, which suggests battery issues. Battery levels are 90+% for most sensors. The lowest being 51% (which actually works fine). Maybe the sensors transmit with less power when batteries run lower? Would be strange though if I had to replace batteries that are reported 90+%...

I have a mesh of 1 Conbee, 4 sockets, all on one floor and this used to work perfectly until recently.
I'm wondering why this is declining so much now.

Is there anyway to provide (long term?) debug information to find out why these devices go offline?

Conbee I.
Version 2.05.67 / 12/12/2019.
Firmware 26330500.

ebaauw, 2019-12-13:
> Batteries reported 60 - 90% before they become unavailable.
> Would be strange though if I had to replace batteries that are reported 90+%...

No, for Xiaomi CR2032-powered devices that means the battery has been depleted.  See explanation is many other issues.

thowi, 2019-12-13:
> No, for Xiaomi CR2032-powered devices that means the battery has been depleted. See explanation in many other issues.

Found this issue:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2161#issuecomment-560168054
You suggest that:
> Double-check the sensor battery. If it's under 100%, replace it and check whether the problem still occurs. If so, double check with `dc_eventlog` if deCONZ sends multiple websocket notifications or if Home Assistant is imagining them.

I didn't know that they already stop working properly below 100%.
Strangely, I've got a multi sensor at 51% that's working perfectly. So it's a little confusing.
But okay, I can try changing those batteries! Maybe I'll actually measure their voltage using my multimeter to see at what point they stop working.

ebaauw, 2019-12-14:
> Maybe I'll actually measure their voltage using my multimeter to see at what point they stop working.

That will depend on many factors:
- The sensor model and probably even the individual sensor (the `lumi.weather` in my bathroom seems more sensitive than my other `lumi.weather` sensors);
- The battery (I use IKEA PLATTBOJ: EUR 1.29 per 8);
- The ambient temperature (the `lumi.weather` sensor in my computer closet (> 30°C) is at 48% and still functioning properly).

I replace the battery when the sensor starts acting strange and deCONZ reports < 100%.

It would be cool to cross-check the measured voltage against the voltage reported in the Xiaomi special attribute report.  I wouldn't be surprised to find differences between individual sensors of the same model here as well.  You should be able to see the reported voltage in the deCONZ log, when running with `--dbg-info=1` or `--dbg-aps=2` (not sure which one, I use both).  Here's one of my `lumi.weather` sensors reporting 29.95V (which deCONZ translates to 98%):
```
Dec 14 00:57:24 pi2 deCONZ[23007]: 00:57:23:723 ZCL attribute report 0x00158D00022CBAD2 for cluster 0x0000, ep 0x01
Dec 14 00:57:24 pi2 deCONZ[23007]: 00:57:23:723 0x00158D00022CBAD2 extract Xiaomi special attribute 0xFF01
Dec 14 00:57:24 pi2 deCONZ[23007]: 00:57:23:723     01 battery 2995 (0x0BB3) 
Dec 14 00:57:24 pi2 deCONZ[23007]: 00:57:23:723     04 unknown 5032 (0x13A8)
Dec 14 00:57:24 pi2 deCONZ[23007]: 00:57:23:723     05 RSSI dB (?) 30 (0x001E)
Dec 14 00:57:24 pi2 deCONZ[23007]: 00:57:23:724     06 LQI (?) 4294967296 (0x0100000000)
Dec 14 00:57:24 pi2 deCONZ[23007]: 00:57:23:724     64 temperature 2089
Dec 14 00:57:24 pi2 deCONZ[23007]: 00:57:23:724     65 humidity 4290
Dec 14 00:57:24 pi2 deCONZ[23007]: 00:57:23:724     66 pressure 978
Dec 14 00:57:24 pi2 deCONZ[23007]: 00:57:23:724     0a unknown 49735 (0xC247)
```

thowi, 2019-12-15:
Thanks for the tips, Erik!
Wondering if all these tips could be compiled in a howto/FAQ, maybe in the wiki? You're sharing very useful information that most people probably don't know (me included) and than I ask stupid questions on the issues here :)

I went through all sensors that went unavailable recently, measured the battery voltage, re-paired them, then checked battery percentage in Phoscon. Here's what I got:
- Door sensor (Bedroom window): 2.96V. Phoscon: 100%.
- Door sensor (Bedroom window atrium): 3.00V. Phoscon: 100%.
- Door sensor (Bedroom door): 3.01V. Phoscon: 91%.
- Door sensor (Guest bedroom door): 2.91V. Phoscon: 98%.
- Door sensor (Guest bedroom window): 3.01V. Phoscon: 98%.
- Door sensor (Living room window): 2.95V. Phoscon: 95%.
- Multi sensor (Bedroom climate): 2.97 V. Phoscon: 88%.
- Multi sensor (Guest bedroom climate): 2.99V. Phoscon: 95%.
- Multi sensor (Outside west climate): 2.94V. Phoscon: 71%.

Couple of strange things that I observed:
1. Phoscon battery is inconsistent with battery voltage. E.g. above you see 100% at 2.96V, and also 91% at 3.01V. I didn't cross-check with debug logs (didn't record the logs, unfortunately). Likely, the sensor measures/reports the voltage inconsistently.
1. I could successfully re-pair all devices that went offline, without changing the battery. They still report data and seem to mostly work fine. So why did they go offline, if they still work after re-pairing? Multi sensors do drop some readings though at these battery levels. I used to see more updates from those sensors, so probably battery does play a role here. Still strange that they disappear, then work again after re-pairing.
1. Possibly unrelated to the sensors going offline: I've also had two powered sockets going offline recently (Bitron and Xiaomi). Battery cannot be the issue here. Re-pairing worked. But why do I even have to do that? Sounds like a maintenance burden if I have to re-pair my devices every couple of weeks.

Thanks again for your reference point on battery (you replace once <100% in Phoscon).
My batteries are mostly around 3.00V or slightly below, and Phoscon reports <100% for most sensors. So possibly that's an issue.
I measured brand new CR2032 and CR1632 batteries and they have voltages >=3.30V, so possibly <=3.00V is already critical. Strange though, that devices stop working at 3.00V, whereas the batteries are actually specified for 3V.
Anyway, should my (now working again) sensors go offline again, I'll swap the batteries and see if that improves things.

Sorry for my somewhat frustrated tone. Some of this is just not intuitive as a user and confusing.
Thanks for all your hard work and tips, though!

chimpy, 2019-12-30:
I'm having sensors drop out too. Then I repair and they work for a while (days) then they drop off again. It's super frustrating. Especially because I can't tell whether they need a new battery or not.

acoloss, 2019-12-30:
I'd just like to point out that my dropping out issue was in fact range.
Could have sworn it wasn't, but it was. My temp sensor being just a little bit around a corner it was intermittent.
Now it works permanent.

Fruchuxs, 2020-03-26:
Same Problem here with RaspBee II
Version 2.05.74
Firmeware 26520700

Frustrating. I buyed the RaspBee for exactly this purpose.

[image]

sstamoulos, 2020-04-16:
> Same Problem here with RaspBee II
> Version 2.05.74
> Firmeware 26520700
> 
> Frustrating. I buyed the RaspBee for exactly this purpose.
> 
> [image]
> 
> EDIT: My bad. The sensor is refreshing if it detects mentionable changes.

what are  mentionable changes? how much?

thowi, 2020-04-16:
Temp sensors usually update if the temperature changes at least 0.5°C,  humidity 6%, or otherwise roughly every hour. Or if you push the button.
See also https://community.home-assistant.io/t/xiaomi-temperature-humidity-sensor-poll-time/26269/14.

BTW, I just dumped that one unreliable sensor. The other ones work pretty well.
I saw a big improvement when plugging the Conbee into a USB extension cable and keeping it at least 50 cm away from other electronics. Previously it was plugged into a USB hub with a Bluetooth dongle next to it. No bueno.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix

klada, 2020-08-14:
I made an interesting observation: when the Xiaomi sensor is meshed to some ZigBee devices like Osram wall plugs it loses it's connection from time to time. Now I forced it to mesh trough a Busch-Jaeger light switch and it's working perfectly ever since.

So it might not just be the sensor, it might also be it's connection partner which may cause issues. So you might want to try pairing the sensor next to a reliable ZigBee device (mains powered) which is close to the sensor's designated location. As far as I know the sensor will always connect to the same ZigBee device which was used for pairing, which might be another reason why people have issues with it.

pergolafabio, 2020-08-14:
Yeah, ditch osram, I also replaced with Ikea, much better