dresden-elektronik/deconz-rest-plugin #1567: Start/stop dimming up/down for Home Assistant automations with dimmer switches?

imbaimbaimba, 2019-06-05:
Hi everyone,

So, all my ZigBee is paired to deCONZ on my NAS with a ConBee stick. It works fine, the automations/bindings etc. in deCONZ work as intended really. Also the Home Assistant integration seems to work fine.

Now my problem is how to get my Z-Wave wall switches(Logic Group ZHC5010) to control the lights properly. Turning on/off is easy, but dimming the lights has certainly been having me pulling off my hair.

In Hubitat there is a start_Level_Change button under every ZigBee bulb, along with a stop_Level_Change button. This can be used to execute on button hold and button release automations, to achieve the functionality pretty easily.

Is something like this possible with deCONZ? I already tried the example in the HA cook book, but it does not work. It does indeed dim/brighten my light as intended, but because of all the commands it is flooding the ZigBee network with, it keeps dimming like 5 seconds after I release the button, and it works very slowly. https://www.home-assistant.io/cookbook/dim_and_brighten_lights/

Hope someone can help me.

chimpy, 2019-06-11:
Try making the pause script delay longer? 1ms is not very long. Try 500ms. Otherwise if you end up doing this is Node Red, I have a great NR dimming routine that works.

imbaimbaimba, 2019-06-12:
Can you svare that NR routine?

chimpy, 2019-06-17:
@imbaimbaimba Sure! https://pastebin.com/ArvzSwZJ

closed 2019-10-22 by stale[bot]