dresden-elektronik/deconz-rest-plugin #1569: [Question] green dot blinking permanently

gpl1998, 2019-06-05:
Hi,

Quick question: just added a third Heiman Smoke Detector to my configuration. In contrast to the other two Heiman Smoke Detectors, there is a green dot blinking every two seconds (see screenshot). I wondering why as all other devices in my configuration are blinking blue (and much less frequent).

(I'm using v2.05.65) 

[image]

Smanar, 2019-06-06:
Some infos here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1496
It's working perfectly ?

gpl1998, 2019-06-08:
now, 3 days later, the device has lost connection (again) :-( 

I've 94 nodes in my configuration, this device is the only one that makes constantly troubles.
I powered off/on as well as re-added this device multiple times, but sooner or later the connection gets lost again. As said before, I have two more Heiman smoke detectors, which work properly without any issues.

Any hints/tips are highly appreciated to get this device working more reliable.

[image]

Smanar, 2019-06-08:
The green blinking can say the "pairing procedure" is not finished.
So on my side, that I would do is trying pairing the device using another router or with direct connexion to coordinator

closed 2019-06-16 by gpl1998