dresden-elektronik/deconz-rest-plugin #1570: Backup not working in v2.05.65

gpl1998, 2019-06-05:
Hi,

after updating to v2.05.65, I'm not able to create a backup file in Phoscon.
The screen "Backup will created" appears, but then nothing happens - I waited 10 minutes, then I pushed the browser's reload-button.

[image]

bubffm, 2019-06-07:
I had this the other day, but for some reason it worked today.

butschi, 2019-06-07:
I have the same. In the console it reports:

```
[Error] Unhandled Promise Rejection: ReferenceError: Can't find variable: apiversion
	(anonymous function) (settings-gateway-functions.js:3:9322)
	initializePromise
	Promise
	createBackup (settings-gateway-functions.js:3:9058)
	(anonymous function) (general-states-backup-states.js:3:718)
	(anonymous function) (xstate.interpreter.js:1:34008)
	forEach
	execute (xstate.interpreter.js:1:33989)
	update (xstate.interpreter.js:1:34127)
	send (xstate.interpreter.js:1:33250)
	serviceSend (settings-gateway-machine.js:3:8705)
	click #saveConfiguration (general-states-backup-view.js:3:1113)
	(anonymous function)
	dispatch (jquery-min.js:3:10322)
```

But since this global variable is used in multiple locations, I wonder if this would affect other things as well …

butschi, 2019-06-07:
Just found out that if you change the url from `…settings-gateway2.html…` to `…settings-gateway.html…` without the "2", backing up works as before.

This tip was posted in #1572

manup, 2019-06-10:
Good catch, will be fixed in 2.05.66

>[Error] Unhandled Promise Rejection: ReferenceError: Can't find variable: apiversion
	(anonymous function) (settings-gateway-functions.js:3:9322)
	initializePromise
	Promise
	createBackup (settings-gateway-functions.js:3:9058)
	(anonymous function) (general-states-backup-states.js:3:718)
	(anonymous function) (xstate.interpreter.js:1:34008)
	forEach
	execute (xstate.interpreter.js:1:33989)
	update (xstate.interpreter.js:1:34127)
	send (xstate.interpreter.js:1:33250)
	serviceSend (settings-gateway-machine.js:3:8705)
	click #saveConfiguration (general-states-backup-view.js:3:1113)
	(anonymous function)
	dispatch (jquery-min.js:3:10322)

manup, 2019-06-22:
Close via https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_66

closed 2019-06-22 by manup