dresden-elektronik/deconz-rest-plugin #1571: Gira 3 gang switch looses connection after some time

rendgeor, 2019-06-06:
Using the latest release Radio:
- The Gira switches can be setup and directly switch lights
- Within the light group also additional actions can be assigned to the other 4 keys

Unfortunately after ~12h the switch does not work anymore:
[image]
Whatś the reason for that?

rendgeor, 2019-08-23:
Just by accident saw:
https://forum.fhem.de/index.php?topic=100772.0

which says: "Nach RS mit der sehr hilfsbereiten und kompetenten Hotline von Dresden Elektronik stellte sich heraus, dass der GIRA Light Link offenbar ein Problem hat, sobald sich Zigbee 3.0 Komponenten im Netzwerk befinden. Ist dies der Fall, fällt der Schalter nach Inaktivität aus dem Netz und verbindet sich anschließend auch nicht mehr..."

Is there any update?
Why couldn't  Dresden E. just write the problem reason also here?

rendgeor, 2019-08-23:
Busch Jäger seems to support firmware updates.
As they seem to use the same HW probably it works too.

https://community.openhab.org/t/busch-jaeger-zigbee-light-link-control-element-stops-working/27676/12

rendgeor, 2019-08-30:
UPDATE: Gira zigbee Link link product will be discontinued.
End of 2019 the product will go out of market.

Reason: Gira says that zigbee 3.0 standard has many problems. Therefore Gira decided that further SW updates for the wall switch would be too much work.

Bad for all people owning theses switched....

closed 2019-08-30 by rendgeor

manup, 2019-08-31:
:( I still hope an OTA update will arrive in the future. We will document the case on the phoscon.de compatibility list and FAQ to make the evolved incompatibility clear.

kaeferfreund, 2020-01-05:
Hopefully Jung will issue an update, that will somehow work with the gira stuff as well.

Still really bad mark for gira there, others seem to get their switches working so it should be possible...

felixstorm, 2020-03-13:
Update for Jung devices available (see https://github.com/Koenkk/zigbee2mqtt/issues/2417#issuecomment-598650782), but apparently Gira does really not intend to release an update for their devices.