dresden-elektronik/deconz-rest-plugin #1572: Gateway tab not working since update to 2.05.65

ademobile, 2019-06-07:
Since updating to v2.05.65 the gateway tab no longer displays correctly or functions. Please note the German at the bottom under the icons.

[image]

manup, 2019-06-07:
Can you please open the developers console (F12) and reload the page. It seems that loading stops because some error is thrown.

As workaround you can also open the old version of this page by changing the url to settings-gateway.html (without the 2).

ademobile, 2019-06-07:
Taking the 2 out worked - thank you. I guess this will be addressed and maybe is this issue with those that cannot do a backup?

Kugelfang666, 2019-07-01:
I can also confirm, removing the "2" works!

manup, 2019-07-01:
The issue will be fixed in version 2.05.67.

manup, 2019-07-01:
Closed to continue further discussion in: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1630

closed 2019-07-01 by manup