dresden-elektronik/deconz-rest-plugin #1573: 'Life control' smart plug doesn't show power and consumption via deconz

Spirituss, 2019-06-07:
The zigbee smart plug of brand 'Life control' with TexasInstruments RICI01 chipset works as light via Phoscon app but doesn't show it's power and consumption, while it can be shown in original 'Life control' app. It would be grateful to add the full support of this power plug.

Link to original site of this power plug:
[Life control smart plug](https://lifecontrol.ru/devices/smart-plug/)

Main clusters info from deconz-gui for this plug:
 
[image]

[image]

[image]

bastshoes, 2019-07-06:
Please add support for metering cluster. Device details could be found in #1541

Spirituss, 2019-09-03:
Any news regarding the issue?

bastshoes, 2019-09-08:
Already supported. Check latest release.

Spirituss, 2019-09-19:
I have updated Deconz and re-added the plug into Deconz. Energy power and consumptions have added but the plug itself become a disaster. Now each 5-10 mins it switches off the relay for just a second and then switches it on again. At the same time there is no any event in HA log regarding it, HA believes that nothing happens. Thus now I can't use the plug because it so regularly restarts a plugged device. 
I have to emphasize that there was no such plug's behavior before the last Deconz update.

Spirituss, 2019-10-07:
Still waiting for the fix. 0.99.3 The device now doesn't work at all with the last deconz. Regular on/off auto switching + sometime loosing the device from deconz.

bastshoes, 2019-10-08:
Hello @Spirituss I have three plugs. All work fine in terms of loosing connection. 
Regrading switching. I've noticed it on one plug before sensors was addded. One plug switches for a second around 1 am. Maybe other too but they usually staying in off stated.

closed 2020-02-12 by stale[bot]

macros0, 2020-02-20:
> @bastshoes I have updated Deconz and re-added the plug into Deconz. Energy power and consumptions have added but the plug itself become a disaster. Now each 5-10 mins it switches off the relay for just a second and then switches it on again. At the same time there is no any event in HA log regarding it, HA believes that nothing happens. Thus now I can't use the plug because it so regularly restarts a plugged device.

What to do, bought a plug in too have all these problems?