dresden-elektronik/deconz-rest-plugin #1575: UPnP Serialnumber != bridgeid

Kane610, 2019-06-07:
Hi @manup !

Working on an issue with HASS component, and the culprit seems to be inconsistencies between the serial number from UPnP and the one gotten from the gateway over the Rest API.

UPnP: 00212E012345
Rest: 00212EFFFF012345

Is there some magic I can do to expand the FFFF on UPnP, or can I remove them from Rest?

The uuid seems to be the same between UPnP and Rest though, how static is that?

/R

robertlandes, 2019-06-08:
I can confirm this. Have the same issue. Not sure if something on hass.io side or on deCONZ side is causing this. However I am experiencing this only after I upgraded to 2.05.65. Since then discovery of hass.io is showing me a second deCONZ Zigbee Gateway was found.

[image]

The upper one without the FFFF was initially found and setup in hass.io. The second one with FFFF is "new". `bridge_id` via REST API is the FFFF version.

Kane610, 2019-06-08:
I replaced the bridgeid in favor of uuid for network discoverymatching. Would be glad to revert it with a proper way of transforming the bridgeid

ebaauw, 2019-06-08:
What version of deCONZ are you running?

I double-checked for deCONZ v2.05.64 and v2.05.65:
- The value of the `hue-bridgeid` tag in the UPnP response message to a search;
- The value of the `hue-bridgeid` tag in the UPnP Alive message;
- The value of the `GWID.phoscon.de` tag in the above UPnP messages;
- The value of the `bridgeid` attribute in the `/config` resource.

In all cases, the value is the (64-bit) ZigBee mac address of the RaspBee or ConBee.

For the Hue bridge, the value is based on the (56-bit) Ethernet mac address of the Hue bridge's NIC.  They add `FFFE` in after the manufacturer code (001788 or ECB5FA) to get the 64-bit value.

Kane610, 2019-06-08:
2.05.65, always the latest @ebaauw.

And as written above, my experience is that it differs between /config and UPnP how the bridgeid is represented.

You mean they add FFF? Or should it be FFFF? Always after the E in the bridgeid?

ebaauw, 2019-06-08:
Ah, of course, I forgot to check `description.xml`.  There, for `serialNumber`, the Hue bridge reports the 56-bit Ethernet mac address.   Looks like they filtered out the middle two bytes of the ZigBee mac address in the deCONZ REST API to report a value that the Hue app accepts in 0dc5eb727db2dc66a82c8e0b7d738d144a8fd15f: https://github.com/dresden-elektronik/deconz-rest-plugin/blob/f017057879556ead624910c7705dfa09185b1aee/upnp.cpp#L69

@KodeCR , @manup, This is the wrong design: this change should be applied only when responding to a request from the Hue app (or similar), i.e. when `ApiModeHue` is set.

robertlandes, 2019-06-08:
@ebaauw `bridgeid `in /config is "00212EFFFF0427B5", however /description.xml lists "00212E0427B5" as serial number. I guess, that's the problem, right?
...
`<serialNumber>00212E0427B5</serialNumber>`
...

ebaauw, 2019-06-08:
> You mean they add FFF? Or should it be FFFF? Always after the E in the bridgeid?

They (Philips) add FFFE.  My BSB002 bridge has serial number `23BFC2`, Ethernet mac address `00:17:88:23:bf:c2` and `bridgeid` `001788FFFE23BFC2`.  The Zigbee mac address of the Hue bridge is unrelated (except for the manufacturer part): `00:17:88:01:01:05:8C:26`.

KodeCR, 2019-06-10:
> @KodeCR , @manup, This is the wrong design: this change should be applied only when responding to a request from the Hue app (or similar), i.e. when `ApiModeHue` is set.

gwBridgeId is set to the Zigbee MAC address unless you use --lan-bridgeid=1 to start deconz in which case its the LAN MAC address. (The Hue app only works with a LAN based serial, so moving to that completely would definitely have my preference but I appreciate that could cause issues.) I can't change description.xml based on ApiModeHue because the ApiMode is set based on the username/apikey, not sure if we could detect the hue app based on the http header? 

I don't know what the "right" design is here, Philips use the MAC-48/EUI-48 version of the LAN MAC address as serial and the EUI-64 version as bridgeid. Unfortunately that seems to be the only thing their app accepts. I initially changed deconz to do the same, but we moved to using the Zigbee MAC address in not to break other apps that already used that, with an option to use the LAN MAC address based on the command line parameter.

ebaauw, 2019-06-10:
Sorry, what I meant with the wrong design is: initialising `description.xml` at startup, instead of rendering it just-in-time with each GET request, taking into account the `ApiHueMode` which has been set based on the HTTP header.

I don't think we can do anything sensible with the whole 48 vs 64 bit mac address (Did I miscount the bits before?  Updated my previous comments.)

For the Ethernet vs Zigbee mac address: if the Hue app accepts a `bridgeid` and `serialNumber` not based on the network mac address, I would keep it like that.  If not, and you want to change it, please do so only when `HueApiMode` is set.  Note that this would cause a different `bridgeid` values, when switching between wired and wireless networks on your Raspberry Pi.

KodeCR, 2019-06-12:
>if the Hue app accepts a bridgeid and serialNumber not based on the network mac address
It doesn't, that's unfortunately exactly the problem, the Hue app only works if the bridgeid is equal to the EUI-64 LAN MAC address. And it indeed changes between wired and wireless (the hue-bridge of course has it's wifi disabled so does not have this problem).

We also can't give different bridgeids based on ApiHueMode for two reasons
- description.xml is part of upnp and also the upnp broadcast contains the bridgeid
- the hue app doesn't use a user-agent in the http header, so we can't know what is requesting without a  username/apikey

Back to this particular issue, why does the serialnumber have to match the bridgeid? As in it doesn't match for the Philips Hue bridge either.
Also description.xml and upnp are both upnp. If you want to check the rest api vs upnp, use the bridgeid from /api/nouser/config, or did I misunderstand something?

Kane610, 2019-06-12:
@KodeCR because serial number and bridgeid are the same thing. Just for some reason one of them have a filter. So you should be able to identify the bridge by comparing them. Because with Home Assistant you have multiple ways of adding a deconz system, if they identify different between the different paths of which you add the bridge you would get to the situation where duplicate pairings might happen.

KodeCR, 2019-06-13:
> @KodeCR because serial number and bridgeid are the same thing

@Kane610 I understand why you might think that, but they are not. They're just very similar because Philips based them on the same thing. Of course deCONZ could do things differently, but I don't think it's a good idea to base code on that assumption, especially when the actual bridgeid is easily available. both via upnp and rest. I don't know much about Home Assistant but the bridgeid is the identifier, ignore the serial.

Kane610, 2019-06-13:
@KodeCR The Bridgeid is only exposed over UPnP as the serial number, that's the whole issue here

KodeCR, 2019-06-13:
@Kane610 Ah, ok, the bridgeid is exposed in the UPnP SSDP announcement as "hue-bridgeid" and also as "GWID.phoscon.de".

closed 2019-10-18 by stale[bot]