dresden-elektronik/deconz-rest-plugin #1577: IKEA Trådfri dimmer. The one you rotate, without buttons.

xydix, 2019-06-08:
I just paired it and it seems to work. When i rotate it Right it give me 2002, when i rotate it Left, it give me 3002.
So far so good.
But if i rotate if fast Left, i get 4002.
When i rotate if fast Right, i still get 2002.
Shouldn't i get a different value when i rotate it fast Right?
According to https://community.home-assistant.io/t/use-ikea-tradfri-dimmer-in-home-assistant/33211/78 it should support fast turn right.

olemr, 2019-06-09:
I noticed the same thing when I tested it out some 6 mnds ago, but a more major problem is that mine just disappear from the net after a few days. Tried re-pairing 4-5 times, bit eventually gave up on it as it felt a bit 'el cheapo'.

xydix, 2019-06-16:
Today I was in an IKEA store and tried this dimmer paired to a bulb. How it acted i would say there should be a a specific message when you turn it fast right. It was possible to go from off/low dim, to 100% light when I turned it fast right.

MickL, 2019-06-28:
So this is the default behavior and the issue can be closed?

xydix, 2019-06-28:
No. I still miss a value. Fast turn right.
Deconz doesn't register this.
I might be wrong but I don't think so.

closed 2019-11-02 by stale[bot]

xydix, 2019-11-03:
This issue isn't solved yet...