dresden-elektronik/deconz-rest-plugin #1579: Device Request

marschallphilipp, 2019-06-09:
I would be very pleased to see the Opus 55 Zigbee switch within the list of supported devices.
[https://www.jaeger-direkt.com/onlineshop/haustechnik/installationsbussysteme/bussystem-hand-wandsender/12383/opus-55-zigbee-wandsender-modul](url)
The switch uses the "PTM 215ZE" Enocean module.
[https://www.enocean.com/de/enocean-module-24ghz/details/ptm-215ze/](url)
I thought it´s the same as all the friends of hue switches are using. Turns out that they might use slightly different modules.
I was able to add the switch within phoscon and I can see some deconz_events within Home Assistant (when using the official hass.io addon). But I´m not able to get events for long presses / holds (I guess because the switch is recognizied as a hue tap, not the newly friends of hue switches).
Unfortunately I can´t see the switch within the Deconz UI in order to make screenshots from the device settings (when using the add-on from marthoc.
[https://github.com/marthoc/hassio-addons](url)
I will do my best and will continue to try making screenshots within deconz UI.

marschallphilipp, 2019-08-05:
Hi there, I would like to kindly ask if we can go further with implementing another kind of "FOHSwitch".
As I saw at line 1072 in https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/de_web_plugin.cpp there is some documentation about the Illumra switches allready. They share the same Enocean module like my Opus 55 switch. This is why I would love to see it implemented in order to be able to use the long press /hold button events as well (right now the opus 55 is recognized as a hue tap and the short presses are working).

What do you need to go ahaed with implementing the Illumra switches?
https://illumra.com/wp-content/uploads/2017/02/ZBT-SxAWH_Integration_Manual.pdf
Here you can find the button status encoding and a pretty solid documentation about the switch.
Please contact me if I can assist in any way (unfortunately I´m not able to see my Opus switch in deconz mesh application even  the switch is imported as a hue tap).

Kane610, 2019-08-06:
Start with this: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support

marschallphilipp, 2019-08-06:
@Kane610: This is what I did already ;-)
But as described I´m not able to see the switch in deconz application (only in phoscon) hence I´m not able to do the necessary steps described in the "Request-Device-Support" page.

ebaauw, 2019-08-06:
Zigbee Green Power (ZGP) devices don’t show in the GUI.

Afaik, there’s two versions of the module - they do differ in model number: an older one used by the Hue tab, and a newer one, used by (most?) newer FoH switches.  The newer model sends different events on button press and release; the older model sends only one event per press/release.  Neither model sends hold nor long release (the power harvesting only works when there’s movement).  The REST API plugin generates the hold and long release button events by timing the delay between the press and hold.

Probably the easiest way to get the info we need, is to pair the switch to a gen2 Hue bridge and check what capabilities the bridge reports.  See #974.

agners, 2019-08-13:
I have a similar situation there Feller (the most prevalent manufacturer of classic light switch and plugs in Switzerland) released a device with the same module PTM 215Z, also says DOLPHIN.

https://www.feller.ch/de/Produktangebot/Funktaster

The device can be paired by pressing a button 10 seconds down (I think A 1, left switch down). However, I actually have a hard time to pair the switch with Phoscon. I only managed twice so far. It appears as "Hue Tap 19". Assigning it to a lamp seems not to work and I do not get any events in HomeAssistant.

This is the debugging log form my second successful pairing:

```
23:10:34:551 APS-DATA.request id: 94, addrmode: 0x02, addr: 0xfffc, profile: 0xA1E0, cluster: 0x0021, ep: 0xF2 -> 0xF2 queue: 0 len: 17 tx.options 0x00
23:10:34:551 send GP pairing
23:10:34:551 SensorNode 19: Hue Tap 19 reactivated
23:10:34:670 APS-DATA.confirm id: 94, status: 0x00 SUCCESS
23:10:34:670 aps request id: 94 finished, erase from queue
23:10:34:692 APS-DATA.request id: 97, addrmode: 0x01, addr: 0xfff0, profile: 0x0104, cluster: 0x0003, ep: 0x01 -> 0xFF queue: 0 len: 5 tx.options 0x00
23:10:34:707 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x01 dstAddrMode: 1, profile: 0x0104, cluster: 0x0003, lqi: 230, rssi: 0
23:10:34:707 FP indication 0x0104 / 0x0003 (0x00212EFFFF04239F / 0x0000)
23:10:34:707                       ...     (0x0000000000000000 / 0x0000)
23:10:34:773 APS-DATA.confirm id: 97, status: 0x00 SUCCESS
23:10:34:773 APS-DATA.confirm request id: 97 -> erase from queue
23:10:34:793 aps request id: 97 finished, erase from queue
23:10:35:033 Mgmt_Lqi_req zdpSeq: 138 to 0xD0CF5EFFFEC18742 start index 9
```

The modules manual can be found on FCC: https://www.fcc.gov/oet/ea/fccid (Grantee Code: "SZV", Product Code: "-PTM215Z").

The operational description also says: "The length of transmission is three pulses within 100ms, each being approximately 1ms long. Through manual operation radio transmissions of device
are fully random."

agners, 2019-08-13:
The manual available from FCC also seems to have more information about how the switch exactly works. Depending on which channel the switch should be commissioned with a different button needs to be long pressed (in my case channel 15 was to left or A 0).

The manual also has information of what button codes the devices sends.

marschallphilipp, 2019-08-14:
@agners : Your Feller switch should be an official "Friends of Hue" swtich and thus be supported in the latest release of deconz.
My Opus switch comes with a PTM 215ZE" Enocean module, not the PTM 215Z" module.

Weired though that your switch isn´t recognized as an friends of hue switch. I guess the model ID isn´t yet present in the code and this is why it isn´t recongnized correctly.

ebaauw, 2019-08-14:
> Depending on which channel the switch should be commissioned with a different button needs to be long pressed (in my case channel 15 was to left or A 0).

Same for the Hue tap, and, I guess, for the other ZGP switches.  When numbering the buttons counter clockwise, 1 is channel 11, 2 channel 15, 3 channel 20, and 4 channel 25.

> I guess the model ID isn´t yet present in the code and this is why it isn´t recongnized correctly.

ZGP doesn’t provide clusters nor endpoints.  That’s why you don’t see these switches in the GUI.  I think the Enocean module sends some sort of ID (that’s how the Hue tap and the FoH switches are distinguished) but I would need to check the code for details.  I wouldn’t know how to ead what code the module in the Opus sends.

> My Opus switch comes with a PTM 215ZE" Enocean module, not the PTM 215Z" module.

So the Hue tap, and the Feller contain a 215Z and are shown in the REST API as Hue tap; the switches with a 215ZE are shown as FoH switches.  Only the latter support press vs release and (through the API) long presses.

agners, 2019-08-15:
@marschallphilipp  I am using 2.05.66 which is at this time the latest release. The switch is an official "Friends of Hue" but still seems not to work.

Sorry I did not meant to hijack this issue. Moved the Feller to #1754 and added some additional info.

marschallphilipp, 2019-11-20:
Anyone here who can help me to get the PTM 215ZE mdoule working correctly?
I managed to import my Opuss 55 switch into Phoscon but its recognized as a Hue Tap.
What I´m missing is the long press option.
I learned that a long press is "simulated" within the code because the modules only send a press / release package.
What lines in code would need to be douplicated and edited in order to get this module recognized correctly and get the button long press working?
Thanks!

mdolnik, 2020-02-09:
> Anyone here who can help me to get the PTM 215ZE mdoule working correctly?

I've had some success making changes to allow a Illumra dual-rocker switch (which uses the PTM215ZE module) to work, but the patch I provided needs work and will break the mapping of any other Friends of Hue switches.

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2108#issuecomment-583906292

Maybe you can see if the patch works for you, and someone can chime in as to what should be added/changed in order to properly add support for this switch.

mdolnik, 2020-02-13:
Added another patch which doesn't break the mapping with other Friends of Hue switches.

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2108#issuecomment-585559746

Waiting for someone with more knowledge to help determine if it's pull request worthy.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue.

closed 2020-06-05 by Mimiix