dresden-elektronik/deconz-rest-plugin #1580: 2.05.65 WEB socket events. (openHAB related?)

olemr, 2019-06-09:
When installing 2.05.65 after running 2.05.64 for a while, my openHAB/deconz binding integration acted up a bit.
I am not sure if it is a deconz or openHAB issue though, but in this case deconz was the only item updated.

This is what happened:
After restarting openHAB, all deconz sensors that had not sent an update was flagged as Offline.
New event showed up in PWA just fine, but device was still Offline in openHAB and no events was coming through.

Now, the only thing needed for it to work again, was a 'systemctl restart deconz', but it only worked for sensor events that were updated just prior to the restart of deconz.

Unsure at which end the trouble is.
Also logged an issue [here.](https://github.com/openhab/openhab2-addons/issues/5691)

Kane610, 2019-06-10:
Works fine with Home Assistant. Don't think it's the web socket.

olemr, 2019-06-10:
No, tested that last night. Same with the HUE-1 remote.
System had been up and running for days. Found my HUE-1 remote that I don't use on a regular basis (only as a device re-setter), no events when hitting buttons. Restarted deconz, back to normal.

manup, 2019-06-10:
This might be a timestamp issue.
Related code:

https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/rest_sensors.cpp#L2163-L2168

Can you please enable debug log info level 2 to confirm this is the issue:

    deCONZ --dbg-info=2 --http-port=80 | tee -a log.txt

olemr, 2019-06-10:
I'm running deconz headless from systemctl. Will a tee work then?

```

[Service]
User=omr
ExecStart=
ExecStart=/usr/bin/deCONZ -platform minimal --http-port=80

```

olemr, 2019-10-09:
Turned out to be an openHAB issue:
https://github.com/openhab/openhab2-addons/issues/5691

closed 2019-10-09 by olemr