dresden-elektronik/deconz-rest-plugin #1581: Several sensors do not use another route when their router goes offline

JBS5, 2019-06-09:
One of my IKEA GU10 went unavailable and the nodes which uses this GU10 spot as router goes offline too: A Tradfri Remote, a Aqara motion sensor and a Aqara multisensor. All three did not search for/find a new route. After power cycling the GU10, all three sensors are back online (using the specific GU10 as router).

Any clue why these nodes did not use another route when their router goes offline?

ebaauw, 2019-06-09:
This seems to be a "feature" of the Xioami devices.  I didn't know the Trådfri remote behaves similarly.  It might depend on how the router "went unavailable" (power off, firmware hang, ...).

JBS5, 2019-06-10:
The router went offline like described here 
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1261

No way to force a Xiaomi device to use another route, other then pair again I guess?

robertlandes, 2019-06-10:
@JBS5 Did you try and press the button on the multi sensor shortly? The blue light should be on for a second and make the sensor go “out of sleep” and send a “heartbeat “

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

JBS5, 2019-06-10:
@robertlandes Yes, tried that with both the Aqara multisensor and the Aqara motion sensor. No luck.

Smanar, 2019-06-10:
On my side, It's working the first hours just after the pairing, I have already see a Xiaomi device changed his router, but after a while, its dead, if the parent go offline, the device leave the network after somes hours, nothing to do.

thecem, 2019-06-11:
#1566 may like same here:
my case: Raspbian latest, latest deconz .65, raspbee latest FW 26330500, 3bplus (same issue with pi zero w latest and conbee I latest).
I have the same an the Xiaomi sensors get lost after time (no routers in zbnw, only battery powerd sensors and switches), pressing the button dosn't help, even it is not possible to repair or delete and new pair the sensor.
What works:
- powercycle the raspi (it is nessecary to powercycle the conbee/raspbee)
- push the sesnor button for 1 sec
After 10 - 20 sec you will see it in deconz updated.
This works a couple of days up to the first sensor is not reachable.....
First I tought this belongs to the range, but if reboot even the farest Sensor works for weeks, and the nearest get lost...
I think there must be a problem within the FW...

Smanar, 2019-06-11:
@thecem , you have all your devices in direct connection to the conbee ?

Have you tried to use an USB cordon to take away the conbee ?
If you have a GUI installation, can you enable LQI and take a look on value on green line ?

thecem, 2019-06-11:
Yes all devices direct connected via raspbee or via conbee (with usb cord between rasp zero w and conbee I), I have changed from zero w to raspi rapbee (and vise versa) and same problems.
sry no GUI installation...

closed 2019-10-16 by stale[bot]

cobch, 2020-03-05:
Hello,

I have exactly the same issue as stated above with my xiaomi multi sensors and motion sensorr.
All my setting is updated to latest version (deconz and Hass) but I spend some hours of adding my sensors that are disconnecting after a couple of hours or days... Anyone find a solution?

I use conbee with Legrand and Ikea trådfri but never get this kind of issue. It seems something related to xiaomi and deconz.

pergolafabio, 2020-03-05:
Indeed, xiaomi don't change route easy... Best practice is to pair it with a repeater, and never take out the repeater again ;(

cobch, 2020-03-05:
Yes. But what to do if I did it? Resetting and adding again the sensors lead each time to uncessfully results )-:

pergolafabio, 2020-03-05:
If I want to pair to specific router ,I make sure the router is far away from conbee, but offcourse still connected... Then pair xiaomi far away from conbee, so it's not able to pair with conbee... but must choose the other router instead...
One mesh complete, it's sticks....
(For xiaomi)

Ikea sensors, they are easy, they change and take the best route...

cobch, 2020-03-19:
Ok, finally I have found a work around to reinclude all Xiaomi sensors (temperature, motion, door or water leak) that is working great to fix the problem of losing connection anymore.

- Remove any sensors you have problem with from Phoscon
- Check that the sensor is removed from Deconz WebUI
- Restart Home assistant
- Repair the sensor with the help of Phoscon, rename it and wait that it is showing in Deconz WebUI
- On Deconz WebUI:  single click on the node (sensor you just added) and choose "Edit" -> "Reset selected nodes". The node will start to reconnect to a new router
- Choose then "Edit" -> "Permit join" and set the time to 60s (for example).
- you can then press shortly on the sensor you are adding for a couple of time. Maybe 20 times.
- After a while that is varying (maybe 1 to 10 minutes) a new green connection is showing for the node. You have managed to reinclude the sensor and then it should not loose the connection anymore (-:

I hope it will help people like me that just spend hours of reincluding  Xiaomi sensors in HA and Deconz.

pergolafabio, 2020-03-19:
Ah, that's a good trick...
I always deleted it completely, then did the join process again, untill it did connect to the router I wanted it to connect too...

thecem, 2020-03-20:
@cobch : which conbee(I/II) /raspbee do you use?

cobch, 2020-03-20:
@thecem I use conbee2.

thecem, 2020-03-20:
I had the same issues, after I migrated (complete reset of all devices) to conbee2 never had this issue.
On the other hand some devices are really picky which battery there is in, some stop working if the batt volt is lower than 2,95V. So try to double check the voltage of the batt.

cobch, 2020-04-14:
This is not due to battery issues but if you start changing let’s say a router that you are moving somewhere else so Xiaomi does not connect to a new router. If you are not changing you config Xiaomi are just working good (-: