dresden-elektronik/deconz-rest-plugin #1584: Install error Debian Buster

AntiHeld889, 2019-06-11:
Hi I wanted to install Deconz on a Debian Buster system today and get the following error.

`sudo apt-get install deconz
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
Die folgenden NEUEN Pakete werden installiert:
  deconz
0 aktualisiert, 1 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
52 nicht vollständig installiert oder entfernt.
Es müssen noch 0 B von 5.447 kB an Archiven heruntergeladen werden.
Nach dieser Operation werden 21,4 MB Plattenplatz zusätzlich benutzt.
(Lese Datenbank ... 190558 Dateien und Verzeichnisse sind derzeit installiert.)
Vorbereitung zum Entpacken von .../deconz_2.05.65-ubuntu-xenial-beta_amd64.deb ...
Entpacken von deconz (2.05.65-ubuntu-xenial-beta) ...
dpkg: Fehler beim Bearbeiten des Archivs /var/cache/apt/archives/deconz_2.05.65-ubuntu-xenial-beta_amd64.deb (--unpack):
 »/usr/lib/systemd/system/deconz-homebridge.service.dpkg-new« kann nicht geöffnet werden: Datei oder Verzeichnis nicht gefunden
Fehler traten auf beim Bearbeiten von:
 /var/cache/apt/archives/deconz_2.05.65-ubuntu-xenial-beta_amd64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
`

On debian stretch (9.9) the installation of the Ubuntu version works without problems.

Mysteriosis, 2019-06-11:
I confirm this bug on the last version of Hassbian, who already uses Buster.

> $ sudo apt install deconz
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following NEW packages will be installed:
  deconz
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/5,235 kB of archives.
After this operation, 20.7 MB of additional disk space will be used.
(Reading database ... 49859 files and directories currently installed.)
Preparing to unpack .../deconz_2.05.64-raspbian-stretch-stable_armhf.deb ...
Unpacking deconz (2.05.64-raspbian-stretch-stable) ...
dpkg: error processing archive /var/cache/apt/archives/deconz_2.05.64-raspbian-stretch-stable_armhf.deb (--unpack):
 unable to open '/usr/lib/systemd/system/deconz-homebridge.service.dpkg-new': No such file or directory
Errors were encountered while processing:
 /var/cache/apt/archives/deconz_2.05.64-raspbian-stretch-stable_armhf.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

akacruel, 2019-07-19:
I also confirm this bug on Hassbian (buster) with the problems which are mentioned above. Version deconz_2.05.66.

theninth, 2019-07-20:
I also have the same problem on Debian Buster amd64 using the ubuntu repos.

as19git67, 2019-08-06:
Same problem here with buster

matfouc, 2019-08-08:
Hi ! i just did a fresh installation of Hassbian on Buster and i also experienced the same issue.

lepatrick, 2019-08-14:
Hey @manup is it planned to fix it or should we go with Ubuntu 18.04?

manup, 2019-08-15:
Hi, yes the build system will be extended with the Debian Buster on AMD64 platform soon. Till than please refer to the Ubuntu version which should also work fine.

Raspbian Buster is already included and works with Raspberry Pi 4.

closed 2019-08-28 by AntiHeld889