dresden-elektronik/deconz-rest-plugin #1585: Routers not joining mesh

OldSystemMan, 2019-06-11:
I am using a Raspbee to run a small network, but the network connections are all direct to the coordinator, with no mesh formed between the routers, and no return value on LQI, as per attached screenshot.
[image]

I am using version 2.05.64, firmware version  0x26330500, and the routers are mainly Gledopto bulbs and Xiaomi curtain motors.
I have restatred Deconz numerous times, but the mesh never forms.

Any suggestions?

[image]

manup, 2019-06-11:
The mesh is likely fine, the reason no interconnected lines are shown is that we don't query the neighbor tables of certain devices, like Xiaomi mains powered devices anymore. These tend to have firmware bugs which got triggered by such requests.

So in short, the mesh should work but just isn't displayed.

OldSystemMan, 2019-06-11:
So Gledopto bulbs aren't queried either? It's just that I have Gledopto bulbs in another installation, and they do show as being in the mesh.