dresden-elektronik/deconz-rest-plugin #1586: change port 443 to something else

Staubgeborener, 2019-06-11:
Is there a chance to change the port 443 from deconz to something else? I want to install nextcloud beside deconz on my raspberry, and i'll get always an error, 'cause nextcloud needs port 443. There is no really a reson, that deconz runs on port 443 (also weird, because i can reach deconz localy on port 90 and this is good, i don't need any further ports for this). 

So i do `sudo netstat -ntulp | grep 443` and i always get the deconz instance. I only see guides here, how to change port 90 to port 8080, but nothing about this port 443. I cant find 443 in any config file of deconz (also, 8080 is blocked by homebridge).

ebaauw, 2019-06-11:
See #238

Staubgeborener, 2019-06-11:
My problem is, that ` /usr/bin/deCONZ --ws-port=8087` want to open the deconz gui. I work with putty. But when i connect over VNC and use this, this command seems to only open the deconz gui with the port 8087. Also there is no `/etc/systemd/system/deconz.service` on my setup to change the ws-port right there (exactly what i was watching about, a conf-file to simply change the ws-port). Basicaly, there is nothing like deconz* in /etc/systemd/system (but yes, deconz is installed and runs as a webserver).

Staubgeborener, 2019-06-11:
Nevermind, it looks like, that this here works now: https://github.com/dresden-elektronik/phoscon-app-beta/issues/26#issuecomment-410553910
I had to create the `override.conf` by myself.

closed 2019-06-11 by Staubgeborener

Patricen1, 2020-05-25:
Hello,

I'm a newbee in the same situation.
How and where did you create override.conf, what is the content?

Thanks

Patricen

Smanar, 2020-05-25:
Hello, perhaps https://github.com/dresden-elektronik/deconz-rest-plugin/issues/295#issuecomment-347615035 ?

Staubgeborener, 2020-05-25:
Hey @Patricen1 , see the answer fo @Smanar 
create `override.conf` in `/etc/systemd/system/deconz.service.d/override.conf` with a content like this:
```shell
# /etc/systemd/system/deconz.service.d/override.conf
[Service]
ExecStart=
ExecStart=/usr/bin/deCONZ -platform minimal --http-port=8080 --ws-port=8081 --auto-connect=1 --dbg-error=1
```

You can also change the ports there.