dresden-elektronik/deconz-rest-plugin #1587: No more lines | every device unavailable

maui1911, 2019-06-11:
Not sure what happened, no errors from logs. But every device is unavailable in the Phoscon app and there are no more lines in the VNC app. 

I'm running deconz in docker with a Conbee v1 stick. Updated firmware to the last version.

[image]

maui1911, 2019-06-12:
I managed to add a new thing, nice new green line.

How do I get the stick to connect to all the old things, please help :(

[image]

Smanar, 2019-06-12:
Arf, all devices are grayed. It mean they are not in the network ATM. It's like you have loose all yours devices.
Have you tried the "network configuration recovery" https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_59 ?
Have you tried unplug/replug the conbee (to check for firmware hangs) ?

maui1911, 2019-06-12:
I think it happend after I did a firmware upgrade.

But thanks a lot man, loading an old configuration worked like a charm!

[image]

closed 2019-06-14 by maui1911