dresden-elektronik/deconz-rest-plugin #1588: ignore new node - limit reached

d-lebed, 2019-06-12:
I'm using Conbee II with 264A0700 Firmware and deCONZ 2.05.65

I'm not able to add more then 5 Devices. Currently I have 2 Xiaomi Plugs and 3 Aquara/Xiaomi Sensors added. When I'm trying to add 6th I can see something like that in logs:

```
02:43:36:554 APS-DATA.indication from unknown node 0x00158D0002A21A66
02:43:36:555 ZCL attribute report 0x00158D0002A21A66 for cluster 0x0000, ep 0x01
02:43:36:555 	payload: 050042126c756d692e766962726174696f6e2e61713101002008
02:43:36:555 0x00158D0002A21A66 skip Xiaomi attribute 0x0000
02:43:36:559 APS-DATA.indication from unknown node 0x00158D0002A21A66
02:43:36:559 ZDP device announce: 0x00158D0002A21A66, 0xCBCA, 0x80
02:43:36:559 ignore new node 0x00158d0002a21a66 - limit reached
02:43:36:567 Node 0x00158D0002C8ED65 is known by 1 neighbors, last seen 700 s
02:43:36:848 poll node 00:15:8d:00:02:8a:cb:96-03-000c
02:43:36:848 Poll ZHAConsumption sensor node Consumption 13
02:43:37:047 Node 0x00158D00027B62D5 is known by 1 neighbors, last seen 1202 s
02:43:37:527 Node 0x00158D000237A113 is known by 1 neighbors, last seen 30 s
02:43:37:687 poll node 00:15:8d:00:02:37:a1:13-01
02:43:37:688 Poll light node xiaomi plug 1
02:43:37:756 APS-DATA.indication from unknown node 0x00158D0002A21A66
02:43:37:757 ZCL attribute report 0x00158D0002A21A66 for cluster 0x0000, ep 0x01
02:43:37:757 	payload: 01ff422e0121770b0328230421a8010521250006240100000000082108030a2161f398212800992100009a25affe5b016904
02:43:37:757 0x00158D0002A21A66 extract Xiaomi special attribute 0xFF01
02:43:37:757 	01 battery 2935 (0x0B77)
02:43:37:757 	03 temperature 35 °C
02:43:37:757 	04 unknown 424 (0x01A8)
02:43:37:757 	05 RSSI dB (?) 37 (0x0025)
02:43:37:757 	06 LQI (?) 4294967296 (0x0100000000)
02:43:37:757 	08 unknown 776 (0x0308)
02:43:37:757 	0a unknown 62305 (0xF361)
02:43:37:758 	98 unknown 40 (0x0028)
02:43:37:758 	99 unknown 0 (0x0000)
02:43:37:758 	9a unknown 193506983373060 (0xAFFE5B016904)
```

I've tried to reset Conbee using Phoscon Gateway -> Reset functionality and, I believe, before the reset I had 6 devices added, but still was not able to add more. As I know Conbee 2 should support 200 devices. On Help -> About in deConz I see next line:

Node limit 5 (max. 200)

Can anybody explain me how these limits works?

agners, 2019-08-05:
Very similar issue here. I connected 5 Ikea lamps (4 GU10 and one panel). I do have two more GU10 bulbs, but they just won't pair. I always get `ignore new node 0x....... - limit reached`. I am using 2.5.66 on arm64.

twallaert, 2019-08-18:
I receive the same log message when trying to add more devices. Currently I have 39 nodes working in deCONZ without any problem (I restored a backup in Phoscon from a previous **amd**64 setup), but now it's not possible anymore to add more devices on my new **arm**64 host (Odroid N2). The log message I find is:

> 09:11:13:157 ignore new node 0x00158d0001e0fa66 - limit reached

deCONZ mentions `Node limit 5 (max. 200)` in the About section.

I am using a Conbee USB stick with marthoc/deconz Docker, running version 2.05.66 (firmware 26330500) on arm64.

What can I do to solve this problem?

wvuyk, 2019-08-18:
Just guessing here, do not really know. Is deCONZ total not limited to 200 nodes, but is one router maybe limited to 5 connections? So if you do not have extra routers (lights/repeaters/plugs/gateway) available, extra devices are not possible? 
39 nodes in this case would suggest you have 5 routers (say 4 lights and gateway) and maybe 34 sensors? 
It is a bit of wild guess here, but one that could point in a direction here?

Smanar, 2019-08-19:
Good question, from the wiki (and no sure it still good values) the maximum direct connection from the gateway is 32, but I haven't info for routers.

agners, 2019-08-19:
@manup confirmed to me by email that this is an accidental limit in the ARM64 port and should be fixed in the next release.

twallaert, 2019-08-19:
Thanks very much indeed for this reassuring reply. I'm eager to see this problem solved in the next release. Is there perhaps already a release date for this update?

Just to be curious: would this issue be solved if I reverted to a previous version of deCONZ in the mean time?

manup, 2019-08-21:
The release is cooking and should be arrive this week.

>Just to be curious: would this issue be solved if I reverted to a previous version of deCONZ in the mean time?

Unfortunately this affects all previous releases.

pvizeli, 2019-09-03:
That is fixed now?

manup, 2019-09-03:
Yes the issue is fixed in version 2.05.67. Affected was only the arm64 platform.

pvizeli, 2019-09-03:
Perfect, the aarch64 version is released

agners, 2019-09-04:
@manup installed the aarch64 version through hass.io and I can confirm that it supports now 200 devices as the 32-bit ARM version did. From my point of view this can be closed. Thanks!

0evmc, 2019-10-26:
Encountered the erroneous node limitation with raspbee using DeCONZ add-on for Hass.io. For some reason I was unable to update the firmware - it would say successful, but the firmware version never changed. I ended up booting a PI to one of the preconfigured images [here](https://phoscon.de/en/raspbee/sdcard), which allowed me to update the raspbee firmware. After that I was able to connect the raspbee back to hass.io and add all the nodes I needed to.

closed 2020-03-02 by stale[bot]