dresden-elektronik/deconz-rest-plugin #1589: 2 hue lights turning crazy (looping between on and off very quickly)

ooii, 2019-06-12:
Hi, 

Since a couple of days, I have two Hue lights which switch between on and off tens of times per second. 
These two lights have been working fine for months with a Hue motion sensor. This latter changes the value of a Clip Generic Status sensor. I'm using such rules and scenario in several other rooms at home that are still working fine.

I attach here a ~1 second capture of `dc_eventlog` where you can see that these lights (52 and 7) change their status very quickly. I can get what's happening. As you can see, there is no sensor changing any value. The groups you see (5, 9, and 4) are groups to which these two lights belong.

Any idea what's going on?
Thanks.

```
/lights/52/state: {"alert":null,"bri":177,"on":true,"reachable":true}
/groups/4/state: {"all_on":false,"any_on":true}
/lights/7/state: {"alert":null,"bri":75,"on":true,"reachable":true}
/lights/52/state: {"alert":null,"bri":177,"on":true,"reachable":true}
/lights/7/state: {"alert":null,"bri":75,"on":true,"reachable":true}
/lights/52/state: {"alert":null,"bri":177,"on":true,"reachable":true}
/sensors/5/config: {"battery":100,"group":"25500","on":true,"reachable":true}
/lights/7/state: {"alert":null,"bri":75,"on":true,"reachable":true}
/lights/52/state: {"alert":null,"bri":177,"on":true,"reachable":true}
/lights/7/state: {"alert":null,"bri":75,"on":true,"reachable":true}
/lights/52/state: {"alert":null,"bri":177,"on":true,"reachable":true}
/groups/4/state: {"all_on":false,"any_on":false}
/lights/7/state: {"alert":null,"bri":75,"on":false,"reachable":true}
/lights/52/state: {"alert":null,"bri":177,"on":true,"reachable":true}
/lights/52/state: {"alert":null,"bri":177,"on":false,"reachable":true}
/groups/5/state: {"all_on":false,"any_on":false}
/groups/9/state: {"all_on":false,"any_on":false}
/lights/7/state: {"alert":null,"bri":75,"on":true,"reachable":true}
/lights/52/state: {"alert":null,"bri":177,"on":true,"reachable":true}
/groups/4/state: {"all_on":false,"any_on":false}
/groups/5/state: {"all_on":false,"any_on":true}
/groups/9/state: {"all_on":true,"any_on":true}
/lights/7/state: {"alert":null,"bri":75,"on":false,"reachable":true}
/lights/52/state: {"alert":null,"bri":177,"on":false,"reachable":true}
/lights/52/state: {"alert":null,"bri":177,"on":false,"reachable":true}
/lights/7/state: {"alert":null,"bri":75,"on":false,"reachable":true}
/groups/5/state: {"all_on":false,"any_on":false}
/groups/9/state: {"all_on":false,"any_on":false}
/lights/52/state: {"alert":null,"bri":177,"on":false,"reachable":true}
/lights/7/state: {"alert":null,"bri":75,"on":false,"reachable":true}
/lights/7/state: {"alert":null,"bri":75,"on":false,"reachable":true}
/lights/7/state: {"alert":null,"bri":75,"on":false,"reachable":true}
/lights/52/state: {"alert":null,"bri":177,"on":true,"reachable":true}
/groups/5/state: {"all_on":false,"any_on":false}
/groups/9/state: {"all_on":false,"any_on":false}
/sensors/36/state: {"lastupdated":"2019-06-12T09:59:51","temperature":1742}
/lights/52/state: {"alert":null,"bri":177,"on":false,"reachable":true}
/lights/7/state: {"alert":null,"bri":75,"on":true,"reachable":true}
/lights/52/state: {"alert":null,"bri":177,"on":false,"reachable":true}
/lights/7/state: {"alert":null,"bri":75,"on":true,"reachable":true}
/groups/5/state: {"all_on":false,"any_on":true}
/groups/9/state: {"all_on":true,"any_on":true}
/lights/7/state: {"alert":null,"bri":75,"on":false,"reachable":true}
/groups/5/state: {"all_on":false,"any_on":false}
/groups/9/state: {"all_on":false,"any_on":false}
/lights/52/state: {"alert":null,"bri":177,"on":true,"reachable":true}
/groups/4/state: {"all_on":false,"any_on":true}
/lights/52/state: {"alert":null,"bri":177,"on":false,"reachable":true}
/groups/4/state: {"all_on":false,"any_on":false}
/lights/7/state: {"alert":null,"bri":75,"on":false,"reachable":true}
/lights/52/state: {"alert":null,"bri":177,"on":false,"reachable":true}
/lights/52/state: {"alert":null,"bri":177,"on":true,"reachable":true}
/lights/7/state: {"alert":null,"bri":75,"on":false,"reachable":true}
/lights/52/state: {"alert":null,"bri":177,"on":false,"reachable":true}
/lights/7/state: {"alert":null,"bri":75,"on":true,"reachable":true}
/lights/52/state: {"alert":null,"bri":177,"on":false,"reachable":true}
/groups/5/state: {"all_on":false,"any_on":true}
/groups/9/state: {"all_on":true,"any_on":true}
/lights/7/state: {"alert":null,"bri":75,"on":false,"reachable":true}
/groups/5/state: {"all_on":false,"any_on":false}
/groups/9/state: {"all_on":false,"any_on":false}
/lights/52/state: {"alert":null,"bri":177,"on":true,"reachable":true}
/lights/7/state: {"alert":null,"bri":75,"on":true,"reachable":true}
/lights/52/state: {"alert":null,"bri":177,"on":true,"reachable":true}
/groups/4/state: {"all_on":false,"any_on":true}
/groups/5/state: {"all_on":false,"any_on":true}
/groups/9/state: {"all_on":true,"any_on":true}
/lights/52/state: {"alert":null,"bri":177,"on":true,"reachable":true}
```

ooii, 2019-07-24:
Hi,

I'm still facing this issue with two groups of lights. 
If someone could help with that please.

Smanar, 2019-07-24:
Hello, have you tried to disable your rules/scenario one by one, to check if the problem don't come from one of them ?

ooii, 2019-07-25:
I have a lot of rules and cannot disable them one by one to check. But I used `dc_eventlog` to follow rules execution and no rule is firing these lights.
I have to disable deconz integration in Home Assistant and check if the issue is not coming from there.

ebaauw, 2019-07-25:
> I have a lot of rules and cannot disable them one by one to check.

You can turn them all off by:
```
$ for rule in $(ph get -al /rules | grep '/status:"enabled"' | cut -d / -f 2) ; do
>   ph put /rules/${rule} '{"status": "disabled"}'
> done
```

> But I used `dc_eventlog` to follow rules execution and no rule is firing these lights.

deCONZ doesn't (yet?) issue a websocket notifcation when a rule fires, so `dc_eventlog` won't be of help here.  You need to follow the deCONZ log (with the appropriate `--dbg-`... parameter) and filer for rule execution:
```
$ sudo journalctl -afn 1000 -u deconz-gui | grep 'trigger rule'
```

ooii, 2019-07-25:
Thanks @ebaauw, I'll try that this evening.

ooii, 2019-07-25:
So, I ran tests with what you have suggested @ebaauw, and I don't have any output in `journalct`. I'm running deConz with the `-platform minimal` option so no `deconz-gui` unit probably but Ì have no `deconz` neither.
I tried without the `-u` option and, except `dc_eventlog`, I don't see any log related to deConz.
Note that I have started it with the following options: `--dbg-info=2 --dbg-aps=1 --dbg-error=1`.

But in deConz logs, I have these lines repeated about one thousand times every second until I stop deConz. And these lights 12 and 13 are the one being crazy.

```
22:34:28:613 discard light state push for 13: state/bri (already pushed)
22:34:28:615 discard light state push for 12: state/on (already pushed)
22:34:28:616 discard light state push for 12: state/bri (already pushed)
22:34:28:617 discard light state push for 13: state/bri (already pushed)
22:34:28:619 discard light state push for 13: state/on (already pushed)
22:34:28:620 discard light state push for 12: state/bri (already pushed)
22:34:28:621 discard light state push for 13: state/on (already pushed)
```

After that, I added option `--dbg-http=1` and now, I can see in the logs a lot (more than 6000 in a minute) of rest messages like this one:

```
22:49:04:417 HTTP API PUT /api/E9B2384618/lights/12/state - 192.168.1.11
22:49:04:417 Text Data: 	{"on": true, "bri": 141}
22:49:04:417 ApiMode: 0
22:49:04:417 ~Resource() /groups 0x7ffcda8e0560
22:49:04:417 add task 6803 type 11 to 0x001788011040F879 cluster 0x0008 req.id 163
22:49:04:418 ~Resource() /groups 0x7ffcda8e0560
22:49:04:418 drop request to zombie (rx = 0)
22:49:04:418 [{"success":{"/lights/12/state/on":true}},{"success":{"/lights/12/state/bri":141}}]
22:49:04:418 HTTP client hdr detected:
PUT /api/E9B2384618/lights/13/state HTTP/1.1
Host: 192.168.1.11
User-Agent: HomeAssistant/0.96.3 aiohttp/3.5.4 Python/3.6
Accept: */*
Accept-Encoding: gzip, deflate
Content-Length: 13
Content-Type: text/plain; charset=utf-8
```

And the brightness value changes from one message to another.

These two lights do not belong to any deconz rule and it seems these messages are issued by home-assistant. So I think I have a first clue about the issue.
But I know I have no automation in HA which uses these lights. So, I'm wondering why HA is doing that. Any idea where I should look, given that HA logs reports only brightness value change with the following options?

```
logs:
    pydeconz: debug
    homeassistant.components.deconz: debug
```

Thanks.

ooii, 2019-07-26:
Hi,

I think I have an idea about the issue but I'm still unable to understand what happens exactly.
In deconz, I have a light named `jardin` but I have also a light named `jardin`.
In HA, I have 3 "lights" containing `jardin` in their name, as you can see in the screenshot.
[image]
In HA's deconz integration panel, I can see only `light.jardin_2` which is unavailable.

Okay, I can understand that I should not have given the same name to a light and a group that contains this same light. But, why switching on/off one of them enters in an infinite loop of switching on/off the group of lights? If you have any idea, I'd be interested.
Thanks.

closed 2019-11-30 by stale[bot]