dresden-elektronik/deconz-rest-plugin #1591: .65 release, my xiaomi pir, is still not reconnecting

pergolafabio, 2019-06-14:
on every floor i have a osram smart plug to act as a router, i have placed them 3 days ago
but my xiaomi pir sensor is just loosing connection to the smart plug after about 1-2 hours
i need to press the button on it to make it work again

i am on latest .65 release with firmware 26330500

see my mesh here below screenshot

the xiaomi_sensor1 is like 2 meters away to plug3

what could be wrong?

screen mesh here : https://www.dropbox.com/s/xfdmcrjmtzxksa2/xiaomi-mesh.JPG?dl=0

pergolafabio, 2019-06-14:
the sensors xiaomi have firmware : Version 20170627
the osram plugs have firmware : Version V1.04.12

can those be updated?

pergolafabio, 2019-06-18:
i am on  2.05.65 , conbee 1 , fw 26330500
for test i have it now verry simple, all end devices are direct connected to the conbee (with extension cable)
when connected, they are all green , so should be good

after a while , i see a drop ... for test , i only have : 

xiaomi pir sensors & ikea remote switch (dimmer)
see here : 
https://www.dropbox.com/s/95xgpemaseh83qj/simplemesh.JPG?dl=0

what can i do ?

manup, 2019-06-18:
Does the connection come back when you press the button of the remote or the sensor?

pergolafabio, 2019-06-18:
yes, for the ikea it comes back

pergolafabio, 2019-06-18:
for the xiaomi, well sometimes, if i shortpress it
sometimes i need to repair completely

manup, 2019-06-18:
When you press the button 5 times quickly on some of the Xiaomi sensors this should invoke a connection test.

Rejoining it completely should only be needed when it sticks to a no longer available parent, but this is not possible in this simple setup, or when it didn't rejoin correctly due a bug in Xiaomi firmware, see: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1595#issuecomment-502462107

pergolafabio, 2019-06-18:
xiaomi firmware? is it possible to update that? or do you mean conbee firmware?

manup, 2019-06-18:
The firmware of the sensor has the bug:
When it receives a "Leave + Rejoin" command, which might happen in various cases, it will just do the Leave- but not the Rejoin part. It happens only in some rare setups with certain router devices.

> is it possible to update that

I'm not aware of a newer firmware for Xiaomi devices which fixes the bug.

pergolafabio, 2019-06-18:
ok, thnx in advance, i indeed have some osram smart plugs to act as routers (not in network now)
if i place them, its even worse, i think connection fails after 2-3 days
when all connected to conbee, its more stable, like 10-20 days

pergolafabio, 2019-06-19:
@manup 
i press the ikea button, it comes back
walking in front the xiaomi, doesnt detect , doesnt come back
pressing button 5 times doesnt do anything
suddenly , a few hours later, connected again
again, few hours later, again xiaomi disconnected

frustating :(

i am on 2.05.65 , conbee 1 , fw 2633050

pergolafabio, 2019-06-21:
@manup 
its getting worse, 2 xiaomis are always failing after about 1-2 days, connection is gone
sometimes it comes up later, or noth at all, need to press those buttons again

do you think .66 will resolve this? i cant update it yet, since using marthoc addon, thats still on  .65 release

or what can i do to investigate this and help troubleshooting?

Smanar, 2019-06-21:
Your 2 xiaomis have direct connexion (you are still not using yours Osram plug)  ?
Can you enable LQI, just to check quality connexion ?

pergolafabio, 2019-06-21:
yes, indeed that was strange, no idea what that endpoint, was if i delete it , it just comes back
also not visible in phoscon
but anyway, i removed all osrams plug now, in this test phase, all xiaomi  are now in direct connection of conbee

pergolafabio, 2019-06-21:
sorry, i am wrong, i thougt you were talking about 0XD860 , connected to plug3
well, i have no idea what that is :)

and yes, seems my xiaomi pir, have 2 links , should that not be possible?

but, now offcourse, just one link , no osrams at this moment

Smanar, 2019-06-21:
Hu, it's realy strange, EndDevice can't have 2 connexions.
Have you try to enable LQI for some minuts before taking a SS ?

pergolafabio, 2019-06-21:
now its like this, osrams are not in the socket

https://www.dropbox.com/s/ql6ilywqc3ssh38/xiaomi-mesh3.JPG?dl=0

you see one xiaomi is now not connected (again) :-)
if i press the buttons on it, it starts working for a short while

Smanar, 2019-06-21:
Not connected but still in network.
Press the "LQI" button to check Link quality (but take carre realy cpu intensive, it's just for tests).

pergolafabio, 2019-06-21:
Ok, Will do that later when I am home...

Should I make the connection first by pressing the buttons on the xiaomi, or just like it's now

Smanar, 2019-06-21:
If the device haven't connection, you can't have LQI, but I don't know how work deconz, you can have a connexion but not displayed on GUI, because not enought data.

pergolafabio, 2019-06-24:
ok, back home
for testing now, i have plugged back in 1 osram in middle of house, rebooted my HassOS and deconz
a few end devices were not working, so took out battery, now they are all up and running
see mesh : https://www.dropbox.com/s/4fl9i5n90qef9rq/xiaomi-mesh4.JPG?dl=0

again a few end devices are connected 2 times :)

also for that LQI testing, how does that work? i can start testing, its sending packets, then what? :)

Smanar, 2019-06-25:
When you press the "LQI" button (on deconz toolbar), deCONZ will write value on "green connexion line". More accurate than color, but don't let it all the night.

BTW the sensor "Xiaomi_sensor_1" that have leave your network last time, haven't a fantastic connexion on your last SS.

And if someone can explain me, how he can have 2 connexions on end-device everytime he make tries  ?

pergolafabio, 2019-06-25:
ok, for testing now, i have plugged in 1 osram, the LQI from xiaomi 1 to plug1 is like  10/0 , thats seems bad (orange color) ?
xiaomi is on groundfloor, osram is on first floor, conbee is on second floor
when i dont use the osram, so 2 floors between, the color is green, strange isnt it?

also for testing now, i changed wifi channel to 25 instead of 11

so next testing, when this one fails, is to place another osram as router

Smanar, 2019-06-25:
So, I m not a pro, so I m not sure, but LQI is link quality, it's not the same thing than the power signal. If you haven't good LQI, you realy need to make something (and yes for me 10 is realy bad).

And it seem osram plug cause problem with Xiaomi device (Xiaomi problem), there is some issue about this problem in this github.

pergolafabio, 2019-06-26:
Updated today to 66 (marthdoc) add-on, no my end devices only have 1 connection ;)

closed 2019-09-21 by pergolafabio