dresden-elektronik/deconz-rest-plugin #1592: Xiaomi Smart Plug miss sensor device (ZHAConsumption)?

stccmaya, 2019-06-14:
I have one Xiaomi smart plug (modelid: "lumi.plug.mitw01", swversion: "01-10-2018")

The plugin creates three devices (lumi.plug; ZHAPower)

Have some problem

1. loss one device (ZHAConsumption)

2. ZHAPower device no value : (

------------------------------------
detail information: (I use: ConebeeII)

ZHAPower

**************************** Object info *****************************
{
"state": {
"lastupdated": "2019-06-06T02:41:11",
"power": -1
},
"config": {
"on": true,
"reachable": true
},
"name": "Power 3",
"type": "ZHAPower",
"modelid": "lumi.plug.mitw01",
"manufacturername": "LUMI",
"swversion": "01-10-2018",
"uniqueid": "00:15:8d:00:02:8b:3d:0c-01-0b04",
"ep": 1,
"etag": "bf511a13f8b44cb357694252ad67edd1"
}
************************* End Object info ***************************
[image]
[image]
[image]

stccmaya, 2019-07-10:
no any respone : (

renedis, 2019-07-19:
Select analog Input on the second device. After that, pres READ on the left side

closed 2019-11-23 by stale[bot]