dresden-elektronik/deconz-rest-plugin #1594: Elko RF dimmer does not respond to brightness 255

ricmik, 2019-06-14:
Hi,

This has been an issue for several deCONZ versions and Conbee firmware versions.
The Elko RF dimmer does not respond when setting brightness to 100% or "255". Setting the brightness to 99% or "254" is working fine.

Even if the actual state of the dimmer is not changing, the state in deCONZ is updated to brightness 255 until the next poll.

The result from the deCONZ REST API is "success" in both cases.

Current version 2.05.65, Firmware 26330500

Examples:

**Non-working:**
```
PUT /api/XXXXXXXX/lights/5/state HTTP/1.1
{
  "bri": 255
}
Result:
[{"success":{"/lights/5/state/bri":255}}]
```

**Working:**
```
PUT /api/XXXXXXXX/lights/5/state HTTP/1.1
{
  "bri": 254
}
Result:
[{"success":{"/lights/5/state/bri":254}}]

ricmik, 2019-10-12:
Is it possible to fix this without doing changes in the deconz source code?

Level 254 is the maximum level (100%) for the Elko dimmer.

ricmik, 2019-10-23:
This should be pretty easy to fix. Any help is appreciated. @ebaauw maybe? :)

ebaauw, 2020-02-20:
Sorry, I must have missed this one before.  On a positive note, stale bot is good for something...

The maximum setting for _Current Level_ in ZigBee is 254; value 255 is used to indicate an invalid value (or previous value for on and poweron settings).  I recently had some issue with this for the window controller devices as well, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2257.

The Hue bridge actually sends _Level_ 254 when `state.bri` is set to 255, but the REST API plugin still doesn't do so.  See https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1111#issuecomment-456464872.

So the dimmer is correct ignoring the 255 value.

Not sure what the best way to fix this.  Ideally, I would not accept value 255 for `state.bri`.  For Hue API compatibility, best map 255 to 254.

uphillbattle, 2020-05-07:
Depending on what you use for home automation, this may be of assistance until the REST API stops sending 255:

I have solved this by use of an automation in Home Assistant:

```
- id: limit_brightness_spisestue
  alias: Limit brightness spisestue
  trigger:
    platform: template
    value_template: >
      {% if is_state('light.spisestue', 'on') %}
        {{ states.light.spisestue.attributes.brightness > 254 }}
      {% else %}  
        {{ False }} 
      {% endif %} 
  condition:
    - condition: template
      value_template: "{{ is_state('light.spisestue', 'on') }}"
  action:
    - service: light.turn_on
      data:
        entity_id: light.spisestue
        brightness: 254

```

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix