dresden-elektronik/deconz-rest-plugin #1595: Dissapointing about Conbee performance :(

jascdk, 2019-06-16:
I just got the Conbee II with the mailman. I have it setup on a Intel Nuc running latest Home Assistant. 

I have no problems on installing the Conbee, the deconz addon and afterwards the Zigbee devices. I use alot of Tradfri bulbs and alot of Xiaomi aqara sensors (door and PIR) and buttons. 

I can add the devices to the Conbee, and it performs ok at first, but after a night or so it looses its connection to alot of my xiaomi sensors. 

The tradfri bulbs are pretty stable, and they are being connected ok. But the xiaomi sensors - arrrrggghhhh :( 

The xiaomi sensors even dies out when close to either a tradfri bulb or the Conbee stick - i dont get it. Have tried to reset the xiaomi devices, reset conbee, restart system, change zigbee channel and so on - no luck:(

I am trying to go back to my tradfri gateway and xiaomi hub, as they were offering a stable system. I hope for some updates on the conbee:)

pergolafabio, 2019-06-16:
yeah, lots of issues here also with xiaomi pir sensors, latest firmware now
they just loose connection after a few hours
readding/pairing doesnt help

also battery are all reporting 100%, not right offcourse :)

jascdk, 2019-06-16:
@pergolafabio Nice to know we agree on each other :D We must hope they will add some updates, which improves the reliability ;) 

I will place the conbee in the drawer for now and use my old setup :)

daaaaa, 2019-06-16:
Same problem here with Xiaomi devices.
I have the Xiaomi two button switch and it does not work most of the night.

While the philips one works perfectly.

robertlandes, 2019-06-16:
> I just got the Conbee II with the mailman. I have it setup on a Intel Nuc running latest Home Assistant.
>
> I have no problems on installing the Conbee, the deconz addon and afterwards the Zigbee [...]

I can't say something about PIR and buttons/switches. I'll have however 8 Xiaomi window/door sensors and 7 Multi sensors (temp, humidity, ..) running perfectly fine with hass.io. I am howver not using the docker container addon provided by hass.io, but running it on a separate raspi (debian stretch) with phoscon/deCONZ REST.

What versions are you running? (HA, Phoscon/deCONZ and Conbee II firmware)

daaaaa, 2019-06-16:
I have a lot window/door sensors and they work prectly too.
I use hassio and both deconz and hassio are updated to the last version.

Still some Xiaomi switches work rarely.

pergolafabio, 2019-06-16:
maybe the battery is just dying? i dont see the battery level, they all report 100% everywhere
In HA, and also in the mesh overview

Smanar, 2019-06-16:
If you are using GUI, can you take a look on choosen router (some of them are realy bad) and LQI value ? (with enabling it in menu).
Have you tried too the extension usb cable ?

pergolafabio, 2019-06-16:
Mine are all direct connected now, all green...

If I place osrams between them, it's even worse . ;)

manup, 2019-06-16:
Can you please provide more details about the setup:
* deCONZ version
* ConBee II firmware version
* Do you use an USB extension cable?

With latest versions Xiaomi devices should stay online at least when connected directly to the gateway and most routers with a few exceptions (see below).

A bug was recently discovered where some Zigbee 3.0 routers ask end-devices to rejoin if they time out in their neighbour table. The Xiaomi end-devices don't implement that correctly and will just leave the network and repairing them to the network is required.

Related issues: #1488 and https://github.com/Koenkk/zigbee2mqtt/issues/1474

Currently known routers which have this problem are:
* OSRAM US Sylvania plug 
* Müller Licht tint lights

Note the bug is in the Xiaomi devices not the routers. As a workaround I suggest powering off all problematic devices during pairing so that the Xiaomi devices select a "working" parent device.

robertlandes, 2019-06-16:
> maybe the battery is just dying? i dont see the battery level, they all report 100% everywhere
> In HA, and also in the mesh overview

Until latest release 2.05.65 there was a problem with battery level for Xiaomi Multi sensors being always `null` through REST API. Somehow HA manages to map that to 100% (for what it's worth) for all of them.
Upgrading to latest deCONZ version fixed this for me. (it might howver take up to 2h for the sensors to report correct battery level after upgrade).

jascdk, 2019-06-17:
@robertlandes and @manup - i am using the latest hassio and latest Deconz hassio addon and latest conbee firmware . Maybe I should try to run it on a separate pi instead ? Now I went back to the Xiaomi gateway for Xiaomi devices and I use the Connbee for the TRÅDFRI bulbs . That works fine - and I am also using the extension usb cord

robertlandes, 2019-06-17:
@jascdk Just to confirm, you are running:

- Software Version 2.05.65 / 24.5.2019
- Firmware 26490700 (0x26490700)

(Check via Phoscon app > Gateway > Advanced)

As mentioned before, upgrading to 2.05.65 fixed my problem with battery level reporting of Xiaomi Multi Sensors and other than that, all my Xiaomi devices (leak sensors, multi sensors, door sensors and motion sensors) are running perfectly fine. I have also a mixed setup running Philips and Ikea lamps and switches, Innr and Osram plugs. (~ 100 devices).

So it might be related to the hass.io docker container. Although I am not using it, I'll install it to check at least software version etc....

jascdk, 2019-06-17:
@robertlandes 

[image]

Yes it is the same software - but the firmware is a little different ? I can't update it - and looks like the newest one?

Nice that yours is working - and also working with the battery states - all my sensors were 100 procent :(

manup, 2019-06-17:
0x26420700 is the initial shipped firmware, it has issues where especially Xiaomi devices drop off after some time. This was fixed in firmware version 0x26490700.

If you have access to a native Linux you can update it manually:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

Still I wonder why the update button isn't shown when running in the Hassio Add-on, I'll check this out so it can be fixed in next deCONZ version.

jascdk, 2019-06-17:
@manup ahhhh that sounds interesting! As for now i have not access to a Rpi with raspian - but can i use a normal linux pc?

Would be great to get fixed in next deconz version!;)

robertlandes, 2019-06-17:
@jascdk So, I migrated my setup (only for you ;) ) from my "old" raspi running Homebridge and deCONZ on raspbian stretch to my hass.io raspi using the docker container from official add-ons repo of HA and after reconfiguring my integration to point to my new deCONZ instance everything is running absolutely perfect. All devices were immediately online (including all Xiaomi devices).

So, I guess it has to be the firmware of the Conbee II you are running (mine was already up to date). We would need to investigate why the firmware is not updated through the docker container in hass.io. **If you have access to another raspi i.e. you could upgrade the firmware there and then go back and plug the Conbee II to your hass.io box. I am quite sure, that this will fix your problems.** 

@manup I can confirm, running hass.io docker container with latest firmware on Conbee II is working fine. However firmware update button is not available (or is it only shown, when an updated version is available?). 

Anyway here are the start up logs of the container:
```
[22:19:21] INFO: Start deCONZ gateway
[22:19:21] INFO: Run deCONZ OTA updater
[22:19:21] INFO: Run IKEA OTA updater
[22:19:21] INFO: Run Hass.io discovery task
{"uuid":"57162e4cdbfb46a3a74be2f3ef7e392c"}
[22:19:21] INFO: Success send discovery information to Home Assistant
[22:19:21] INFO: deCONZ is setup and running
libpng warning: iCCP: known incorrect sRGB profile
This plugin does not support propagateSizeHints()
This plugin does not support propagateSizeHints()
This plugin does not support propagateSizeHints()
22:19:22:179 HTTP Server listen on address 0.0.0.0, port: 40850, root: /usr/share/deCONZ/webapp/
22:19:22:239 CTRL. 3.16.222:19:22:322 dev /dev/ttyAMA0
22:19:22:322 COM: /dev/ttyACM0 / serialno: 
22:19:22:322 COM: --dev: /dev/ttyACM0 (ConBee II)
22:19:22:322 ZCLDB init file /data/.local/share/dresden-elektronik/deCONZ/zcldb.txt
22:19:22:587 parent process bash
22:19:22:588 gw run mode: docker/hassio
22:19:22:588 GW sd-card image version file does not exist: /data/.local/share/dresden-elektronik/deCONZ/gw-version
22:19:22:588 sd-card cid: 035344535033324780ffffffff0127d1
22:19:22:590 DB sqlite version 3.16.2
22:19:22:592 DB PRAGMA page_count: 70
22:19:22:592 DB PRAGMA page_size: 4096
22:19:22:592 DB PRAGMA freelist_count: 2
22:19:22:592 DB file size 286720 bytes, free pages 2
22:19:22:592 DB PRAGMA user_version: 6
22:19:22:592 DB cleanup
22:19:22:594 DB create temporary views
22:19:22:709 don't close database yet, keep open for 900 seconds
22:19:22:710 started websocket server at port 40860
22:19:22:714 discovery updated announce interval to 10 minutes
22:19:22:717 found node plugin: libde_rest_plugin.so - REST API Plugin
22:19:22:721 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin
22:19:24:956 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin
22:19:24:985 dev /dev/ttyAMA0
22:19:24:985 COM: /dev/ttyACM0 / serialno: 
22:19:24:985 COM: --dev: /dev/ttyACM0 (ConBee II)
22:19:25:050 dev /dev/ttyAMA0
22:19:25:050 COM: /dev/ttyACM0 / serialno: 
22:19:25:050 COM: --dev: /dev/ttyACM0 (ConBee II)
22:19:25:079 dev /dev/ttyAMA0
22:19:25:079 COM: /dev/ttyACM0 / serialno: 
22:19:25:079 COM: --dev: /dev/ttyACM0 (ConBee II)
22:19:25:127 New websocket 192.168.178.40:49373 (state: 3) 
PROTO: CRC error
22:19:25:264 Device firmware version 0x26490700
22:19:25:277 unlocked max nodes: 200
```

jascdk, 2019-06-17:
@robertlandes Ohhh thank you so much! :D It might be the solution. But i dont have access to a pi at the moment :( Maybe i will wait to the deconz gets updated:) Thank you so much for your help - it looks alot like it is the problem!

robertlandes, 2019-06-17:
@manup I've had a look inside the hass.io Docker container and as far as I can see the required firmware files are there:

```
root@core-deconz:/# ls -al /usr/share/deCONZ/firmware/
total 692
drwxr-xr-x 2 root root   4096 Jun 11 13:40 .
drwxr-xr-x 9 root root   4096 Jun 11 13:40 ..
-rw-rw-r-- 1 root root 159048 Apr 12 09:08 deCONZ_ConBeeII_0x26480700.bin.GCF
-rw-rw-r-- 1 root root 158999 Apr 22 19:44 deCONZ_ConBeeII_0x26490700.bin.GCF
-rw-r--r-- 1 root root 126202 Jul 11  2018 deCONZ_Rpi_0x261f0500.bin.GCF
-rw-r--r-- 1 root root 125386 Mar 20 11:31 deCONZ_Rpi_0x26320500.bin.GCF
-rw-rw-r-- 1 root root 125698 Apr 22 19:44 deCONZ_Rpi_0x26330500.bin.GCF
```

Also `GCFFlasher_internal` is available inside the container image. What's a little weird though is, that a list of the devices shows me a Rasbee and a Conbee II?

```
root@core-deconz:/usr/share/deCONZ/firmware# GCFFlasher_internal -l
GCFFlasher V3_05 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyAMA0     | 0x0000 | 0x0000  |            | RaspBee
/dev/ttyACM0     | 0x1CF1 | 0x0030  |            | ConBee II
```
Is this expected?

It would be possible to do a firmware update inside the docker container:

1. First kill the deCONZ process (device is otherwise locked)
`kill -s 9 34`
(where `34 `is the process id of the `deCONZ `process -> get it via `ps afx`)

2. Then change into firmware directory
`cd /usr/share/deCONZ/firmware/`

3. Then update Conbee II firmware to latest with 
`GCFFlasher_internal -d /dev/ttyACM0 -f deCONZ_ConBeeII_0x26490700.bin.GCF`

Output should be:
```
GCFFlasher V3_05 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device /dev/ttyACM0 (ConBee II)
action: update firmware after 66 ms
flashing 158985 bytes: |==============================|
verify: .
SUCCESS
Wait 10 seconds until application starts
```
4. Go to hass.io and reboot deDONZ add-on

**_And btw. shell access to the container can be obtained by installing Portainer.io add-on (be careful what you do though)_**

CC: @jascdk

manup, 2019-06-18:
>Also GCFFlasher_internal is available inside the container image. What's a little weird though is, that a list of the devices shows me a Rasbee and a Conbee II?

>root@core-deconz:/usr/share/deCONZ/firmware# GCFFlasher_internal -l
GCFFlasher V3_05 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyAMA0     | 0x0000 | 0x0000  |            | RaspBee
/dev/ttyACM0     | 0x1CF1 | 0x0030  |            | ConBee II
>Is this expected?

Yes, RaspBee is shown just based on the existence of /dev/ttyAMA0 file. It can can be read as an option to try. We thought of probing the devices but this may fail when no firmware is installed or another process is currently using the serial port.

ConBee II can be uniquely identified by it's USB product and vendor code, unfortunately this is not possible for RaspBee and ConBee I.

meizenga, 2019-06-18:
-----------------+--------+---------+------------+-------
/dev/ttyACM0 | 0x0000 | 0x0000 | | ConBee II
/dev/ttyACM1 | 0x1CF1 | 0x0030 | | ConBee II

This is how mine showed up...

pergolafabio, 2019-06-18:
i am on  2.05.65 , conbee 1 , fw 26330500
for test i have it now verry simple, all end devices are direct connected to the conbee (with extension cable)
when connected, they are all green , so should be good

after a while , i see a drop ... for test , i only have : 

xiaomi pir sensors & ikea remote switch (dimmer)
see here : 
https://www.dropbox.com/s/95xgpemaseh83qj/simplemesh.JPG?dl=0

what can i do ?

robertlandes, 2019-06-18:
@jascdk There is an updated version of the hass.io Container available for installation and it includes latest release 2.05.66. I already did the upgrade and everything is fine.

What I could see while starting up it found latest firmware for update, so maybe the FW update issue is fixed. Give it a try!

`21:17:27:483 GW update firmware found: /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x26490700.bin.GCF`

jascdk, 2019-06-18:
@robertlandes Great - thanks Robert ! I have just managed to update the Conbee FW via the raspberry pi. It seems more stable now - wuhooo :D The updated 2.05.66 version is not viewable for me right now - should I not be able to download it now through the HASSIO add-on? - it aint visible :(

robertlandes, 2019-06-18:
M
>@robertlandes Great - thanks Robert ! I have just managed to update the Conbee FW via the raspberry pi. It seems more stable now - wuhooo :D The updated 2.05.66 version is not viewable for me right now - should I not be able to download it now through the HASSIO add-on? - it aint visible :(

@jascdk Reloading the supervisor engine under Hass.io > System > Hass.io Supervisor should trigger check for updates. If that doesn't work press CTRL + F5 because sometimes the interface doesn't refresh. If still no luck, restart HA server under Configuration > General 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

eskimos2000, 2019-06-27:
I have a big problem after firmware update  conbee 1, 2.05.66 ,  to  fw 26330500. 
Upgrade went well I reckon as on the end was verified, SUCCESS. But it was not I have lost all connection to xiaomi smart plugs, sensors, switches, even hue bulb is unreachable.  I tried re-pair plugs but with no luck at all.

Log on deconz looks like that with TTL value changed. 

`16:17:04:897 GW firmware version: 0x26330500
16:17:04:897 GW firmware version is up to date: 0x26330500
16:17:05:578 discovery network reply error: Error transferring http://dresden-light.appspot.com/discover - server replied: Service Unavailable
16:17:58:629 Current channel 15
16:17:58:661 Device TTL 7080 s flags: 0x7
16:18:58:626 Current channel 15
16:18:58:658 Device TTL 7020 s flags: 0x7
16:19:58:629 Current channel 15
16:19:58:662 Device TTL 6960 s flags: 0x7
16:20:58:624 Current channel 15
16:20:58:656 Device TTL 6900 s flags: 0x7
16:21:04:931 GW firmware version: 0x26330500
16:21:04:931 GW firmware version is up to date: 0x26330500
16:21:58:626 Current channel 15
16:21:58:658 Device TTL 6840 s flags: 0x7`

Any help please, is possible to downgrade the firmware or it is not recommended?

manup, 2019-06-27:
Please try the Zigbee configuration page as described in the following link, to reload a former working configuration:

https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_59

eskimos2000, 2019-06-27:
I chacked service discover:http://dresden-light.appspot.com/discover  and I got this all the time

`Over Quota
This application is temporarily over its serving quota. Please try again later.`

eskimos2000, 2019-06-27:
I have tried it no luck at all  ;-(

eskimos2000, 2019-06-27:
No discovery star working but still no luck with connecting xiaomi plugs.

`GW firmware version: 0x26330500
17:17:04:898 GW firmware version is up to date: 0x26330500
17:17:16:491 Announced to internet
17:17:16:492 discovery server date: Thu, 27 Jun 2019 07:17:15 GMT
17:17:16:492 	 local time seems to be ok
17:17:58:630 Current channel 15
17:17:58:663 Device TTL 3480 s flags: 0x7
17:18:58:622 Current channel 15
17:18:58:654 Device TTL 3420 s flags: 0x7
17:19:58:629 Current channel 15
17:19:58:661 Device TTL 3360 s flags: 0x7
17:20:58:620 Current channel 15
17:20:58:653 Device TTL 3300 s flags: 0x7`

Did I brick my conbee stick ?

manup, 2019-06-27:
Can you please provide a screenshot what is shown on the Zigbee configuration page.
Is the firmware version 0x26330500 shown in the Phoscon App > Menu > Settings > Gateway?

>Did I brick my conbee stick ?

That's not possible, the bootloader always will allow to reflash the firmware, but based on the logs I don't think that's needed here.

eskimos2000, 2019-06-27:
[image]

Thx

manup, 2019-06-27:
Ok that means the connection between ConBee and deCONZ is working.

What about the Zigbee configuration page as described in:
https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_59

Can you please provide a screenshot here as well?

eskimos2000, 2019-06-27:
[image]
It seems that I have only one current configuration, wow, the rest disappeared, my bad. Thx for helping me @manup .

Cheers

eskimos2000, 2019-06-28:
[image]
I had deconz installed on my laptop so I just plug in it and thanks to God, It had all my old configs as well so this page on laptop looks much different. Hope that's going to help.

eskimos2000, 2019-06-28:
OK I made some progress thx to connecting conbee to my laptop and some reaserches. I loaded old configuration from id 13 and all smart plugs and hue light were able to connect to the stick, hurry. 

And I plugged stick to my odroid and .... there were no devices on phoscan app again on homeassistant logs where I am debbugging deconz there two errors:
`Error handling request
6:04 PM components/deconz/config_flow.py (ERROR)

Error doing job: Task exception was never retrieved
6:01 PM components/deconz/config_flow.py (ERROR) - message first occured at 6:01 PM and shows up 3 times`

When I click on then they are more indepth:

`Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/aiohttp/web_protocol.py", line 418, in start
    resp = await task
  File "/usr/local/lib/python3.7/site-packages/aiohttp/web_app.py", line 458, in _handle
    resp = await handler(request)
  File "/usr/local/lib/python3.7/site-packages/aiohttp/web_middlewares.py", line 119, in impl
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/real_ip.py", line 33, in real_ip_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 67, in ban_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 216, in auth_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/view.py", line 115, in handle
    result = await result
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 155, in post
    return await super().post(request, flow_id)
  File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 46, in wrapper
    result = await method(view, request, *args, **kwargs)
  File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 86, in post
    result = await self._flow_mgr.async_configure(flow_id, data)
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 87, in async_configure
    flow, cur_step['step_id'], user_input)
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 123, in _async_handle_step
    result = await getattr(flow, method)(user_input)  # type: Dict
  File "/usr/src/homeassistant/homeassistant/components/deconz/config_flow.py", line 116, in async_step_link
    session, **self.deconz_config)
TypeError: async_get_api_key() missing 1 required positional argument: 'host'`

`Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 71, in async_init
    return await self._async_handle_step(flow, flow.init_step, data)
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 123, in _async_handle_step
    result = await getattr(flow, method)(user_input)  # type: Dict
  File "/usr/src/homeassistant/homeassistant/components/deconz/config_flow.py", line 175, in async_step_ssdp
    for flow in self._async_in_progress()):
  File "/usr/src/homeassistant/homeassistant/components/deconz/config_flow.py", line 175, in <genexpr>
    for flow in self._async_in_progress()):
KeyError: 'bridgeid'`

My son who is handling arch server when home assistant with deconz is working can see on firewall logs that docker is trying heavily to get connected to some IP probably to https://dresden-light.appspot.com/discover, as this command in REST on odroid not working  bit when I connected to my laptop which does not have such a powerful firewall it is working. 

So does conbee need an internet connection to work? What to unlock in my firewall to get it works?

manup, 2019-06-28:
It looks like your Zigbee configuration parameters got scrambled multiple times (this could happen with older firmware). When you compare the top 5 entries you see that PANID and Network key have different values.

So one of these should be the correct one, likely from the version which you were running for the longest time. You can restore any of these to see which one brings the network back to life.

> My son who is handling arch server when home assistant with deconz is working can see on firewall logs that docker is trying heavily to get connected to some IP probably to [...]

>So does conbee need an internet connection to work? What to unlock in my firewall to get it works?

No, the https://dresden-light.appspot.com/discover is only used for the discover server and to query for new updates of deCONZ which are shown in the Phoscon App. Other than that it has no purpose and can safely be blocked from the firewall. The URL is also specified in the zll.db database file and can be emptied to disable discovery.

Looks like we run into some quotas of App Engine here, will be fixed soon.

eskimos2000, 2019-06-28:
OK thx, so I will wait, but in the meantime can I do something to let deconz run with HA somehow?

eskimos2000, 2019-07-02:
Is this problem with quota solved? I tried to GET https://dresden-light.appspot.com/discover and getting an error :
`503 Over Quota</H1>
This application is temporarily over its serving quota.  Please try again later.`

gdreelin, 2019-09-17:
Several times it is mentioned about an extension USB cables, just curious what is the best to use to help improve the Conbee's distance? I have tried a few from Amazon but they do not seem to detect the Conbee II on the end. I used a 3.0 3' extension and a 2.0 1' extension but no joy on detection. I was just wondering if others have had success using a certain type?  Thanks

Zebble, 2019-09-19:
Looks like "over quota" issue again...

manup, 2019-09-19:
>Looks like "over quota" issue again...

We are currently moving to a new solution without Google App Engine.

It's in beta but can already be used, deCONZ 2.05.69 will switch from https://dresden-light.appspot.com to https://phoscon.de/discover when the quota fails (currently after 10 attempts).

In 2.05.70 the new location will be enabled by default. It can also be enabled by editing the zll.db sqlite database config2 table where the discover server location is set.

ebaauw, 2019-09-19:
Does deCONZ register with the new service over IPv4, IPv6, or both?  See #41.

manup, 2019-09-20:
Currently the phoscon.de domain is IPv4 only. In time both will be supported, which also means the gateway needs to discover and announce IPv4 and IPv6 (if available) to the service, in order for clients to be able to match.

The dresden-light.appspot.com domain does already supports IPv4 and IPv6 which is problematic since the current service requires that the gateway as well as the client need to use the same type.

dartfrogdk, 2019-11-24:
Still got this issue with xiaomi sensors and switches, after a few minutes they loose connection to the Conbee ii, im using hassio and the builtin addon deconz, both the hassio and conbee has the newest firmware hassio is version  0.102.1 and conbee version 2.05.71 / 17.11.2019 firmware 26420700.
i removed and added the sensors many times, same result after anh hours or less, no connection

Smanar, 2019-11-24:
Have you tried the USB extension cable ? It can help for theses one who have connexion problem.

dartfrogdk, 2019-11-24:
I got an extension, no luck

[image: Mailtrack]
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality5&>
Sender
notified by
Mailtrack
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality5&>
24.11.19
15.24.08

alp-by, 2019-12-01:
I have the similar issue: I bought Conbee II a week ago. I have the latest FW.
Version: 2.05.69
Firmware: 264A0700

I managed to pair Xiomi Multi sensors (two sensors) without any issues, but after several hours (2-3) no update any more :( I tried with extension cable - no luck.

Any ideas how to solve it?

Flurkmark, 2019-12-02:
Since some have no issues at all, are there different fw versions for the Xaiomi?

alp-by, 2019-12-02:
hm ... makes sense to check. If there any way to check FW without xaiomi hub? maybe someone knows it?

dartfrogdk, 2019-12-02:
Use the latest beta firmware for the conbee, it worked for me

[image: Mailtrack]
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality5&>
Sender
notified by
Mailtrack
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality5&>
02.12.19
13.45.56

alp-by, 2019-12-02:
"Use the latest beta firmware for the conbee, it worked for me" - Do you mean Conbee or Conbee II. If you mean Conbee II - where did you get the latest BETA firmware for Conbee II? (Because I have the latest firmware according to the web site)

dartfrogdk, 2019-12-02:
i will find a link

[image: Mailtrack]
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality5&>
Sender
notified by
Mailtrack
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality5&>
02.12.19
17.58.31

dartfrogdk, 2019-12-02:
https://www.phoscon.de/de/conbee2/install

[image: Mailtrack]
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality5&>
Sender
notified by
Mailtrack
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality5&>
02.12.19
18.00.52

Den man. 2. dec. 2019 kl. 17.58 skrev Dartfrog <dartfrogdk@gmail.com>:

alp-by, 2019-12-02:
Thx. I updated it to 2.05.71 and will test it now.

Zebble, 2019-12-02:
If you haven't already, you may also want to add a mains-powered device (lamp, plug, repeater) to act as a zigbee router to improve your mesh/signalling.  Any of the IKEA Tradfri plugged-in devices seem to work quite well in my setup, and I'm using several Xiaomi devices (temp/humidity sensors and angle sensors).

alp-by, 2019-12-02:
I will order OSRAM Smart+ Plug (it is zigbee)

Zebble, 2019-12-03:
Sounds good.  FYI, keep in mind that not all mains-powered devices can be repeaters, but most apparently are.  I have no personal experience with OSRAM so I can't confirm if it has repeater functionality.  I've only used Tradfri bulbs and plugs and so far and all of them (as well as their USB-based repeater) ave acted as repeaters.

thowi, 2019-12-03:
For me, these worked as repeaters for Xiaomi devices:
Osram Socket, Osram Bulb Bitron Socket, Innr Socket, Ikea Tradfri Repeater.

pergolafabio, 2019-12-03:
i had osram as repeater, it worked for xiaomi, but range was not good
replaced with ikea smart plug, they work too and range much better, one disadvantage on ikea, they dont have a hardware on/off button on it , but you get a nice extra remote switch with it

alp-by, 2019-12-09:
well... osram is bought and arrived. It is a little bit a problem to get it worked in Phoscon :( (not able to add it) ... but let's check if it works as repeater

[3 more replies]

closed 2020-06-05 by Mimiix