dresden-elektronik/deconz-rest-plugin #1596: Unable to join network

dejager-mc, 2019-06-16:
Apologies if this is not the correct place for this question.

OS: Windows 10
deCONZ: 2.05.65
Conbee: updated to newest firmware at start of deCONZ, can't find it now

Problem:
deconz sees my conbee usb stick. I can change its configuration. However I can not let it "join" a network. Nothing happens when I press the button.

Goal:
I have an ikea tradfri gateway with 20 devices connected to it. I want to stop using the gateway and start using the conbee usb stick as my gateway. I am using openhab to control the gateway.

What I have tried:
- Resetting the conbee stick
- Inserting the conbee stick in a different usb port
- Changed the channel mask
- Changed the security mode to no security and "High - No master"
- Looking at the logging with the options "--dbg-error=1 --dbg-info=1 --dbg-warn=1 --dbg-aps=1 --dbg-zdp=1"

When I press the "Join" button in deconz this is the logging I get:

```
[47916] 12:09:56:260 Device protocol version: 0x010B
[47916] 12:09:56:320 Current channel 11
[47916] 12:09:56:332 CTRL got nwk update id 1
[47916] 12:09:56:337 CTRL ANT_CTRL 0x03
[47916] 12:09:56:363 Device protocol version: 0x010B
[47916] 12:09:56:427 Current channel 11
[47916] 12:09:56:439 CTRL got nwk update id 1
[47916] 12:09:56:445 CTRL ANT_CTRL 0x03
[47916] 12:09:56:475 NET ZDO start network status 0xC4
```

All the guides I have seen, like deCONZ-BHB-en.pdf, suggest that I have done all that I need to do. All the steps I have done were the same as in the guides. Everything works just like in the guides, except I can not create a network.

manup, 2019-06-17:
`0xC4` stands for NWK_STARTUP_FAILURE_STATUS, this may have various reasons.
* Do you already use an extension cable? This can prevent interference with other periphery of the pc. 
* If you can provide a screenshot of the configuration I can check if it is valid.

Normally looks like this:

[image]

dejager-mc, 2019-06-17:
Turns out it was something small. I had it plugged into a USB hub and I guess it couldn't draw enough power. I plugged it directly into a USB port of my pc and it worked immediately. Enough power to start but not enough for everything.

Thanks for the help!

closed 2019-06-17 by dejager-mc