dresden-elektronik/deconz-rest-plugin #1598: Conbee II bootloop and USB issues

Oliviakrkk, 2019-06-16:
Hi

Since I upgraded to Conbee II I have tons of issues with my network.  I did some investigation and looks like Conbee II makes USB controller  to "go mad" after some time.  

I have tons of entries like this in my dmesg:
```
[Jun16 13:36] usb 1-7: USB disconnect, device number 12
[  +0.000177] cdc_acm 1-7:1.0: failed to set dtr/rts
[  +0.311314] usb 1-7: new full-speed USB device number 13 using xhci_hcd
[  +0.149492] usb 1-7: New USB device found, idVendor=1cf1, idProduct=0030
[  +0.000002] usb 1-7: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0.000001] usb 1-7: Product: ConBee II
[  +0.000001] usb 1-7: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[  +0.000001] usb 1-7: SerialNumber: DE1962673
[  +0.000857] cdc_acm 1-7:1.0: ttyACM1: USB ACM device
[Jun16 13:37] usb 1-7: USB disconnect, device number 13
[  +0.347853] usb 1-7: new full-speed USB device number 14 using xhci_hcd
[  +0.149544] usb 1-7: New USB device found, idVendor=1cf1, idProduct=0030
[  +0.000004] usb 1-7: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0.000003] usb 1-7: Product: ConBee II
[  +0.000003] usb 1-7: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[  +0.000003] usb 1-7: SerialNumber: DE1962673
[  +0.002566] cdc_acm 1-7:1.0: ttyACM0: USB ACM device
[Jun16 13:39] perf: interrupt took too long (2510 > 2500), lowering kernel.perf_event_max_sample_rate to 79500
[Jun16 14:05] wmiir[30693]: segfault at 0 ip 00007fe1dae7d676 sp 00007ffeb4ccb728 error 4 in libc-2.24.so[7fe1dadfd000+195000]
[Jun16 14:08] usb 1-7: USB disconnect, device number 14
[  +0.000203] cdc_acm 1-7:1.0: failed to set dtr/rts
[ +21.859248] usb 1-7: new full-speed USB device number 15 using xhci_hcd
[  +0.149435] usb 1-7: New USB device found, idVendor=1cf1, idProduct=0030
[  +0.000001] usb 1-7: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0.000001] usb 1-7: Product: ConBee II
[  +0.000001] usb 1-7: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[  +0.000001] usb 1-7: SerialNumber: DE1962673
[  +0.001056] cdc_acm 1-7:1.0: ttyACM0: USB ACM device
[  +7.181029] usb 1-7: USB disconnect, device number 15
[  +0.348461] usb 1-7: new full-speed USB device number 16 using xhci_hcd
[  +0.149582] usb 1-7: New USB device found, idVendor=1cf1, idProduct=0030
[  +0.000004] usb 1-7: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0.000002] usb 1-7: Product: ConBee II
[  +0.000003] usb 1-7: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[  +0.000002] usb 1-7: SerialNumber: DE1962673
[  +0.001070] cdc_acm 1-7:1.0: ttyACM0: USB ACM device
[ +14.670456] wmiir[31150]: segfault at 0 ip 00007f3956c4b676 sp 00007ffddee2e9d8 error 4 in libc-2.24.so[7f3956bcb000+195000]
```

Also if there is any HDD connected to the same USB controller it disappears from the system with IO error after Conbee II start bootlooping. 

Any isue whet to do to fix this?

manup, 2019-06-16:
Can you please provide more details on your setup:
* Which OS distribution and version
* Do you use an USB extension cable
* Which deCONZ version
* Which firmware version is installed on ConBee II
* Any other USB devices attached

Not sure if it is related but the wmiir crashing with segfault in libc doesn't look good either.

Oliviakrkk, 2019-06-16:
**Which OS distribution and version**

Docker host: 
Ubuntu 18.04.2 LTS
I had same issues on Debian Stretch. 

**Do you use an USB extension cable**

Currently no, but I had the same issues when i used extension. 

**Which deCONZ version**
  2.05.65 (docker)

**Which firmware version is installed on ConBee II**
0x26490700

**Any other USB devices attached**
Some bluetooth dongle, pendrive for /boot, a hub with 3x USB HDDs

Oliviakrkk, 2019-06-16:
To be sure that Conbee II is the problem I switched back to Conbee I - everything else stayed the same. 
I used method from https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_59. 

Conbee II usually started bootlooping within couple hours after startup. Will see how it works with older stick.

manup, 2019-06-17:
>**Any other USB devices attached**
>Some bluetooth dongle, pendrive for /boot, a hub with 3x USB HDDs

But the ConBee II was connected directly or via USB hub?
What system is this, a Raspberry Pi?

Since versions are fine I can only guess that the devices may make trouble together on the USB ports. This can only be ruled out by testing one device at the time.

Oliviakrkk, 2019-06-17:
I have dell 3060 mff. Conbee II was tested on:
1. USB 2.0 port
2. USB 2.0 port with extension
3. USB 3.0 port
4. USB 3.0 port with extension
5. Active USB 3.0 hub connected to USB 3.0 port. 

Issue happens when there is any USB disk attached to any USB port. 

I now replaced it with Conbee I and everything works. On every port... Also I do not see the segfault anymore. 

Also have you tested Conbee II memory when getting hot? After each issue it was hot. Maybe that is the problem?

manup, 2019-06-19:
>Issue happens when there is any USB disk attached to any USB port.

Interesting is this is spinning disk or flash? I can only imagine some kind of interference might happen here.

Oliviakrkk, 2019-06-19:
Spinning. SSD is all the time in m.2 slot. Working great. 

I returned my Conbee II to Amazon. When I get refund I will order new one. Maybe those issues were connected to faulty stick...

bjorne, 2019-06-19:
I'm seeing the exact same messages with a loop of USB disconnect and reboot in my kernel log. I'm not completely sure of the timeline, but I think this started after I was tinkering with firmware flashing that failed. This wasn't initially a problem during the first few hours that the stick was connected.

I tried to recover from not seeing the device connect properly in the Phoscon app. Can't remember whether the last firmware I successfully flashed was 0x26480700 or 0x26490700, but the flashing was working very intermittently to begin with -- perhaps related to spinning USB disks?

I've tested the same stick in the current state with an Intel NUC (with spinning USB disks connected) and a Raspberry Pi 3B (without any USB devices) with equal results. I'm going to get a replacement stick as well to see whether the problems persists.

manup, 2019-06-20:
> perhaps related to spinning USB disks?
>Spinning. SSD is all the time in m.2 slot. Working great.

Just had a discussion with a colleague, at least for Raspberry Pi when you're using a external drives you need to make sure that they have either a separate power supply or are connected via an external powered USB-hub.

The Raspberry Pi alone is likely not able to deliver the required powerer for all these USB devices. We suggest using at least the 2.5A power supply to use the RaspBee, ConBee and ConBee II, depending on what else is connected to the Raspberry Pi an even higher power supply is needed.

Oliviakrkk, 2019-06-20:
In my case HDDs are connected to Unitek HUB (Unitek HUB 7xUSB 3.0 + BC ) with external power supply. Power is not the issue here. It is something else...

oscopeaaron, 2019-06-21:
I have the same issue.  Similar logs.  It happens every hour for me.  I did upgrade the firmware but do not recall ever having a failed update.  I am also running on a docker host running Debian Stretch and Kernel 4.19

manup, 2019-06-22:
Can you please provide more details of your setup:

* Do you use an USB extension cable
* Which deCONZ version
* Which firmware version is installed on ConBee II
* Any other USB devices attached

bjorne, 2019-06-22:
My external disks are also powered from separate power supply. Also, for me, this is not affecting RPi but an Intel NUC.

Anyway, I replaced my stick. The new stick has been connected and shows the hourly boot loop. Unfortunately, I cannot easily test without the USB disks as they host system partitions, but I’ll see if I can find the time.

Tried with and without extension cable. Firmware version is the default 0x26420700. Deconz version 2.05.65 via Docker on Ubuntu. For me the Phoscon app shows “Firmware: Not connected” most of the time but once in a while it shows the firmware version. The deconz GUI reflects this as “Not in network”. I am never able to discover any devices.

Here are some things I’m thinking may be of interest:

* Do we share the same USB host controller? Mine is an `USB controller: Intel Corporation Sunrise Point-LP USB 3.0 xHCI Controller (rev 21)` as reported by `lspci`
* Do we share the same chip for the disk controller? Mine uses a chipset called JMS567
* Is the `failed to set dtr/rts` message of importance?

bjorne, 2019-06-22:
For the record I’m also seeing the wmiir segfault occasionally.

Oliviakrkk, 2019-06-22:
I have Dell Optiplex 3060 with 

00:14.0 USB controller: Intel Corporation Cannon Lake PCH USB 3.1 xHCI Host Controller (rev 10)

@manup are you able to test Conbee II with some intel nuc?

oscopeaaron, 2019-06-23:
I am using an HP z210.  The USB controller is: `Intel Corporation 6 Series/C200 Series Chipset Family USB Enhanced Host Controller #2 (rev 04)`

- Do you use an USB extension cable
Yes
- Which deCONZ version
`2.05.64`
- Which firmware version is installed on ConBee II
`0x26490700`
- Any other USB devices attached
lsusb output
```
Bus 002 Device 074: ID 1cf1:0030 Dresden Elektronik 
Bus 002 Device 003: ID 0bda:0181 Realtek Semiconductor Corp. 
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 006: ID 0764:0501 Cyber Power System, Inc. CP1500 AVR UPS
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

oscopeaaron, 2019-06-24:
So I updated the docker container to a version with 2.05.66 and the issue seams to have gone away.

I get this error about every minute but it seems to work now:
```
17:02:44:633 Device TTL 2397 s flags: 0x7
17:02:54:610 unsupported device type 1 (TODO)
17:02:59:611 remove outdated neighbor 0x2932
17:03:04:611 unsupported device type 1 (TODO)
17:03:14:610 unsupported device type 1 (TODO)
17:03:24:611 unsupported device type 1 (TODO)
17:03:34:610 unsupported device type 1 (TODO)
```

Oliviakrkk, 2019-06-25:
WOW. That quite promising.

I am still waiting for my replacement Conbee II. Can't wait to see if it actually works :-).

bjorne, 2019-06-25:
For me 2.05.66 did not change anything. What changes in this release do you think make a difference?

Anyway, here is my log output from deconz:

```
deconz            | [marthoc/deconz] Starting deCONZ...
deconz            | [marthoc/deconz] Current deCONZ version: 2.05.66
deconz            | [marthoc/deconz] Web UI port: 15901
deconz            | [marthoc/deconz] Websockets port: 15902
deconz            | [marthoc/deconz] VNC Display: :10000 on port 15900
deconz            | /usr/bin/xauth:  file /root/.Xauthority does not exist                                              
deconz            |                                             
deconz            | New 'planck:10000 (root)' desktop at :10000 on machine planck   
deconz            |                                   
deconz            | Starting applications specified in /etc/X11/Xvnc-session
deconz            | Log file is /root/.vnc/planck:10000.log                                   
deconz            |                                                         
deconz            | Use xtigervncviewer -SecurityTypes VncAuth,TLSVnc -passwd /root/.vnc/passwd planck:10000 to connect to the VNC server.
deconz            |                                                         
deconz            | libpng warning: iCCP: known incorrect sRGB profile
deconz            | 19:51:12:929 HTTP Server listen on address 0.0.0.0, port: 15901, root: /usr/share/deCONZ/webapp/
deconz            | 19:51:12:939 CTRL. 3.16.219:51:12:968 COM: /dev/ttyACM0 / serialno:
deconz            | 19:51:12:968 COM: --dev: /dev/ttyACM0 (ConBee II)               
deconz            | 19:51:12:968 ZCLDB init file /root/.local/share/dresden-elektronik/deCONZ/zcldb.txt
deconz            | 19:51:13:049 parent process /bin/sh                     
deconz            | 19:51:13:049 gw run mode: docker                                          
deconz            | 19:51:13:050 GW sd-card image version file does not exist: /root/.local/share/dresden-elektronik/deCONZ/gw-version
deconz            | 19:51:13:050 DB sqlite version 3.16.2       
deconz            | 19:51:13:050 DB PRAGMA page_count: 30
deconz            | 19:51:13:051 DB PRAGMA page_size: 4096
deconz            | 19:51:13:051 DB PRAGMA freelist_count: 0    
deconz            | 19:51:13:051 DB file size 122880 bytes, free pages 0
deconz            | 19:51:13:051 DB PRAGMA user_version: 6
deconz            | 19:51:13:051 DB cleanup                       
deconz            | 19:51:13:051 DB create temporary views                                                         
deconz            | 19:51:13:053 don't close database yet, keep open for 900 seconds
deconz            | 19:51:13:054 started websocket server at port 15902
deconz            | 19:51:13:055 discovery updated announce interval to 10 minutes
deconz            | 19:51:13:056 found node plugin: libde_rest_plugin.so - REST API Plugin
deconz            | 19:51:13:057 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin
deconz            | 19:51:13:061 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin
deconz            | 19:51:13:075 COM: /dev/ttyACM0 / serialno:
deconz            | 19:51:13:075 COM: --dev: /dev/ttyACM0 (ConBee II)
deconz            | 19:51:13:125 DEV config changed event
deconz            | 19:51:13:166 Device firmware version 0x26420700
deconz            | 19:51:13:168 scan skip host .2
deconz            | 19:51:13:170 unlocked max nodes: 200
deconz            | 19:51:13:244 Device protocol version: 0x0108
deconz            | 19:51:13:255 new node - ext: 0x00212effff0407a3, nwk: 0x0000
deconz            | 19:51:13:406 Current channel 15
deconz            | 19:51:13:422 CTRL ANT_CTRL 0x03
deconz            | 19:51:13:446 Device protocol version: 0x0108
deconz            | 19:51:13:494 Current channel 15
deconz            | 19:51:13:510 CTRL ANT_CTRL 0x03
deconz            | 19:51:13:534 Device protocol version: 0x0108
deconz            | 19:51:13:582 Current channel 15
deconz            | 19:51:13:598 CTRL ANT_CTRL 0x03
deconz            | 19:51:13:622 Device protocol version: 0x0108
deconz            | 19:51:13:670 Current channel 15
deconz            | 19:51:13:686 CTRL ANT_CTRL 0x03
deconz            | 19:51:13:718 Device protocol version: 0x0108
deconz            | 19:51:13:767 Current channel 15                                                                                       deconz            | 19:51:13:783 CTRL ANT_CTRL 0x03
deconz            | 19:51:13:806 Device protocol version: 0x0108
deconz            | 19:51:13:854 Current channel 15
deconz            | 19:51:13:870 CTRL ANT_CTRL 0x03                                                                                       deconz            | 19:51:13:894 Device protocol version: 0x0108
deconz            | 19:51:13:942 Current channel 15
deconz            | 19:51:13:959 CTRL ANT_CTRL 0x03
deconz            | 19:51:13:982 Device protocol version: 0x0108                                                                          deconz            | 19:51:14:030 Current channel 15
deconz            | 19:51:14:046 CTRL ANT_CTRL 0x03
deconz            | 19:51:14:070 NET ZDO start network status 0xC4
[... 40 lines cut ...]
deconz            | 19:51:13:942 Current channel 15
deconz            | 19:51:13:959 CTRL ANT_CTRL 0x03
[... cut ...]
deconz            | 19:53:34:841 New websocket 192.168.0.24:52112 (state: 3)
deconz            | 19:53:35:706 Remove websocket 192.168.0.24:52112 after error Unknown error
deconz            | 19:53:36:773 New websocket 192.168.0.24:52113 (state: 3)
deconz            | 19:53:48:898 Device protocol version: 0x0108
deconz            | 19:53:48:948 Current channel 15
deconz            | 19:53:48:964 CTRL ANT_CTRL 0x03
```

As I haven’t run deconz successfully yet I don’t really know what to look for. What does `NET ZDO start network status 0xC4` mean?

Oliviakrkk, 2019-06-29:
Yesterday I received my new Conbee II. I moved my network onto the new stick. For now I stayed on 0x26420700. It is working and it is stable for full 24 hours.  Only Xiaomi buttons are not working at all. I think I need to upgrade firmware to make them usable.  Will see whether new firmware breaks things. 

BTW.  The segfault:
```wmiir[27559]: segfault at 0 ip 00007f00e0317676 sp 00007fff84594fe8 error 4 in libc-2.24.so[7f00e0297000+195000]```
is for 100% caused by Conbee II. I have container with VNC enabled, but deconz GUI works despite the error. So this may be a red herring.

manup, 2019-06-30:
>As I haven’t run deconz successfully yet I don’t really know what to look for. What does NET ZDO start network status 0xC4 mean?

0xC4 means the Zigbee network couldn't be started. Usually this means that the radio has a hard time 
to access the media channel due interference, which can be improved with an USB extension cable. 

However I think the main problem here might be the setup with the external USB disks. It might be interference due the spinning disk or USB power issues (the Raspberry Pi 4 seems to have improvements). We need to test and measure this in our lab to get a clearer picture.

Also it looks like you're running on old firmware, please update to 0x26490700 (but it won't help for the "disk issue"):

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

Kane610, 2019-06-30:
I'm experiencing similar issues. I had Conbee2 connected to the front of my Dell server together with a usb memory stick and a aeotec zwave stick. Got a bunch of issues resulting in unmounting the usb stick and putting it in read only mode. In some cases all other drivers would also get the same treatment. First messages would appear after 15-17 minutes taking 7-9 hours before drives getting unmounted. Moving the Conbee2 to the place where I had the old Conbee on the backside and on an extension cord seems to have mitigated at least the initial messages. Will see later today if anything will get the unmount treatment

Do you use an USB extension cable - Now yes, not when connected to the front
Which deCONZ version - 2.05.66
Which firmware version is installed on ConBee II - 26490700
Any other USB devices attached - Not on the back of the computer, on the front one usb stick and one aeotec zwave stick

Oliviakrkk, 2019-06-30:
@manup -> ok, I updated firmware to 
``` GW firmware version: 0x26490700```

Conbee II is connected to back of Dell 3060, to USB 2.0 controller. A disk in enclosure with separate power source is connected to the front to USB 3.0,  so different controller. Will see how it works in this configuration. 

Also take under consideration that most of us, with problems, aren't using Raspberry Pi.

manup, 2019-07-06:
Here is an interesting paper from Intel:

#### USB 3.0* Radio Frequency Interference on 2.4 GHz Devices

https://www.intel.com/content/www/us/en/products/docs/io/universal-serial-bus/usb3-frequency-interference-paper.html

ebaauw, 2019-07-06:
This is going to be particularly “interesting” when people start attaching USB 3.0 disks to their Raspberry Pi 4B systems, running deCONZ using a RaspBee.  Cross-referencing #1643.

manup, 2019-07-06:
I guess so. We've ordered the Raspberry Pi 4 and will do various tests with USB disks in the lab to get a get a better understanding about  interference issues and find mitigations.

Google yields familiar issues with wireless 2.4 GHz mouses and keyboards which aren't reliable when used on USB 3.0 ports or near them. Intel and Logitech recommend using USB 2.0 ports and USB extension cables which solves the issue for most people.

For RaspBee this isn't a solution, but I hope we can figure out something in the lab :)

Oliviakrkk, 2019-07-06:
Hm. New stick in USB 2.0 is stable.  This is getting interesting.

ABeumer525, 2019-07-30:
What is the status on this issue?
I am experiencing similar issues. Mine are combined with a m.2 ssd hooked up to a pi 3b+

It always worked with debian 9 but now running on 10 is showing here mentioned issues.

Oliviakrkk, 2019-08-03:
@manup 

I just got hit by that problem again. I am unable to start my deconz. Any progress ?

Oliviakrkk, 2019-08-10:
@manup maybe this helps... 

I read multiple issues. Everyone is saying that Conbee II is not working on the newest raspbian and Ubuntu. Then somebody mention it worked in older Debian. So I looked into kernel versions for all those distros. 

I installed kernel 5.0 and looks like Combee II is stable in USB 2.0. So the issue may be in the kernel 4.4 and 4.15.

ABeumer525, 2019-08-12:
@Oliviakrkk did you do this on a raspberry? If yes, can you give some pointers?

Oliviakrkk, 2019-08-12:
No, on Dell 3060 mff (amd64). 

I currently do not own raspberry.

flexy2dd, 2019-08-14:
Hello
I have the same issue with Synology NAS :(
Conbee I fully fonctionnal but my Conbee II loop 
I use extend cable

manup, 2019-08-15:
>I installed kernel 5.0 and looks like Combee II is stable in USB 2.0. So the issue may be in the kernel 4.4 and 4.15.

I don't think it is related to the kernel version, I'm running several Arch Linux boxes with recent kernel and ConBee II. For now we figure most problems due USB 3.0 periphery and USB disks/sticks connected near ConBee II.

A new firmware is cooking which highly improves device startup and non-responding ConBee II, but it won't help much for tricky setups near other USB gadgets.

>Conbee I fully fonctionnal but my Conbee II loop
I use extend cable

Do you have a recent firmware installed? Should be either 0x26490700 or 0x264A0700.

Oliviakrkk, 2019-08-15:
@manup That's great :-). Any ETA on the new firmware?

ABeumer525, 2019-08-15:
@manup what is considered to be "tricky setups near other USB gadgets"?

I run a rpi3b+ with a ssd via usb, p1 powermeter sensor, a aeotec Z-Wave and a Conbee II. Considered to be?

Adminiuga, 2019-08-15:
And does extension cable counts as "next" to devices?

Smanar, 2019-08-15:
I think the extension cable counts as good to take away conbee from perturbation.

Oliviakrkk, 2019-08-15:
Just make sure that the extension is USB 2.0

manup, 2019-08-16:
> @manup what is considered to be "tricky setups near other USB gadgets"?
> 
> I run a rpi3b+ with a ssd via usb, p1 powermeter sensor, a aeotec Z-Wave and a Conbee II. Considered to be?

According to the paper mentioned above I would guess that the SSD might be the challenge. If possible you may try to use a USB extension cable for the SSD as well, and bring the possible interference source away from the other Raspberry Pi USB ports.

But I wonder that you said it worked on Debian Stretch (9).

>It always worked with debian 9 but now running on 10 is showing here mentioned issues.

bfayers, 2019-08-17:
> But I wonder that you said it worked on Debian Stretch (9).
> 
> > It always worked with debian 9 but now running on 10 is showing here mentioned issues.

This would make sense in my case since I upgraded from proxmox 5.4 to 6.0 (which goes from debian 9 base to a debian 10 base) and I started having the issues, If i just wait a while after (lots of) resets it eventually settles down.

TPX01, 2019-08-17:
Running into similar issues on the newly arrived Conbee 2 stick. Can't even reflash the older software as it disconnects to frequently.

```
[So Aug 18 00:30:10 2019] usb 3-3: new full-speed USB device number 6 using xhci_hcd
[So Aug 18 00:30:10 2019] usb 3-3: New USB device found, idVendor=1cf1, idProduct=0030
[So Aug 18 00:30:10 2019] usb 3-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[So Aug 18 00:30:10 2019] usb 3-3: Product: ConBee II
[So Aug 18 00:30:10 2019] usb 3-3: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[So Aug 18 00:30:10 2019] usb 3-3: SerialNumber: DE1996707
[So Aug 18 00:30:10 2019] cdc_acm 3-3:1.0: ttyACM0: USB ACM device
[So Aug 18 00:30:14 2019] usb 3-3: USB disconnect, device number 6
[So Aug 18 00:30:14 2019] cdc_acm 3-3:1.0: failed to set dtr/rts
```

Do I understand it correctly that there's no solution for this issue yet? (based on discussions above..)

**Which OS distribution and version**
Ubuntu 18.04.2 LTS (clean install)
**Do you use an USB extension cable**
No
**Which deCONZ version**
2.05.65
**Which firmware version is installed on ConBee II**
0x26490700
**Any other USB devices attached**
No

Oliviakrkk, 2019-08-18:
Hi

Try to install and boot from kernel 5.0. It helped me.

TPX01, 2019-08-18:
Doesn't seem to be a kernel issue for me. Upgraded to kernel 5.0.0-25-generic but still have the issue that the stick gets disconnected.

On the other hand I tried to force it by reflashing it using retry mechanism provided by the GCFFlasher:
```
GCFFlasher_internal -d /dev/ttyACM0 -x 3 -f deCONZ_ConBeeII_0x26490700.bin.GCF -R 1000
```
It took approx 500 retries before it finally succeeded! Seems alls stable again now 👍 no more disconnects (for a start). I can highly suggest this option for people that are still stuck.

manup, 2019-08-19:
Based on your previous comment it seems that you had already installed firmware version 0x26490700?
I guess there must be another reason for the instability if you can please use a USB extension cable, preferably on a USB 2.0 port.

SimplyGardner, 2019-08-21:
I am running conbee ii on a raspberry pi 3b+ (running hassio) and it was having issues. Tried resetting and wiping everything off again... did not work. Until I heard the extension trick. I tried the extension and everything seems to be working again?

Maybe I missed something, what was the issue? Interference?

Is there a future software update? I’m running 0x26490700

TPX01, 2019-08-25:
> Based on your previous comment it seems that you had already installed firmware version 0x26490700?
> I guess there must be another reason for the instability if you can please use a USB extension cable, preferably on a USB 2.0 port.

Same issue, with or without extension cable.. :) Anyway, brute forcing seems to do the trick.

ABeumer525, 2019-08-28:
Just updated kernel version of raspbian from 4.19.58 to 4.19.66. It's christmas now, the network got worse every 15-30 minutes a restart/reboot is required.

To be brutally honest. What is/can be done to fix this? I'm on holiday now so I haven't had the chance to try the 1000 retries flashing as proposed by @TPX01 When I get home I'll give that a shot. But then again... what can be done about it?

USB extension cables as a solution doesn't feel that comfortable and whilst other usb device work I think the challenge lies with conbee / deconz?

Is this hardware or software? To me it feels like software...

ABeumer525, 2019-08-28:
> ```
> GCFFlasher_internal -d /dev/ttyACM0 -x 3 -f deCONZ_ConBeeII_0x26490700.bin.GCF -R 1000
> ```

Why don't you use the 0x264a0700.bin.GCF version?

manup, 2019-08-28:
>USB extension cables as a solution doesn't feel that comfortable and whilst other usb device work I think the challenge lies with conbee / deconz?

I'm afraid without an USB extension cable and a USB 3.0 SSD nearby you likely won't get happy. This the USB 3.0 interference on 2.4 GHz can't be fixed in software.

cgiraldo, 2019-08-28:
I am hitting a similar issue. It seems to be interference between a usb3.0 and the conbee II device.

** Host **
Pine Rock64 board
**Which OS distribution and version**
Ubuntu 18.04 LTS
**Do you use an USB extension cable**
Yes, connected to the USB 2.0 port.
**Which deCONZ version
2.05.67
Which firmware version is installed on ConBee II
0x264A0700
**Any other USB devices attached**
Seagate Firecuda through USB3.0 SATA converter.

If conbee II and usb3.0 hdd are connected to the rock64, phoscon app is not able to detect sensors. It works if If I dettach the disk.

I can even hit the problem if I attach the USB disk to another host (my laptop) and put it closed to the Rock64 board!

ABeumer525, 2019-09-01:
@TPX01 Why don't you use the 0x264a0700.bin.GCF version?

manup, 2019-09-01:
Note 0x26490700 and 0x264a0700 (newer) version are both fine. The new version has fixes which are relevant only for non deCONZ clients.

ABeumer525, 2019-09-09:
If I can be so blunt this all stinks. It keeps being wonky? What is the target date for this new firmware?

Getting nearer to the point of wanting my money back.

ABeumer525, 2019-09-14:
Has anyone got some tips on alternative sticks other than the dresden stuff?
Even thinking of moving back to zigbee2mqtt as this was way more reliable.

TPX01, 2019-09-14:
It is reliable on my part though. Upgrades of firmwares are a pain in the *** but ther than that why let that bother you? 😃

manup, 2019-09-16:
>USB extension cables as a solution doesn't feel that comfortable and whilst other usb device work I think the challenge lies with conbee / deconz?

The challenge is USB 3.0 vs. 2.4 GHz devices. See also https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1598#issuecomment-508904699

To describe it differently USB 3.0 is a screaming elephant while 2.4 GHz next to it have the voice of a mice. As I noted earlier:

>I'm afraid without an USB extension cable and a USB 3.0 SSD nearby you likely won't get happy. This the USB 3.0 interference on 2.4 GHz can't be fixed in software.

So no matter what Zigbee USB-dongle, Bluetooth or WiFi device you use nearby USB 3.0 devices, the signal quality will drastically drop.

While the new firmware which is scheduled during the next weeks will solve many issues of reliable starting and bootloops, it can't fix radio interference of USB 3.0 vs. 2.4 GHz.

ABeumer525, 2019-09-21:
Finally it all works. Reinstalled deconz, reflashed the stick, reset the settings on the pi. What did the trick I don't know. With the zha zigpy solution it still kept being wonky. So for now I will keep it with deCONZ.

deCONZ 2.0.5.67
Firmware 264A0700
Raspbian Buster

Even removed the extension cable. For me this was not an issue.

jossie67, 2019-09-24:
I have the same issue on my NUC NUC6CAYH

Version 2.05.69 / 9/6/2019
Firmware 26A0700
Intel NUC NUC6CAYH
USB extension cable:  yes
USB 3.1
Ubuntu 18.04.3 LTS

@manup , can I set the new firmware?

andras-tim, 2019-10-09:
I have same issue:

```
[okt 9 03:09] usb 1-1.1.2: USB disconnect, device number 74
[  +0,234434] usb 1-1.1.2: new full-speed USB device number 75 using xhci_hcd
[  +0,102348] usb 1-1.1.2: New USB device found, idVendor=1cf1, idProduct=0030
[  +0,000005] usb 1-1.1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0,000003] usb 1-1.1.2: Product: ConBee II
[  +0,000002] usb 1-1.1.2: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[  +0,000002] usb 1-1.1.2: SerialNumber: DE1997960
[  +0,001563] cdc_acm 1-1.1.2:1.0: ttyACM1: USB ACM device
[  +3,501625] usb 1-1.1.2: USB disconnect, device number 75
[  +0,227104] usb 1-1.1.2: new full-speed USB device number 76 using xhci_hcd
[  +0,103441] usb 1-1.1.2: New USB device found, idVendor=1cf1, idProduct=0030
[  +0,000005] usb 1-1.1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0,000002] usb 1-1.1.2: Product: ConBee II
[  +0,000003] usb 1-1.1.2: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[  +0,000003] usb 1-1.1.2: SerialNumber: DE1997960
[  +0,001704] cdc_acm 1-1.1.2:1.0: ttyACM1: USB ACM device
```

Docker: `marthoc/deconz:amd64-2.05.69`
Product: **ConBee II**
Version: **2.05.69 / 9/6/2019**
Firmware: **264A0700**
USB: **3.0 Host + 3.0 HUB**
USB extension cable: **yes**

[28 more replies]

closed 2020-06-06 by Mimiix