dresden-elektronik/deconz-rest-plugin #1599: Conbee II Doesn't show up

Cabalforces, 2019-06-16:
The Conbee 2 device i have ordered doesn't show up in either Win 10, Hassio or Raspbian either.

manup, 2019-06-17:
Hmm strange, can you please provide the output of following command on a Linux machine:

    lsusb

Shurov, 2019-06-18:
I have the same problem. My output is:
`~$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 003: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 002: ID 0781:5583 SanDisk Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`

Cabalforces, 2019-06-18:
pi@phoscon:~ $ lsusb
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Shurov, 2019-06-18:
Looks like `/dev/usb` is a proper address that should be used for me

olivertheobald, 2019-06-18:
Can you please provide the output of dmesg. Are you on the latest docker Image?

eggman0131, 2019-06-22:
I think I may have the same issue, so I'll post here rather than raising a new one.

I'm using ubuntu 18.04 and have a conbee II.  I will be using marthoc's docker image and so haven't installed deconz.  In the host O/S the stick is showing up when I run lsusb, I can see entries in dmesg, but there's nothing in /dev  (assuming it should be /dev/ttyACM0 or /dev/ttyUSB0) - I have no other USB devices.

**lsusb output**
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 002: ID 1cf1:0030 Dresden Elektronik
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

**dmesg output**
[    9.464250] usb 3-1: new full-speed USB device number 2 using uhci_hcd
[    9.922364] usb 3-1: New USB device found, idVendor=1cf1, idProduct=0030
[    9.922926] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    9.923596] usb 3-1: Product: ConBee II
[    9.923941] usb 3-1: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[    9.924588] usb 3-1: SerialNumber: DE1963697

I've been poking round the internet and found references to udev or installing FTDI drivers but I'm pretty sure this should just work so am holding off doing anything that might make things worse :-)

**Full disclosure:**
The conbee ii is plugged in to an intel NUC running proxmox and passed through to the Ubuntu VM.  If I don't pass through, proxmox sees it as /dev/tty/ACM0.  It _may_ be an issue with pass-through to the VM (not an LCX container) but then I wouldn't expect to see it listed in lsusb ...

manup, 2019-06-22:
Hmm the lsbusb and dmesg output looks fine. Not sure why the /dev/ttyACM* devices aren't visible.

FTDI or any other driver isn't needed since ConBee II has native USB.

You may check if Gnome Modem Manager is running and doing some strange things I've seen that in the past but haven't investigated in detail yet.

eggman0131, 2019-06-22:
I've had a look, I don't think I'm running it.

```
eggman@has01:~$ ps -ef | grep modem
eggman    2317  2131  0 07:57 pts/0    00:00:00 grep --color=auto modem
eggman@has01:~$ mmcli

Command 'mmcli' not found, but can be installed with:

sudo snap install modem-manager  # version 1.8.0-8, or
sudo apt  install modemmanager

See 'snap info modem-manager' for additional versions.
```

manup, 2019-06-22:
>I've had a look, I don't think I'm running it.

Sorry I should clarify that, the modem manager isn't required, I was just asking to see if it is running and might make trouble.

Can you please try the following command to see system log, and then unplug, wait a few seconds  and replug ConBee II again.

    sudo journald -xb -f

eggman0131, 2019-06-22:
OK, (thanks for the help btw).

```
Jun 22 08:13:16 TestConbee sudo[1746]:   eggman : TTY=pts/0 ; PWD=/home/eggman ; USER=root ; COMMAND=/bin/journalctl -xb -f
Jun 22 08:13:16 TestConbee sudo[1746]: pam_unix(sudo:session): session opened for user root by eggman(uid=0)
Jun 22 08:13:38 TestConbee kernel: usb 3-1: USB disconnect, device number 2
Jun 22 08:13:57 TestConbee kernel: usb 3-1: new full-speed USB device number 3 using uhci_hcd
Jun 22 08:13:58 TestConbee kernel: usb 3-1: New USB device found, idVendor=1cf1, idProduct=0030
Jun 22 08:13:58 TestConbee kernel: usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Jun 22 08:13:58 TestConbee kernel: usb 3-1: Product: ConBee II
Jun 22 08:13:58 TestConbee kernel: usb 3-1: Manufacturer: dresden elektronik ingenieurtechnik GmbH
Jun 22 08:13:58 TestConbee kernel: usb 3-1: SerialNumber: DE1963697
Jun 22 08:14:04 TestConbee kernel: usb 3-1: USB disconnect, device number 3
Jun 22 08:14:21 TestConbee kernel: usb 3-1: new full-speed USB device number 4 using uhci_hcd
Jun 22 08:14:22 TestConbee kernel: usb 3-1: New USB device found, idVendor=1cf1, idProduct=0030
Jun 22 08:14:22 TestConbee kernel: usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Jun 22 08:14:22 TestConbee kernel: usb 3-1: Product: ConBee II
Jun 22 08:14:22 TestConbee kernel: usb 3-1: Manufacturer: dresden elektronik ingenieurtechnik GmbH
Jun 22 08:14:22 TestConbee kernel: usb 3-1: SerialNumber: DE1963697
```

eggman0131, 2019-06-22:
I only took it out once, it did the 2nd USB disconnect by itself.  

That was a couple of minutes ago and there's not been another.

manup, 2019-06-22:
The second disconnect is actually expected, the current bootloader will:
* Attach the usb interface
* Wait up to 10 seconds for firmware update instructions
* Detach the usb interface
* Start the deCONZ firmware (application code) which does attach the usb interface again

We're currently working on an improved bootloader where the firmware, if present, is started directly. The usb device will therefore attached only once in normal operation. This mitigates a few issues we've seen in the past. But I guess in your case it isn't related.

Shurov, 2019-06-22:
I have no new messages under `dmesg` upon connecting/disconnecting the stick (I've used all USB ports on my server, hub and no-hub). 
Is it broken? Shall I return it back to seller?

5mauggy, 2019-09-05:
Hi,
I am facing the same (or similar) issue. deCONZ can't see my ConBee II after a server reboot (or a ConBee reset or unplug/re-plug the ConBee). The only fix I found so far is to run `udevadm trigger` and then the ConBee II shows up as expected. I am using the ConBee II on an Ubuntu 18.04 server. There are two other USB devices plugged in (a Sony PlayTV and a dongle for a Microsoft Wireless All-in-One Media Keyboard). I also tried without these 2 devices and it didn't change anything. Motherboard is a Gigabyte Z97N-Gaming.

```console
$ lsusb 
Bus 002 Device 002: ID 8087:8001 Intel Corp. 
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:8009 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 023: ID 1cf1:0030 Dresden Elektronik 
Bus 003 Device 003: ID 045e:0800 Microsoft Corp. 
Bus 003 Device 002: ID 1415:0003 Nam Tai E&E Products Ltd. or OmniVision Technologies, Inc. 
Bus 003 Device 008: ID 8087:07dc Intel Corp. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

```console
$ GCFFlasher_internal -l
GCFFlasher V3_06 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
```

```console
$  ls -l /dev/ttyACM*
crw-rw---- 1 root dialout 166, 0 sept.  5 18:43 /dev/ttyACM0
```

```console
$ sudo udevadm trigger
$ sudo GCFFlasher_internal -l
GCFFlasher V3_06 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyACM0     | 0x1CF1 | 0x0030  | DE1234567  | ConBee II 
```

```console
$ dmesg
[52879.427964] usb 3-9: USB disconnect, device number 116
[52897.626909] usb 3-9: new full-speed USB device number 118 using xhci_hcd
[52897.776488] usb 3-9: New USB device found, idVendor=1cf1, idProduct=0030
[52897.776497] usb 3-9: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[52897.776502] usb 3-9: Product: ConBee II
[52897.776506] usb 3-9: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[52897.776510] usb 3-9: SerialNumber: DE1234567
[52897.777331] cdc_acm 3-9:1.0: ttyACM0: USB ACM device
```

<details><summary>ConBee firmware update output (click to expand)</summary><p>

```console
$ sudo GCFFlasher_internal -d /dev/ttyACM0 -f /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x264a0700.bin.GCF -x 3
GCFFlasher V3_06 (c) dresden elektronik ingenieurtechnik gmbh
using firmware file: /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x264a0700.bin.GCF 
ls dev: /dev/ttyACM0 (0x1CF1/0x0030) sn: DE1234567
Reboot device /dev/ttyACM0 (ConBee II)
serial port open device
deCONZ firmware version 26490700
set watchdog ttl status: 0x00
close serial port
serial port open device
bootloader write CMD_GetID after 3200 ms
bootloader start after 3208 ms
GCF_ResetDeviceDone
action: update firmware after 6735 ms
flashing 158985 bytes: |=GET 00 00 3B
transmitting page 1/621
GET 01 00 3B
[... 134 lines cut ...]
sept. 05 19:54:21 server upowerd[4345]: treating change event as add on /sys/devices/pci0000:00/0000:00:1b.0/sound/card1/input8/event8
sept. 05 19:54:21 server upowerd[4345]: treating change event as add on /sys/devices/pci0000:00/0000:00:1b.0/sound/card1/input12/event12
sept. 05 19:54:21 server upowerd[4345]: treating change event as add on /sys/devices/pci0000:00/0000:00:1b.0/sound/card1/input9/event9
sept. 05 19:54:21 server upowerd[4345]: treating change event as add on /sys/devices/pci0000:00/0000:00:1b.0/sound/card1/input13/event13
sept. 05 19:54:21 server upowerd[4345]: treating change event as add on /sys/devices/pci0000:00/0000:00:1b.0/sound/card1/input11/event11
sept. 05 19:54:21 server upowerd[4345]: treating change event as add on /sys/devices/pci0000:00/0000:00:1b.0/sound/card1/input14/event14
sept. 05 19:54:21 server systemd-udevd[17840]: Process '/sbin/crda' failed with exit code 234.
sept. 05 19:54:21 server upowerd[4345]: treating change event as add on /sys/devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.1/0003:045E:0800.0002/input/input4/event4
sept. 05 19:54:21 server upowerd[4345]: treating change event as add on /sys/devices/virtual/input/mice
sept. 05 19:54:21 server systemd-logind[2428]: Watching system buttons on /dev/input/event20 (Sony PlayTV)
sept. 05 19:54:21 server systemd-logind[2428]: Watching system buttons on /dev/input/event3 (Microsoft MicrosoftÂ® Nano Transceiver v2.0)
sept. 05 19:54:21 server systemd-logind[2428]: Watching system buttons on /dev/input/event5 (Microsoft MicrosoftÂ® Nano Transceiver v2.0)
sept. 05 19:54:21 server gnome-shell[7995]: g_array_unref: assertion 'array' failed
sept. 05 19:54:22 server gnome-shell[7995]: g_array_unref: assertion 'array' failed
sept. 05 19:54:22 server gnome-shell[7995]: g_array_unref: assertion 'array' failed
sept. 05 19:54:22 server gnome-shell[7995]: g_array_unref: assertion 'array' failed
sept. 05 19:54:22 server gnome-shell[7995]: g_array_unref: assertion 'array' failed
sept. 05 19:54:22 server gnome-shell[7995]: g_array_unref: assertion 'array' failed
sept. 05 19:54:22 server gnome-shell[7995]: g_array_unref: assertion 'array' failed
sept. 05 19:54:22 server gnome-shell[7995]: g_array_unref: assertion 'array' failed
sept. 05 19:54:25 server sudo[18080]: linuxuser : TTY=pts/0 ; PWD=/home/linuxuser ; USER=root ; COMMAND=/usr/bin/GCFFlasher_internal -l
sept. 05 19:54:25 server sudo[18080]: pam_unix(sudo:session): session opened for user root by (uid=0)
sept. 05 19:54:25 server sudo[18080]: pam_unix(sudo:session): session closed for user root
```
</p></summary>

Thiesy65719, 2019-09-26:
I have a similar problem, my phoscon-app does not show the ConBee II (firmware) connected. Everything seems normal (dmesg, /dev/ttyACMx), 

I added the '--dev=/dev/serial/by..' option to deconz. No help.

I am running deconz on a raspi 3B raspbian buster headless. Attached via USB in addition to the ConBeeII (FW 264A0700, out of the box) are a USB-ethernet-adapter and two USB-sticks (sda, sdb).
From reading the above it appeared that usb dicovery initialization seems an issue.

Added nofail option in fstab to allow unplugging without boot hangup.
After unplugging all USB-devices except ConBeeII resulted in the successful connection indicated in phoscon-app as desired.

Added the USB ethernet adapter and it failed.

Removed the adapter again and added the two USB and deconz/phoscon runs as desired
Hotplugged the USB ethernet adapter, restarted networking and still everything runs as desired.

Below is the syslog (filtered for 'usb'-entries)
showing the USB-ethernet-adapter being plugged after ConBeeII is up and running.

The ethernet-adapter is interfering with the ConBee-dongle during boot. Plugging the ethernet-adapter once the system is running is a possible workaround.

[dmesg.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/3657442/dmesg.txt)

Sorry, I am new to github and don't know how to properly attach the lines from syslog.

Thiesy65719, 2019-09-27:
If have tried to find what action causes deconz to loose the connection to the ConBee II:
First again my setup

Hardware:           Raspberry Pi 3 Model B Rev 1.2
OS:                     Raspian Buster lite, headless
                           Linux sauna 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
hostname:          sauna
deconz:               2.05.67-raspbian-buster-stable

USB-Devices:      USB-Stick,      USB-Stick, Edimax USB Ethernet-Adapter (10/100M), ConBee II
ConBee II FW:    0x264a0700

ZigBee Devices:   3 GL-C-008S       1 Philips Dimmschwitch

I used phoscon-App to configure the switch and lights so as to switch the lights on/off with the respective button. This gives me a full test. I was in error when I assumed that deconz communicates with ConBee II if the App shows the FW-Version.

I am suspection the USB-ethernet-adapter  to interfere with deconz/ConBee II:

I did the following test sequenz:
- shutdown
- unplug ethernet adapter
- boot
- Dimmswitch controls the lights successfully
- 'ip addr' does not show an eth1
- unplug Ethernetcable from adapter
- plug adapter to USB
- switch controls lights successfully
- 'ip addr' shows eth1 is down
- 'sudo systemctl restart networking'
- switch controls lights successfully
- 'ip addr' shows eth1 is down as expected with no cable plugged
- plug the cable
- switch **fails** to control lights
- 'ip addr' shows eth1 is up

Next test:
- everything plugged as I need it
- boot
- light control **fails**
- unplug Ethernet cable but leave adapter plugged
- light control **works**
- plug Ethernet cable
- light control **fails again**

As far as I can tell dmesg and syslog show nothing out of the ordinary.
I have reached the limits of my skills. Let me know which logs I should send to allow you to troubleshoot.

closed 2020-02-01 by stale[bot]

mikkelzzz3, 2021-02-23:
I am also having the exact same problem on Ubuntu server 18.04.after every reboot o have to manually unplug and plug the USB back in.

I tried unbind the USB bus and binding the bus but it does not work.

Thiesy65719, 2021-02-24:
I finally resolved the problem by replacing the USB-Ethernet Adapter and the Raspberry. It was my impression that there was a hardware problem in the USB system. Pinpointing the exact cause is beyond me. Possibly the Ethernet adapter interfered with ConBee II, or the USB interface on Rpi failed. The Rpi runs 24/7 and was giving me increasing trouble with WiFi. The Rpi had run about 3 years.

For trouble shooting check whether your system finds the ConBee II with all other USB devices removed. Should this cure the problem continue and remove on at a time. If you are able to identify a device that collides with ConBee II replace that device, if possible try a different brand.

Hope this helps

mikkelzzz3, 2021-02-24:
@Thiesy65719 I managed to find a solution as well this morning. I made a script that cron runs on every reboot:

`#!/bin/bash
/bin/echo usb3 > /sys/bus/usb/drivers/usb/unbind
sleep 1
/bin/echo usb3 > /sys/bus/usb/drivers/usb/bind
sleep 45
# Turns on my hassos kvm addon, that has the conbee statically added# /usr/bin/virsh start hassos`

This turns off the usb bus and then back on. Simulating physically removing the conbee and plugging it back in. 

Ofcourse one would need to adjust the script to their own system. My conbee is attached on bus 3 = usb3. The bus can be identified with "lsusb -t" when the conbee is successfully working on the system. (Or any other usb drive that sits in the usb port you plan to put the conbee in)