dresden-elektronik/deconz-rest-plugin #1600: headless Image - password for Phoscon-Gateaway-FFCF

flex-0, 2019-06-16:
Hey guys,

so I installed the headless version of the images. I get that it runs in AP mode and I can change that within the app. 

But how can I access the Phoscon-Gateaway-FFCF?

I can't find a password and "raspbeegw" is not working so could you tell me how to access the AP?

manup, 2019-06-17:
Normally on a fresh installation you're asked to specify a new password for the Phoscon App.

https://phoscon.de/en/app/doc#commissioning-login-password-en

Sennevds, 2019-09-13:
@manup he's talking about the sd-image. If you boot the first time it creates a access point but there isn't any information about the password of the wifi.
The ssid of the wifi is something like 'Phoscon-Gateway-[random 4 letters]'.
It's nice that it creates a access point to configure the the app but we need a password!

closed 2020-01-19 by stale[bot]

jsemmler, 2020-03-13:
same problem