dresden-elektronik/deconz-rest-plugin #1601: Centralite Pearl Touch Screen Thermostat - 3157100-WE

rb2k, 2019-06-17:
These are the logs around the time that I added it:

```
17:07:22:069 device announce 0x000D6F00055B811A (0x499A) mac capabilities 0x80
17:07:22:069 set fast probe address to 0x000D6F00055B811A (0x499A)
17:07:22:069 FP indication 0x0000 / 0x0013 (0x000D6F00055B811A / 0x499A)
17:07:22:069                       ...     (0x000D6F00055B811A / 0x499A)
17:07:22:069 device announce 0x000D6F00055B811A (0x499A) mac capabilities 0x80
17:07:22:947 [1] get node descriptor for 0x000d6f00055b811a
17:07:36:438 FP indication 0x0000 / 0x8002 (0x000D6F00055B811A / 0x499A)
17:07:36:438                       ...     (0x000D6F00055B811A / 0x499A)
17:07:36:438 ZDP indication search sensors 0x000D6F00055B811A (0x499A) cluster 0x8002
17:07:36:438 ZDP indication search sensors 0x000D6F00055B811A (0x499A) clear timeout on cluster 0x8002
17:07:36:444 [2] get active endpoints for 0x000d6f00055b811a
17:07:36:675 FP indication 0x0000 / 0x8005 (0x000D6F00055B811A / 0x499A)
17:07:36:675                       ...     (0x000D6F00055B811A / 0x499A)
17:07:36:675 ZDP indication search sensors 0x000D6F00055B811A (0x499A) cluster 0x8005
17:07:36:675 ZDP indication search sensors 0x000D6F00055B811A (0x499A) clear timeout on cluster 0x8005
17:07:39:299 FP indication 0x0104 / 0x000A (0x000D6F00055B811A / 0x499A)
17:07:39:299                       ...     (0x000D6F00055B811A / 0x499A)
17:07:44:370 ZCL attribute report 0xF0D1B80000019542 for cluster 0x0300, ep 0x01
17:07:44:411 ZCL attribute report 0xF0D1B80000019542 for cluster 0x0300, ep 0x01
17:08:04:306 Current channel 25
17:08:04:338 Device TTL 4920 s flags: 0x7
17:08:08:371 verified group capacity: 255 and group count: 1 of LightNode 0x000d6ffffe28349c
17:08:08:371 0x000d6ffffe28349c found group 0x0006
17:08:17:491 CTRL skip polling while permit join is set
17:08:20:353 Search sensors done
17:08:25:965 Websocket disconnected 192.168.1.82:60417 (state: 0) 
17:08:28:225 New websocket 192.168.1.82:60468 (state: 3) 
17:08:29:015 Websocket disconnected 192.168.1.82:60468 (state: 0) 
17:08:30:076 New websocket 192.168.1.82:60469 (state: 3) 
17:08:31:020 Websocket disconnected 192.168.1.82:60469 (state: 0) 
17:08:32:149 New websocket 192.168.1.82:60472 (state: 3) 
17:08:42:747 binding for cluster 0x0000 of 0x000D6FFFFE28349C exists (verified by reporting)
17:08:42:747 binding for cluster 0x0006 of 0x000D6FFFFE28349C exists (verified by reporting)
17:08:42:747 binding for cluster 0x0008 of 0x000D6FFFFE28349C exists (verified by reporting)
17:08:42:900 ZCL configure reporting rsp seq: 189 0x000D6FFFFE28349C for cluster 0x0006 attr 0x0000 status 0x00
17:08:42:964 ZCL configure reporting rsp seq: 190 0x000D6FFFFE28349C for cluster 0x0008 attr 0x0000 status 0x00
17:09:02:481 Unhandled node key 16777249
17:09:04:309 Current channel 25
17:09:04:341 Device TTL 4860 s flags: 0x7
17:09:05:529 Unhandled node key 16777249
17:09:07:846 Unhandled node key 16777251
17:09:07:939 Unhandled node key 16777248
17:09:08:036 Unhandled node key 36
17:09:09:702 Unhandled node key 16777251
17:09:09:723 Unhandled node key 16777248
17:09:09:834 Unhandled node key 36
17:09:11:919 Unhandled node key 16777248
17:09:11:920 Unhandled node key 16777251
17:09:12:095 Unhandled node key 36
17:09:17:782 CTRL skip polling while permit join is set
```

These are the requested screenshots:

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

wvuyk, 2019-06-21:
This one would be really nice if supported.......

rb2k, 2019-06-24:
I wonder if there's anything else I have to do?
I tried to follow https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support and did everything besides the debug logging. I think this should be enough?

manup, 2019-06-25:
Thanks, I think this should be enough data to get a first implementation ready.

rb2k, 2019-07-22:
Is there any ETA on these kind of things?
I'm thinking about getting rid of my Nest thermostat over the next month and was wondering if I should just wait it out or switch to something already supported :)

closed 2019-11-27 by stale[bot]