dresden-elektronik/deconz-rest-plugin #1603: Support request for Paulmann 500.67 Zigbee Remote

a005, 2019-06-18:
Is it possible to support Paulmann 500.67 Zigbee Remote Controller?
Product page is [here](https://en.paulmann.com/indoor-lighting/smart-home/zigbee/controlling/smarthome-zigbee-remote-control-gent-with-wall-mounting-bracket/50067).
Some comments about connection to Philips Hue Bridge were found [here (customer questions and answers)](https://www.amazon.de/Paulmann-500-67-Smarthome-Fernbedienung-Wandhalterung/dp/B071WZZ12Y).
Node info:
[image]

Basic cluster info:
[image]

If needed, I'm ready to provide additional information.
Thank you.

closed 2019-10-23 by stale[bot]

kayhannay, 2020-01-23:
I would also like to see support for this controller. Is there something I can provide to support the integration?

ITSSeu, 2020-03-11:
> 
> 
> I would also like to see support for this controller. Is there something I can provide to support the integration?

I also like to see support for this controller, thanks!

svrnwnsch, 2020-05-23:
Pleas reopen the Issue as I would also love support for this controller.