dresden-elektronik/deconz-rest-plugin #1605: Adding devices to Home Assistant after initial configuration

hvddrift, 2019-06-18:
Hi

I did my initial setup of all my devices on a dedicated Raspberry PI. Then I synced/configured all the devices via the authentication section of the gateway. 

All is working great. 

I just bought another zigbee switch that I have added. Is there a method to make it visible to home Assistant without the fully sync method I did above?

robertlandes, 2019-06-18:
Resources of type light should be discovered automatically. For other types you would need to do one of two things: either restart the HA service from Configuration > General or call the corresponding service from developer tools.

[image]

[image]

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

robertlandes, 2019-06-18:
Ah, sorry I just read your post again and switches are not visible as separate instances like sensors or lights. You will probably only find a batter level sensor for your switch. Switches are only presented by HA as events of type `deconz_event`. To use switches in automations or node-red you have to use the events as triggers. Please have a look at section `Remote Control Devices` here: https://www.home-assistant.io/components/deconz/ 

To be clear light switches and/or remotes are handled as remote control devices. _Switches_ in HA are actually only smart plugs like Innr SP 120 or Osram Smart+ plugs (See at the very bottom of that website)

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

closed 2019-10-24 by stale[bot]

oywino, 2019-12-21:
Friends, I have ConBee II connected to a Win10 PC, running fine and tested with a single IKEA bulb. Firmware and SW updated to latest versions.
Next step is to get it integrated to my Home Assistant, but I do not understand how to do that. I know how to use the deCONZ to initiate App Authentication, but how do I get HA to find it?
The integration description talks about something called "the deCONZ configurator in Home Assistant frontend" - but what is that and where do I find this configurator?

robertlandes, 2019-12-22:
@oywino So if i understand correctly:

You want to integrate your deCONZ setup running on a Windows PC into HA, right?
1. Make sure that your Windows Firewall isn't blocking the ports of the REST-API
2. Go to Your HA install and open Settings -> Integration and click on the + button
3. Search for deCONZ Zigbee Gateway and add it
[image]
4. Enter IP address and port of your install (port is REST-API port)
5. Do the authentication thing through deCONZ web interface
6. That's it

As a side note: If you are running HA via Hass.io on let's say a Raspbery PI, I highly recommend installing the deCONZ add-on from HA and plug your Conbee II into the PI.

And usually HA should actually auto discover running deCONZ instances inside you network.

And further I think this is not the right place to open and issue and ask for support as this is more an Home Assistant issue.

Good resources to get support are
- HA community
https://community.home-assistant.io/c/configuration/zigbee
- HA Github
https://github.com/home-assistant/home-assistant
- HA Documentation
https://www.home-assistant.io/integrations/deconz/

And here is the community supported container add-on:
- https://github.com/marthoc/docker-deconz

oywino, 2019-12-22:
Many thanks for your detailed reply. I really appreciate that.
Please help me one step further. Here are some details:
I run deCONZ on a Windows 10 machine. Phoscon runs fine and my ZigBee devices ara all visible and controllable.
When use the method you advice above to establish the integration from HA, I persistently get "Couldn't get an API key"
In your bullet #4 you say "Enter the IP and Port".  Where am I supposed to enter that? And how do I determine the which IP and port deCONZ has assigned to its REST-API ?