dresden-elektronik/deconz-rest-plugin #1607: Konke motion sensor

sweebee, 2019-06-19:
I bought this zigbee sensor off aliexpress. It looks nice but it doesnt work yet with deconz.

https://nl.aliexpress.com/item/33014773963.html?spm=a2g0s.9042311.0.0.5dff4c4d8Csnuz

What to do to get this sensor supported? I tried to pair it as a sensor but nothing happens.

manup, 2019-06-19:
We need to add it to the REST-API first. Can you please provide the information as described here:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support

sweebee, 2019-06-19:
I have raspbee but no interface on my rasbian. does the debug info give enough info?

```deCONZ --dbg-info=1 --dbg-aps=1 --dbg-zcl=1 --dbg-zdp=1 --http-port=80```

sweebee, 2019-06-20:
I installed the desktop environment, but when i start the deconz app nothing shows up.

manup, 2019-06-20:
This gives us the manufacturer code. We'd also need the attributes from the basic cluster like model identifier and the endpoint clusters as shown in the above screenshots.

sweebee, 2019-06-20:
[image]

[image]

BigWumpus, 2019-06-30:
here are some other data
[image]
[image]
[image]
[image]
[image]
[image]

sweebee, 2019-07-08:
@manup Do you need any more information?

goermezer, 2019-07-16:
Any progress with this sensor? I bought one too in hope to get a motion stop status. Want to replace Xiaomi motion sensors.

vandalon, 2019-07-18:
I just bought the same little sensor. I'm happy to test where wanted :)

phdelodder, 2019-07-27:
Any timeline on the integration of this device? Just bought one.

Smanar, 2019-07-27:
Lol, I have just discovered this devices, there is same device than Xiaomi, but cheaper.

jurriaan, 2019-07-31:
@manup I've got the same device, also got their temperature and humidity sensor and contact sensor.

Both the contact sensor and motion sensor work with IAS.

Got this log of the motion sensor when it detects motion:
```
18:43:06:268 APS-DATA.indication srcAddr: 0xfaf4, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x0000, lqi: 255, rssi: -43
18:43:06:269 APS-DATA.request id: 218, addrmode: 0x02, addr: 0xfaf4, profile: 0x0000, cluster: 0x8000, ep: 0x00 -> 0x00 queue: 1 len: 12 tx.options 0x04
18:43:31:031 APS-DATA.indication srcAddr: 0xfaf4, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x0000, lqi: 255, rssi: -48
18:43:31:031 APS-DATA.request id: 151, addrmode: 0x02, addr: 0xfaf4, profile: 0x0000, cluster: 0x8000, ep: 0x00 -> 0x00 queue: 1 len: 12 tx.options 0x04
```

And contact sensor (while opening and closing the contact):
```
18:45:32:616 APS-DATA.indication srcAddr: 0x201e, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x0000, lqi: 255, rssi: -44
18:45:35:808 APS-DATA.indication srcAddr: 0x201e, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x0000, lqi: 255, rssi: -49
18:45:37:637 APS-DATA.indication srcAddr: 0x201e, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x0000, lqi: 255, rssi: -43
18:45:37:641 APS-DATA.indication srcAddr: 0x201e, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x0000, lqi: 255, rssi: -44
```

For temperature/humidity I got these logs:
```
18:12:33:298 APS-DATA.indication srcAddr: 0xc3ca, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0405, lqi: 255, rssi: -52
18:12:33:298 APS-DATA.request id: 171, addrmode: 0x02, addr: 0xc3ca, profile: 0x0104, cluster: 0x0405, ep: 0x01 -> 0x01 queue: 1 len: 5 tx.options 0x00
18:17:15:472 APS-DATA.indication srcAddr: 0xc3ca, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0402, lqi: 255, rssi: -51
18:17:15:473 APS-DATA.request id: 11, addrmode: 0x02, addr: 0xc3ca, profile: 0x0104, cluster: 0x0402, ep: 0x01 -> 0x01 queue: 1 len: 5 tx.options 0x00
18:36:58:098 APS-DATA.indication srcAddr: 0xc3ca, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0402, lqi: 255, rssi: -52
18:36:58:098 APS-DATA.request id: 63, addrmode: 0x02, addr: 0xc3ca, profile: 0x0104, cluster: 0x0402, ep: 0x01 -> 0x01 queue: 0 len: 5 tx.options 0x00
18:42:47:434 APS-DATA.indication srcAddr: 0xc3ca, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0405, lqi: 255, rssi: -51
18:42:47:435 APS-DATA.request id: 36, addrmode: 0x02, addr: 0xc3ca, profile: 0x0104, cluster: 0x0405, ep: 0x01 -> 0x01 queue: 1 len: 5 tx.options 0x00
```

jurriaan, 2019-07-31:
After rejoining the contact sensor and setting the IAS_CIE_Address to the gateway IEEE address I do get more log messages:

```
19:00:58:759 APS-DATA.indication srcAddr: 0xc62f, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x0000, lqi: 255, rssi: -52
19:00:59:767 APS-DATA.indication srcAddr: 0xc62f, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0500, lqi: 255, rssi: -52
19:01:00:775 APS-DATA.indication srcAddr: 0xc62f, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0500, lqi: 255, rssi: -52
19:01:00:822 APS-DATA.indication srcAddr: 0xc62f, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x0000, lqi: 239, rssi: -64
19:01:00:827 APS-DATA.indication srcAddr: 0xc62f, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x0000, lqi: 239, rssi: -64
19:01:01:884 APS-DATA.indication srcAddr: 0xc62f, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0500, lqi: 239, rssi: -64
19:01:01:975 APS-DATA.indication srcAddr: 0xc62f, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x0000, lqi: 255, rssi: -52
19:01:01:984 APS-DATA.indication srcAddr: 0xc62f, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x0000, lqi: 255, rssi: -52
19:01:02:792 APS-DATA.indication srcAddr: 0xc62f, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0500, lqi: 255, rssi: -51
19:01:02:840 APS-DATA.indication srcAddr: 0xc62f, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x0000, lqi: 255, rssi: -61
19:01:02:845 APS-DATA.indication srcAddr: 0xc62f, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x0000, lqi: 255, rssi: -61
19:01:03:807 APS-DATA.indication srcAddr: 0xc62f, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0500, lqi: 255, rssi: -61
19:01:04:269 APS-DATA.indication srcAddr: 0xc62f, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x0000, lqi: 255, rssi: -53
19:01:04:276 APS-DATA.indication srcAddr: 0xc62f, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x0000, lqi: 255, rssi: -52
19:01:04:813 APS-DATA.indication srcAddr: 0xc62f, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0500, lqi: 255, rssi: -53
19:01:05:822 APS-DATA.indication srcAddr: 0xc62f, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0500, lqi: 255, rssi: -57
19:01:06:864 APS-DATA.indication srcAddr: 0xc62f, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0500, lqi: 255, rssi: -49
19:01:07:855 APS-DATA.indication srcAddr: 0xc62f, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0500, lqi: 255, rssi: -55
19:01:08:852 APS-DATA.indication srcAddr: 0xc62f, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0500, lqi: 255, rssi: -52
19:01:09:864 APS-DATA.indication srcAddr: 0xc62f, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0500, lqi: 255, rssi: -57
```

Same for the motion sensor:
```
18:59:28:831 APS-DATA.indication srcAddr: 0xfaf4, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x0000, lqi: 255, rssi: -45
18:59:28:832 APS-DATA.request id: 142, addrmode: 0x02, addr: 0xfaf4, profile: 0x0000, cluster: 0x8000, ep: 0x00 -> 0x00 queue: 2 len: 12 tx.options 0x04
18:59:49:430 APS-DATA.indication srcAddr: 0xfaf4, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x0000, lqi: 255, rssi: -47
18:59:49:432 APS-DATA.request id: 32, addrmode: 0x02, addr: 0xfaf4, profile: 0x0000, cluster: 0x8000, ep: 0x00 -> 0x00 queue: 1 len: 12 tx.options 0x04
18:59:50:492 APS-DATA.indication srcAddr: 0xfaf4, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0500, lqi: 255, rssi: -46
19:00:18:763 APS-DATA.indication srcAddr: 0xfaf4, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x0000, lqi: 255, rssi: -47
19:00:18:764 APS-DATA.request id: 16, addrmode: 0x02, addr: 0xfaf4, profile: 0x0000, cluster: 0x8000, ep: 0x00 -> 0x00 queue: 1 len: 12 tx.options 0x04
19:00:19:867 APS-DATA.indication srcAddr: 0xfaf4, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0500, lqi: 255, rssi: -47
```

jurriaan, 2019-08-01:
I got the motion sensor and the window/door sensor to work, but it requires some manual actions still to get it set up. Hopefully someone more familiar with the matter can help me out. See #1722.

Hypfer, 2019-08-18:
https://github.com/Koenkk/zigbee2mqtt/issues/1689#issue-464244888

According to this issue, the sensors only work on channel 15, 20 and 25.
Is this correct? That might be the reason why mine won't show up in the deconz gui.

phdelodder, 2019-08-27:
Can't see my konke motion sensor in phoscon but visible in home assistant.

sweebee, 2019-08-27:
the sensor is only supported by the api, not by the frontend (phoscon).

Hypfer, 2019-08-28:
@phdelodder @sweebee 

Which zigbee channel are you using?

phdelodder, 2019-08-28:
@Hypfer I'm using channel 15.

@sweebee thanks for the info.

vandalon, 2019-08-30:
Am I the only one or is this sensor slow in reaction to motion? It takes up to 3 seconds for me for the sensor to report motion.

sweebee, 2019-08-31:
Yea, to be honest the sensor is quite bad. Slow reaction. And no “no motion” trigger.

Hypfer, 2019-08-31:
It does however push state updates every few seconds instead of the 1 minute weirdness of the xiaomi motion sensor.

Maybe a combination of those two would make sense? Use the Xiaomi for brightness & fast initial response + the Konke to detect end of motion?

Semi-OT: Is there a proper Motion sensor which does everything well?

sweebee, 2019-08-31:
I have one hue motion sensor and it works great. But also quite expensive. 

The xiaomi pirs work better if you mod them.

ivan-leschinsky, 2019-08-31:
I had two Xiaomi motion sensors( latest version for now, with light sensor). I modified them accurately but they stopped sending any motion at all, pinky illumination data. Even when I returned all the changes I made to the sensors hardware

closed 2019-08-31 by sweebee

subzero79, 2019-09-19:
I managed to set the IAS_CIE address

[image]

channel is set to 25 in deconz, device pairs so it shows in deconz

Also in the logs i can see the response everytime i pass my hand over it

```sh
docker logs -f --tail 50 deconz | grep -i 'b267'
13:07:13:090     * neighbor: 0x086BD7FFFE91562A (0xB267), LQI: 42, relation: 0x01 rxOnWHenIdle: 0
13:07:29:205 APS-DATA.indication srcAddr: 0xb267, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0500, lqi: 255, rssi: -53
13:07:52:349 APS-DATA.indication srcAddr: 0xb267, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0500, lqi: 255, rssi: -53
13:08:00:127     * neighbor: 0x086BD7FFFE91562A (0xB267), LQI: 42, relation: 0x01 rxOnWHenIdle: 0
13:08:07:682 APS-DATA.indication srcAddr: 0xb267, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0500, lqi: 255, rssi: -53
```

But no matter what it doesn't appear in ha, if i query all sensors to the API endpoint also no traces. 
Anything else to try?

nicedevil007, 2019-12-05:
is there anything new to this one?

chrisjbawden, 2019-12-11:
Looks like this is closed.

Currently doesn’t look to be working as of 102.3.

Just in case it is useful to someone working on this in the future (konke motion sensor) here is a dump of some logs of joining the device to a raspberry pi with hassio deployed - 

[0x0000:zdo] ZDO request 0x0036: [60, <Bool.false: 0>]
Device 0xa89a (08:6b:d7:ff:fe:e9:70:18) joined the network
[0xa89a] Requesting 'Node Descriptor'
[0xa89a] Extending timeout for 0x99 request
Device 0xa89a (08:6b:d7:ff:fe:e9:70:18) joined the network
Skip initialization for existing device 08:6b:d7:ff:fe:e9:70:18
[0xa89a:zdo] ZDO request 0x0013: [0xa89a, 08:6b:d7:ff:fe:e9:70:18, 128]
[0xa89a] Node Descriptor: <Optional byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=4712 maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=11264 maximum_outgoing_transfer_size=82 descriptor_capability_field=0>
[0xa89a] Discovering endpoints
[0xa89a] Extending timeout for 0x9b request
[0xa89a] Discovered endpoints: [1]
[0xa89a:1] Discovering endpoint information
[0xa89a] Extending timeout for 0x9d request
[0xa89a:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=260 device_type=1026 device_version=0 input_clusters=[0, 1, 3, 1280, 64704] output_clusters=[3, 64704]>
[0xa89a] Extending timeout for 0x9f request
[0xa89a:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=159 command_id=Command.Read_Attributes_rsp>
[0xa89a:1] Manufacturer: Konke
[0xa89a:1] Model: 3AFE28010402000D
Checking quirks for Konke 3AFE28010402000D (08:6b:d7:ff:fe:e9:70:18)
Considering <class 'zigpy.quirks.smartthings.SmartthingsMultiPurposeSensor'>
Fail because input cluster mismatch on at least one endpoint
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1}
Considering <class 'zhaquirks.lutron.lzl4bwhl01remote.LutronLZL4BWHL01Remote2'>
Fail because profile_id mismatch on at least one endpoint
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because input cluster mismatch on at least one endpoint
Considering <class 'zhaquirks.osram.a19twhite.A19TunableWhite'>
Fail because endpoint list mismatch: {3} {1}
Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
Fail because endpoint list mismatch: {1, 2} {1}
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee_io.XbeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xiaomi.mija.smoke.MijiaHoneywellSmokeDetectorSensor'>
Fail because input cluster mismatch on at least one endpoint
device - 0xa89a:08:6b:d7:ff:fe:e9:70:18 entering async_device_initialized - is_new_join: True
device - 0xa89a:08:6b:d7:ff:fe:e9:70:18 has joined the ZHA zigbee network
[0xa89a](3AFE28010402000D): started configuration
[0xa89a:ZDO](3AFE28010402000D): channel: 'async_configure' stage succeeded
[0xa89a] Extending timeout for 0xa1 request
[0xa89a] Extending timeout for 0xa3 request
[0xa89a:1:0x0500]: started IASZoneChannel configuration
[0xa89a] Extending timeout for 0xa5 request
[0xa89a] Invalid state on future for 0xa1 seq -- probably duplicate response
[0xa89a:1:0x0000]: bound 'basic' cluster: Status.NOT_AUTHORIZED
[0xa89a:1:0x0000]: finished channel configuration
[0xa89a] Extending timeout for 0xa7 request
[0xa89a:1:0x0001]: bound 'power' cluster: Status.SUCCESS
[0xa89a] Extending timeout for 0xa9 request
[0xa89a:1:0x0500]: bound 'ias_zone' cluster: Status.SUCCESS
[0xa89a] Extending timeout for 0xab request
[0xa89a:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=167 command_id=Command.Read_Attributes_rsp>
[0xa89a:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=169 command_id=Command.Configure_Reporting_rsp>
[0xa89a:1:0x0000]: initializing channel: from_cache: False
[0xa89a:1:0x0000]: channel: 'async_configure' stage succeeded
[0xa89a:1:0x0500] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=171 command_id=Command.Write_Attributes_rsp>
[0xa89a:1:0x0001]: reporting 'battery_voltage' attr on 'power' cluster: 3600/10800/1: Result: '[[<ConfigureReportingResponseRecord status=Status.SUCCESS>]]'
[0xa89a:1:0x0500]: wrote cie_addr: 00:0d:6f:00:0e:43:e3:90 to 'ias_zone' cluster: [<WriteAttributesStatusRecord status=Status.SUCCESS>]
[... 32 lines cut ...]
[0xa89a:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=181 command_id=Command.Read_Attributes_rsp>
[... cut ...]
[0xa89a:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=191 command_id=Command.Read_Attributes_rsp>
[0xa89a:1:0x0001]: initializing channel: from_cache: False
[0xa89a:1:0x0001]: channel: 'async_initialize' stage succeeded
[0xa89a](3AFE28010402000D): power source: Battery or Unknown
[0xa89a](3AFE28010402000D): completed initialization

It does pair and show in HA but initially as a water sensor (which you can change it class). After initial pairing it doesn’t look to get any more updates.

[image]

nicedevil007, 2019-12-22:
How did you all achieve writing the IEEE Adress into this special clusterfield?

[image]

Hypfer, 2019-12-22:
@nicedevil007 like this but without the write failing 🙃

nicedevil007, 2019-12-22:
@Hypfer wahhhh I tryed it on my Windows PC now, and also writing failure :( Did you do anything else special to achieve this? Channel is 25 as you wrote in an other thread. Should I press the button on the PIR or even multiple times or do I have to bring the conbee Stick into pairing mode or activate the authenticate thing for 3rd party apps first?

airens, 2020-01-06:
Hello, can anyone tell me, how I can make my second Konke motion sensor (which I bought in december) to appear in HA? I suppose it's not exposed via REST API because of manufacturer code which differs from 0x1002. Am I right?
[image]
[image]

ebaauw, 2020-01-06:
As does the _Model Identifier_.  You’d need to whitelist the new model here
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/bb47cdbc19c257e3000fdb8475aa456830e6df94/de_web_plugin.cpp#L103
You might want to check (search the code for the old _Model Identifier_) if the old model is whitelisted anywhere else, and also whitelist the new model there.

airens, 2020-01-06:
> As does the _Model Identifier_. You’d need to whitelist the new model here [...]

So, I added new MacPrefix `0x14b4570000000000ULL`, new supported device `3AFE28010402000D` but couldn't find, where to add new manufacture code. It have to be set instead of VENDOR_EMBER?

ebaauw, 2020-01-06:
> It have to be set instead of VENDOR_EMBER?

Yes.  You might want to lookup to whom the max prefix belongs and who is vendor 0x1268 for the right names for the #defines.

airens, 2020-01-06:
> > It have to be set instead of VENDOR_EMBER?
> 
> Yes. You might want to lookup to whom the max prefix belongs and who is vendor 0x1268 for the right names for the #defines.

Sorry,  I couldn't find where does the previous manufacture code `0x1002` set to just use it as example. Also, I don't understand, why previous model is set to `VENDOR_EMBER` which contains mask `emberMacPrefix` but not, for example, new `VENDOR_KONKE` #define with `konkeMacPrefix`?

ebaauw, 2020-01-06:
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/bb47cdbc19c257e3000fdb8475aa456830e6df94/de_web_plugin_private.h#L253

Typically these are the codes belonging to the OEM manufacturer.  They're used in devices from multiple, different commercial vendors.

Some extensive googling reveals:
https://github.com/wireshark/wireshark/blob/13703434da30977805ed44f8252c6ea94e119d72/epan/dissectors/packet-zbee.h#L958

And `14:B4:57 SiliconL	Silicon Laboratories` on https://www.wireshark.org/tools/oui-lookup.html.

airens, 2020-01-06:
> https://github.com/dresden-elektronik/deconz-rest-plugin/blob/bb47cdbc19c257e3000fdb8475aa456830e6df94/de_web_plugin_private.h#L253
>
> Typically these are the codes belonging to the OEM manufacturer. [...]

Thanks. Looks like I have done [https://github.com/dresden-elektronik/deconz-rest-plugin/pull/2293](url)

Juanezm, 2020-01-06:
> > https://github.com/dresden-elektronik/deconz-rest-plugin/blob/bb47cdbc19c257e3000fdb8475aa456830e6df94/de_web_plugin_private.h#L253
> > 
> > Typically these are the codes belonging to the OEM manufacturer. They're used in devices from multiple, different commercial vendors.
> > Some extensive googling reveals:
> > https://github.com/wireshark/wireshark/blob/13703434da30977805ed44f8252c6ea94e119d72/epan/dissectors/packet-zbee.h#L958
> > And `14:B4:57 SiliconL Silicon Laboratories` on https://www.wireshark.org/tools/oui-lookup.html.
> 
> Thanks. Looks like I have done it: #2293.

Does this change mean that Phoscon would detect the Konke motion sensor?

airens, 2020-01-06:
> Does this change mean that Phoscon would detect the Konke motion sensor?

No, it’s just all about the REST plugin

dbweb-ch, 2020-01-10:
I've got the same sensor and it shows up in the network. It did read node info as following:

[image]

It also gets updated when motion is detected (Last Seen Updates). I am not able to read any cluster basic info, and IAS_CIE_Address as mentioned from others here is at 0 too.

What do I have to do to expose it to the RestAPI and finally get it into iobroker? I am not very familiar with deconz, just used it for some xiaomi buttons so far...

mrdago, 2020-03-30:
Yesterday I've updated my deCONZ RasBee GW to Version 2.05.74/Firmware 26350500. Looking to the messages received by the deCONZ zigbee web server on port 443, I was surprised by notifying event messages from the Konke Motion Sensor (Zigbee channel 15). Now, I can use this small, nice sensor within my Home Automation System. Unfortunately the sensor doesn't show up in Phoscon. How much effort is necessary to list the device in the Phoscon app?  I would be pleased if this device would be available in Phoscon in one of the next versions.
Here are the messages I receive from the decCONZ zigbee web server:
```
device:  {
  e: 'changed',
  id: '48',
  r: 'sensors',
  state: {
    lastupdated: '2020-03-30T05:04:23',
    lowbattery: false,
    presence: true,
    tampered: false
  },
  t: 'event',
  uniqueid: '14:b4:57:ff:fe:46:a6:1d-01-0500'
}
device:  {
  e: 'changed',
  id: '48',
  r: 'sensors',
  state: {
    lastupdated: '2020-03-30T05:05:23',
    lowbattery: false,
    presence: false,
    tampered: false
  },
  t: 'event',
  uniqueid: '14:b4:57:ff:fe:46:a6:1d-01-0500'
}

```

Doreinie, 2020-08-17:
Hi, it seems the sensor still does not show in Phscon but in deconz. 
Is there a plan to have this one fully integrated?

aap82, 2020-09-11:
> Hi, it seems the sensor still does not show in Phscon but in deconz.
> Is there a plan to have this one fully integrated?

I opened a request to add here --> https://github.com/dresden-elektronik/phoscon-app-beta/issues/193

Doreinie, 2020-09-12:
Seems your request was closed and referred back to this thread. 
I do not know why but maybe there is no solution for this manufacturer?
Possible I do not understand some of the previous post for the Konke sensor.

> > Hi, it seems the sensor still does not show in Phscon but in deconz.
> > Is there a plan to have this one fully integrated?
> 
> I opened a request to add here --> [dresden-elektronik/phoscon-app-beta#193](https://github.com/dresden-elektronik/phoscon-app-beta/issues/193)

retrography, 2020-10-27:
There is an issue with the current implementation: It reports battery level as null, resulting in a low battery status in gateways such as homebridge. At the same time I can see that the device itself reports a lowbattery state of false. Can't we disable battery percentage reporting altogether and only keep reporting the lowbattey status?

Smanar, 2020-10-27:
I have done a correctif some days ago > https://github.com/dresden-elektronik/deconz-rest-plugin/pull/3347
You are not concerned by it ?

retrography, 2020-10-27:
It looks like it, thanks! I will report back if the beta doesn't solve the issue.

retrography, 2020-12-03:
Nope, this didn't solve the issue. The battery value is still not reported. Apparently @SilverWx wanted to have a look at the issue later, but that never happened: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3239#issuecomment-700332510

Smanar, 2020-12-03:
I don't remember all the story, do you have the device JSON ?

SilverWx, 2020-12-03:
> Nope, this didn't solve the issue. The battery value is still not reported. Apparently @SilverWx wanted to have a look at the issue later, but that never happened: [#3239 (comment)](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3239#issuecomment-700332510)

I will try again this weekend and see if I can catch any battery value

SwoopX, 2020-12-03:
@SilverWx No need to try, you won't get anything. Apparently, consideration of the battery reports has been overlooked. Will be available with the next release.

SilverWx, 2020-12-04:
You mean that konke missed to report the battery ? And only new hardware will get that correction ?

retrography, 2020-12-04:
I think it means that the Konke battery patch didn't make it to the current release. That's what I gather from the commit that is referenced in the post (above).