dresden-elektronik/deconz-rest-plugin #1609: within deconz is a group 'All' this group can not be deleted why?

holli73, 2019-06-19:
hello,

after updating to .66 (but maybe releases before - i'm not always checking the ui) - i have now a group called 'All' but this group can not be deleted why?

thanks
holli

robertlandes, 2019-06-19:
That was a a special type of hidden group, where all devices belonged to starting with the latest release it has been implemented as a standard group `/group/0` so that’s probably why you see it now.

See https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_66

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

manup, 2019-06-20:
The closing this issue in favor of https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1612 to track the bugfix.

closed 2019-06-20 by manup