dresden-elektronik/deconz-rest-plugin #1610: Suddenly, all lights and sensors are unreachable

rsedlmeier, 2019-06-19:
Without any changes, all sensors and lights are unreachable.
Even, when I use a backup - complete pi3 backup on other SD cards - I get the same effect.

I am absolutely desperate, cause I have no idea what's going on that machine.
I usually have spare SD cards with older versions and that doesnt work either.

Once I disable the system service and run it manually, I see the deCONZ mask which cannot connect to RaspBee.

manup, 2019-06-19:
Please try the hidden Zigbee configuration page as described here:

https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_59

It is often helpful for recovering a broken configuration.

rsedlmeier, 2019-06-20:
a restore of the raspbee gateway configuration by using WebApp solved the problem.
open is still, why raspbee lost the configuration and had a bad network id of 0x000.

closed 2019-06-20 by rsedlmeier