dresden-elektronik/deconz-rest-plugin #1611: [Request Device Support] Heimann HS1HT-E

bedaes, 2019-06-19:
I've just received a Heimann HS1HT-E. There is the HS1HT on the supported devices page, so I'm not entirely sure if they are the same? At least I can't add the sensor through Phoscon App.

The sensor has two nodes:

[image]

### Node 01 Info

[image]

### Node 01 Cluster Info

[image]

[image]

[image]

[image]

[image]

### Node 02 Info

[image]

### Node 02 Cluster Info

[image]

[image]

[image]
 
[image]

ebaauw, 2019-06-19:
Looks like a newer model, with the new _Manufacturer Code_.  It simply needs to be whitelisted for support by the REST API plugin.  Note to self: double check that it’s whitelisted for setting up  attribute reporting.

Are you sure it exposes the same _Model Identifier_ on both endpoints?  The old model uses different values.

bedaes, 2019-06-19:
That was quick :+1: 

I double checked and yes, the _Model Identifier_ is reported as `TH-EM` on both endpoints.

bedaes, 2019-10-19:
I have successfully added the sensor using Phoscon App and it's working fine using Home Assistant. It doesn't show up in Phoscon App though, therefore I can't rename the sensor. Is this to be expected?

closed 2020-02-23 by stale[bot]