dresden-elektronik/deconz-rest-plugin #1612: /groups/0 bugs in v2.05.66

ebaauw, 2019-06-19:
All of a sudden, `/groups/0` is returned when GETting `/groups`.  Note that the lights are missing.  I have no idea where the scenes come from.  Also note the `id`: 0.
```
$ ph get /groups
{
  "0": {
    "action": {
      "bri": 127,
      "colormode": "hs",
      "ct": 0,
      "effect": "none",
      "hue": 0,
      "on": false,
      "sat": 127,
      "scene": null,
      "xy": [
        0,
        0
      ]
    },
    "devicemembership": [],
    "etag": "2a5d53cba47db543c2e362451d984169",
    "id": "0",
    "lights": [],
    "name": "All",
    "scenes": [
      {
        "id": "1",
        "lightcount": 0,
        "name": "Scene 1",
        "transitiontime": 0
      },
      {
        "id": "2",
        "lightcount": 0,
        "name": "Scene 2",
        "transitiontime": 0
      },
      {
        "id": "3",
        "lightcount": 0,
        "name": "Scene 3",
        "transitiontime": 0
      },
      {
        "id": "4",
        "lightcount": 0,
        "name": "Scene 4",
        "transitiontime": 0
      },
      {
        "id": "5",
        "lightcount": 0,
        "name": "Scene 5",
        "transitiontime": 0
      },
      {
        "id": "6",
        "lightcount": 0,
        "name": "Scene 6",
        "transitiontime": 0
      }
    ],
    "state": {
      "all_on": false,
      "any_on": false
    },
    "type": "LightGroup"
  },
  ...
}
```
The lights are present when GETting `/groups/0`.  However, now the scenes are absent and `id` returns the Zigbee group ID:
```
$ ph get /groups/0
{
  "action": {
    "bri": 127,
    "colormode": "hs",
    "ct": 0,
    "effect": "none",
    "hue": 0,
    "on": false,
    "sat": 127,
    "scene": null,
    "xy": [
      0,
      0
    ]
  },
  "devicemembership": [],
  "etag": "50566800a798635e56506d4fe8298f72",
  "id": "65520",
  "lights": [
    "2",
    "3",
    "1",
    "4",
    "6",
    "7"
  ],
  "name": "All",
  "scenes": [],
  "state": {
    "all_on": false,
    "any_on": true
  },
  "type": "LightGroup"
}
```
The same response is returned when GETting `/groups/65520`.  Note that this resource is _not_ included in the response of GET `/groups`.

manup, 2019-06-20:
Yes our had the first reports of some users getting confused by this.

How is the hue api handling this? From the documentation I would assume they hide group 0 and the related real Zigbee group entirely from the API.

KodeCR, 2019-06-20:
Sorry, it should be hidden for /groups (and shows as you have it for /groups/0), probably some check I missed porting over from #1471 , I'll have a look.
The difference between /groups and /groups/0 is surprising though.

KodeCR, 2019-06-20:
Hmm, the one from /groups is the old group 0 (with id = 0) which should have had it's state set to deleted:
```
    // create default group
    if (gwGroup0 == 0)
    {
        // get new id and replace old group0 and get new id
        for (uint16_t i = 0xFFF0; i > 0; i--) // 0 and larger than 0xfff7 is not valid for Osram Lightify
        {
            Group* group = getGroupForId(i);
            if (!group)
            {
                gwGroup0 = i;
                // delete old group 0
                Group* group = getGroupForId(0);
                if (group) 
                {
                    group->setState(Group::StateDeleted);
                }
                queSaveDb(DB_CONFIG | DB_GROUPS, DB_LONG_SAVE_DELAY);
                break;
            }
        }
    }

    Group* group = getGroupForId(gwGroup0);
    if (!group)
    {
        // new default group
        Group group;
        group.setAddress(gwGroup0);
        group.setName("All");
        groups.push_back(group);
        queSaveDb(DB_GROUPS, DB_LONG_SAVE_DELAY);
    }
```
Ah, I see the problem now. getGroupForId(0) translates 0 to gwGroup0. So "gwGroup0 = i;" should be moved to after the deletion. And now we need to add code to always delete the old group 0, since I missed our one time chance... @manup do you want me to create a pull request for that or have you got it from here?

manup, 2019-06-20:
> @manup do you want me to create a pull request for that or have you got it from here?

A PR would be best :)

ebaauw, 2019-06-20:
> How is the hue api handling this? From the documentation I would assume they hide group 0 and the related real Zigbee group entirely from the API.

The Hue API assigns resource IDs to `/groups` resources (just as to `/lights` resources), and hides the Zigbee group ID completely.  `/groups/0` is not included in the response to GET `/groups`.

KodeCR, 2019-06-21:
The PR fixes /groups, /groups/0 remains accessible in the same way the hue bridge allows this. The id shows the real zigbee id, the hue bridge does not have this, we could hide this but I'm not sure if that would cause other problems.

eskimos2000, 2019-07-22:
When I made GET /groups I have group 0 which is very confusing. I have this in Phoscan app as well. 

`{
    "0": {
        "action": {
            "bri": 128,
            "colormode": "hs",
            "ct": 0,
            "effect": "none",
            "hue": 0,
            "on": false,
            "sat": 128,
            "scene": null,
            "xy": [
                0,
                0
            ]
        },
        "devicemembership": [],
        "etag": "d23a2e9c8bd30ec9ce33c9d5600ecfbf",
        "id": "0",
        "lights": [
            "2",
            "3"
        ],
        "name": "Group 0",
        "scenes": [],
        "state": {
            "all_on": false,
            "any_on": false
        },
        "type": "LightGroup"
    }
}`

Any help with this, please?

closed 2019-11-26 by stale[bot]