dresden-elektronik/deconz-rest-plugin #1614: Hue bulbs do not reconnect after power cycle

l-mb, 2019-06-19:
I have a few Hue bulbs and lights paired successfully. (A White Ambiance light, and a bathroom ceiling and mirror light.)

However, once someone accidentally uses the actual power switch to cycle the lights, they do not automatically reconnect when powered on; Phoscon shows them as stale and my connected Home Assistant shows them as "Unavailable".

When I activate "discover new lights" however, they reconnect fine, but I'd rather not do that every time.

I've tried to reset them via the dimmer switch, and while that worked, the problem did not disappear.

My various other devices from Dresden Electronic, Osram, Busch-Jaeger, GLEDOPTO, Illuminize etc don't show this problem and reconnect immediately.

This is running 2.05.66 with a RaspBee right now. I'm not quite sure when this behaviour arose.

robertlandes, 2019-06-20:
Which firmware do those bulbs have? And what are the exact models?

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

l-mb, 2019-06-20:
The two most important ones:
- Philips LTW017 (Hue Adore LED Mirror light, CCT light), 1.16.1_r19181
- Philips LTC021 (Hue Adore Ceiling Light), 1.16.1_r19181

https://www2.meethue.com/de-de/support/veroffentlichungshinweis/lamps-luminaires suggests I might want to update that. Looks like deCONZ can't do it though, so I guess I need to order a Hue bridge ...

ebaauw, 2019-06-20:
> Looks like deCONZ can't do it though

Actually it can, but Philips don’t publish their firmware.  So far we have limited success figuring out where the Hue bridge gets the firmware from, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/270.

brianajensen2, 2019-06-21:
I have the same problem

lordkev, 2019-06-22:
I'm not sure if this is the same issue or not, but after rebooting my Pi running Hassio, all of my lights in Phoscon are greyed out with no ability to control them (all Hue lights). I don't see anything in the log that would indicate a problem of any sort. Tried running through the add new lights wizard and that didn't help either.

manup, 2019-06-22:
Would be interesting what exactly happens when the light is power-cycled. for example does is send a Device Announce command and is it discovered in Neighbour tables of other nodes.

Can you please start deCONZ with debug enabled:

    deCONZ --http-port=80 --dbg-info=1 --dbg-zdp=1 | tee log.txt

Then power-cycle the Hue light. In the log we should see if there is any life sign.

l-mb, 2019-06-22:
Here's that log. The two lights in question are, creatively, called Bath Ceiling / Mirror ;-) They were on and connected when I started deCONZ and waited until the graph appeared to have settled. Then after a minute or two I turned them both off, and a minute later or so turned them back on.

[log.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/3317279/log.txt)

l-mb, 2019-06-25:
The firmware update (combined with setting the network security to high) has resolved this issue.

closed 2019-06-25 by l-mb