dresden-elektronik/deconz-rest-plugin #1615: Support for Merten Wiser

phil239, 2019-06-20:
Hi,

Are you planning to support the Wiser system by Merten, especially the Wiser Home Touch (which integrates all Wiser devices into a ZigBee network) in near future?

Thanks,
Philipp

manup, 2019-06-20:
Are you sure this is Zigbee? I can't find anything mentioning Zigbee on their page.

phil239, 2019-06-20:
Yes, very sure. Have a look at the product specifications:

http://pdf.schneider-electric.nu/Generatepdf.aspx?salesorgid=42&dist=600&lang=10&form=no&documenttype=plain&productid=MEG5050-0000&render=eShopV3&TemplateID=1

manup, 2019-06-20:
Interesting, but I'm afraid this is only the touch panel (Zigbee coordinator/gateway?). Question is if the other devices are also using Zigbee. The following looks more like Bluetooth:

https://www.merten.de/fileadmin/import/files/merten/anleitungen_dokumentationen/MEG51_Wiser_DE_Web.pdf

phil239, 2019-06-20:
Wiser devices support both Zigbee and Bluetooth. When you don‘t have the Wiser Home Touch panel you control the devices directly with your smartphone via Bluetooth. When you do have the panel, it switches your devices into Zigbee mode

phil239, 2020-04-04:
Hi,

I would like to re-open this issue because the Wiser system is gaining more and more relevance. And in combination with the Wiser Home Touch Gateway it is completely compatible with Zigbee - which also applies to the controllers:

E.g.

https://zigbeealliance.org/de/zigbee_products/weiser-Druckknopfmodul-1-Gang-Shutter-Controller/

Any chance to get this supported?

wolfgang-steinberg, 2020-07-29:
Hi,

I would also like to see this re-opened. I called the Wiser technical support today, they said they don't have any documentation available, and do not officially support ZigBee.

Has anyone tried to discover controllers with what deconz offers out of the box?

anmag, 2020-08-17:
Hi,

I've had little luck trying to connect wiser devices ([Rotary dimmer](https://zigbeealliance.org/zigbee_products/wiser-rotary-dimmer/)) to home assistant using conbee on raspberry pi. I know, however, that this has been done with [Homey](https://homey.app/en-us/) using ZigBee without Schneider Electric's Home Touch Gateway.

The driver written for Wiser on the Homey plattfrom can be found here:
[https://github.com/Signalare/com.se.wiserapp](https://github.com/Signalare/com.se.wiserapp)

Hopefully, the necessary information can be found there such that these devices can be supported.

axellebot, 2020-10-16:
Hey !

If it can help 

From Homey community :
- App : https://homey.app/sv-se/app/com.se.wiserapp/Schneider-Electric-Wiser/ 
- Source code : https://github.com/Signalare/com.se.wiserapp

From Zigate community :
- Forum : https://easydomoticz.com/forum/viewtopic.php?t=9187
- Source code : https://github.com/pipiche38/Domoticz-Zigate/search?q=schneider

Mimiix, 2020-10-20:
We need someone with the device to provide screenshots. @anmag do you have one?

anmag, 2020-10-20:
I've tried to connect the device to ConBee 2 using deConz on Windows 10, without luck. I'd be happy to provide the screenshots if you can tell me how I can make the connection.

So far I have created a network in deConz and scanned for devices in both phoscon and the webapp, but the Wiser device doesn't connect.

Otherwise, you can try extracting the information you need from this [file].(https://github.com/Signalare/com.se.wiserapp/blob/master/drivers/ZB_NHROTARY_DIMMER_1/interview.json)

axellebot, 2020-10-20:
On my side I have this product: https://www.se.com/ww/en/product/NU550818/new-unica---wiser-blind-control-switch---white-%28z%29/ (maybe this product on zigbee alliance website: https://zigbeealliance.org/zigbee_products/wiser-blinds-and-shutters-control-switch/)

My tests to make the switch joining the zigbee network have all failed :/
I'm maybe bad at using conbee software. Even if I already added 2 xiaomi sensor through the zigbee integration on home-assistant, I can't see them on the network management software.

anmag, 2020-10-21:
I finally got the the screenshots, turned out that the device was in pairing mode for bluetooth and not zigbee when I tried earlier.
[image]
[image]
[image]
[image]
[image]

Mimiix, 2020-10-21:
Looks integrated for me. 
Need to wait for a dev tho ^^ 
We do need screens of every cluster being read tho.

anmag, 2020-10-21:
Here are screenshots of all clusters on endpoint 3 (Dimmable Light)
Let me know if you need anything else.
[clusters.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5418596/clusters.zip)

polyzois, 2020-10-28:
Hi, I have the puck dimmer of the same Wiser series (https://github.com/Signalare/com.se.wiserapp/tree/master/drivers/ZB_PUCK_DIMMER_1). If needed I could probably provide data from it.

SwoopX, 2020-11-22:
The device looks ok for usage from here.

closed 2020-12-25 by stale[bot]

yorklars, 2021-01-09:
[image]

hey,
here i found my blinds shutter from merten, but the deconz app said its a light, what can i do?

phil239, 2021-02-23:
> [image]
> 
> hey,
> here i found my blinds shutter from merten, but the deconz app said its a light, what can i do?

Were you able to fix this? What did you do to get your devices recognized by the deconz app?

Is anyone still planning to integrate Wiser „officially“?

tomillr, 2021-03-16:
I'd second this request. The Merten Zigbee-device would be a nice option to retrofit existing installations with matching buttons, dimmers and switches.

Mimiix, 2021-03-16:
@tomillr I suggest doing a new device request following the issue template.