dresden-elektronik/deconz-rest-plugin #1616: state.bri missing for dimmable light since 2.05.66

DOliana, 2019-06-20:
I have a Philips LWB010 setup in deConz (which is connected to OpenHab - I am using it as a wake up light). 

Since 2.05.66 when I look at the state through postman the state looks like this:
```json
   "state": {
        "alert": "none",
        "on": true,
        "reachable": true
    }
```

Since I am using deConz in docker I quickly siwtched to version 2.05.65 and realized that the state there is 
```json
    "state": {
        "alert": "none",
        "bri": 10,
        "on": true,
        "reachable": true
    },
```

Thus I found out why my wake uplight is not working :-)

AntiHeld889, 2019-06-20:
Same problem

manup, 2019-06-22:
Can you please provide the complete REST-API output of the light, I need to check the light.type.

ebaauw, 2019-06-22:
I can confirm the issue, but oddly, only for my Hue dimmable lights, e.g.:
```
{
  "etag": "054eff2283559cc8cdc2022523b8258b",
  "hascolor": false,
  "manufacturername": "Philips",
  "modelid": "LWB006",
  "name": "Entrance 1",
  "state": {
    "alert": "none",
    "on": false,
    "reachable": true
  },
  "swversion": "5.127.1.26420",
  "type": "Dimmable light",
  "uniqueid": "00:17:88:01:10:52:ad:e6-0b"
}
```
Whereas non-Hue lights seem unaffected, e.g. innr:
```
{
  "etag": "5e9b3458c75ae8ac4c8ff0d62544e862",
  "hascolor": false,
  "manufacturername": "innr",
  "modelid": "UC 110",
  "name": "Kitchen Left",
  "state": {
    "alert": "none",
    "bri": 254,
    "on": false,
    "reachable": true
  },
  "swversion": "0x10000c46",
  "type": "Dimmable light",
  "uniqueid": "00:15:8d:00:00:cd:d7:f7-01"
}
```
iCasa:
```
{
  "etag": "719b475da420152dd38563c24b11a1f4",
  "hascolor": false,
  "manufacturername": "icasa",
  "modelid": "ICZB-IW11D",
  "name": "Dining Room Table",
  "state": {
    "alert": "none",
    "bri": 254,
    "on": false,
    "reachable": true
  },
  "swversion": "2.4.1_r30",
  "type": "Dimmable light",
  "uniqueid": "00:0d:6f:00:10:65:6f:f1-01"
}
``` 
and GLEDOPTO:
```
{
  "etag": "e3531f64f5276b6d6b37ab0d80b1d325",
  "hascolor": false,
  "manufacturername": "GLEDOPTO",
  "modelid": "GLEDOPTO",
  "name": "Balcony 1",
  "state": {
    "alert": "none",
    "bri": 1,
    "on": false,
    "reachable": true
  },
  "swversion": "1.0.2",
  "type": "Dimmable light",
  "uniqueid": "00:12:4b:00:13:0a:5f:ed-0a"
}
```

ebaauw, 2019-06-22:
The iCasa dimmers are Zigbee 3.0, with a ZHA light endpoint (so probably a different device type); the innr and GLEDOPTO lights have a ZLL light endpoint, just like the Hue lights.  They don't have a ZGP endpoint though (not sure if that's even related).

manup, 2019-06-22:
> I can confirm the issue, but oddly, only for my Hue dimmable lights, e.g.:

Which device type does the Simple Descriptor of the LWB006 have?

ebaauw, 2019-06-22:
I don't see a difference:
Hue:
```
IEEE 802.15.4 Data, Dst: 0x0000, Src: 0x5027
ZigBee Network Layer Data, Dst: 0x0000, Src: 0x5027
ZigBee Application Support Layer Data, Dst Endpt: 0, Src Endpt: 0
ZigBee Device Profile, Simple Descriptor Response, Device: 0x5027, Status: Success
    Sequence Number: 159
    Status: Success (0)
    Device: 0x5027
    Simple Descriptor Length: 24
    Simple Descriptor
        Endpoint: 11
        Profile: 0xc05e
        Application Device: 0x0100
        Application Version: 0x0002
        Input Cluster Count: 7
        Input Cluster List
            Input Cluster: 0
            Input Cluster: 3
            Input Cluster: 4
            Input Cluster: 5
            Input Cluster: 6
            Input Cluster: 8
            Input Cluster: 4096
        Output Cluster Count: 1
        Output Cluster List
            Output Cluster: 25
```
innr:
```
Frame 8818: 162 bytes on wire (1296 bits), 162 bytes captured (1296 bits) on interface 0
Ethernet II, Src: Raspberr_dd:8a:80 (b8:27:eb:dd:8a:80), Dst: Apple_ca:3b:53 (a4:5e:60:ca:3b:53)
Internet Protocol Version 4, Src: 192.168.76.83, Dst: 192.168.76.41
User Datagram Protocol, Src Port: 49181, Dst Port: 17754
ZigBee Encapsulation Protocol, Channel: 25, Length: 88
IEEE 802.15.4 Data, Dst: 0x0000, Src: 0x8507
ZigBee Network Layer Data, Dst: 0x0000, Src: 0x19d9
ZigBee Application Support Layer Data, Dst Endpt: 0, Src Endpt: 0
ZigBee Device Profile, Simple Descriptor Response, Device: 0x19d9, Status: Success
    Sequence Number: 189
    Status: Success (0)
    Device: 0x19d9
    Simple Descriptor Length: 38
    Simple Descriptor
        Endpoint: 1
        Profile: 0xc05e
        Application Device: 0x0100
        Application Version: 0x0002
        Input Cluster Count: 8
        Input Cluster List
            Input Cluster: 0
            Input Cluster: 4
            Input Cluster: 3
            Input Cluster: 6
            Input Cluster: 8
            Input Cluster: 5
            Input Cluster: 65535
            Input Cluster: 25
        Output Cluster Count: 7
        Output Cluster List
            Output Cluster: 0
            Output Cluster: 4
            Output Cluster: 3
            Output Cluster: 6
            Output Cluster: 8
            Output Cluster: 5
            Output Cluster: 25
```

manup, 2019-06-22:
Can you please try if the following commit fixes the issue: https://github.com/dresden-elektronik/deconz-rest-plugin/commit/1bacf461494372496772b84682d9c0b124170a4b

0x0100 is DEV_ID_HA_ONOFF_LIGHT which is handled in
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/light_node.cpp#L347-L355

ebaauw, 2019-06-22:
No, it doesn't.

ebaauw, 2019-06-22:
But the above commit does.  Not sure why it wasn't working, though.

We might actually want to ditch the check on `manufacturerCode() == VENDOR_PHILIPS`, since this probably holds for any ZHA/Zigbee 3.0 _On/Off Light_ that exposes _Level Control_ to support _Move to Level (with On/Off)_.

manup, 2019-06-27:
>We might actually want to ditch the check on manufacturerCode() == VENDOR_PHILIPS, since this probably holds for any ZHA/Zigbee 3.0 On/Off Light that exposes Level Control to support Move to Level (with On/Off).

Yes, I'm just a bit afraid that it might affect other devices which don't implement the standard properly.

ebaauw, 2019-06-28:
OK, I'll submit a PR with the vendor check present.

closed 2019-07-09 by manup

DOliana, 2019-08-08:
Thank you for fixing the issue @manup . I am just wondering if it is contained in the current release, since the version did not change. (it seems that it is not the case, since 2.05.66 is the latest build and it was created before the PR was merged.)

lettore, 2019-08-18:
I have the same problem, LWB10 bulbs without brightness sliders and I'm on the lastest version 2.05.66. I use a docker container addon of Home Assistant, so I'm waiting for a new version that will resolve this bug.

manup, 2019-08-19:
2.05.67 will arrive in a few days and will contain the submitted fixes.

commit c12dae1f (2019-07-09): Update light_node.cpp

Fix #1616.