dresden-elektronik/deconz-rest-plugin #1618: ConBee II reconnect loop

Skwaaier, 2019-06-20:
Dear all,

A while ago I bought a ConBee II stick to communicate with my lights and remote from within home assistant, which is running on a raspi 3b+. I have installed the deCONZ add-on to home assistant accordingly, assigned the device to /dev/ttyACM0 and disabled the bluetooth in the config.txt file.

System info:
HassOS version 2.12
Home assistant version 0.93.2
deCONZ add-on version 2.7
Firmware version 2.5.65 (as shown by home assistant integrations)

The ConBee II is connected via an extension cable and no other devices are connected via USB.

It appears that the ConBee II stick is found, however it is connected and disconnected every 15 seconds as shown below. 

```
[Hass.io] deCONZ device: /dev/ttyACM0
[Hass.io] Starting in minimal mode, without VNC GUI
libpng warning: iCCP: known incorrect sRGB profile
This plugin does not support propagateSizeHints()
This plugin does not support propagateSizeHints()
20:25:27:508 HTTP Server listen on address 0.0.0.0, port: 8080, root: /usr/share/deCONZ/webapp/
20:25:27:534 CTRL. 3.16.220:25:27:615 dev /dev/ttyAMA0
20:25:27:616 COM: --dev: /dev/ttyACM0 (ConBee II)
20:25:27:616 ZCLDB init file /data/.local/share/dresden-elektronik/deCONZ/zcldb.txt
20:25:27:784 parent process /bin/sh
20:25:27:784 gw run mode: docker/hassio
20:25:27:784 GW sd-card image version file does not exist: /data/.local/share/dresden-elektronik/deCONZ/gw-version
20:25:27:784 sd-card cid: 9f5449534433324761ffffffff012a17
20:25:27:785 DB sqlite version 3.16.2
20:25:27:787 DB PRAGMA page_count: 30
20:25:27:787 DB PRAGMA page_size: 4096
20:25:27:787 DB PRAGMA freelist_count: 0
20:25:27:787 DB file size 122880 bytes, free pages 0
20:25:27:787 DB PRAGMA user_version: 6
20:25:27:787 DB cleanup
20:25:27:788 DB create temporary views
20:25:27:794 don't close database yet, keep open for 900 seconds
20:25:27:795 started websocket server at port 8443
20:25:27:800 discovery updated announce interval to 10 minutes
20:25:27:803 found node plugin: libde_rest_plugin.so - REST API Plugin
20:25:27:806 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin
20:25:27:821 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin
20:25:27:852 dev /dev/ttyAMA0
20:25:27:852 COM: --dev: /dev/ttyACM0 (ConBee II)
20:25:28:533 wait reconnect 15 seconds
20:25:28:572 dev /dev/ttyAMA0
20:25:28:572 COM: --dev: /dev/ttyACM0 (ConBee II)
20:25:29:484 wait reconnect 14 seconds
20:25:30:481 wait reconnect 13 seconds
20:25:31:481 wait reconnect 12 seconds
20:25:32:481 wait reconnect 11 seconds
20:25:33:033 dev /dev/ttyAMA0
20:25:33:228 Announced to internet
20:25:33:228 discovery server date: Wed, 19 Jun 2019 18:25:33 GMT
20:25:33:229 	 local time seems to be ok
20:25:33:229 discovery found version 2.04.35 for update channel stable
20:25:33:481 wait reconnect 10 seconds
20:25:34:481 wait reconnect 9 seconds
20:25:35:481 wait reconnect 8 seconds
20:25:36:482 wait reconnect 7 seconds
20:25:37:481 wait reconnect 6 seconds
20:25:38:481 wait reconnect 5 seconds
20:25:39:481 wait reconnect 4 seconds
20:25:40:481 wait reconnect 3 seconds
20:25:41:482 wait reconnect 2 seconds
20:25:42:481 wait reconnect 1 seconds
20:25:42:524 dev /dev/ttyAMA0
20:25:42:524 COM: --dev: /dev/ttyACM0 (ConBee II)
20:25:43:028 dev /dev/ttyAMA0
20:25:43:481 wait reconnect 15 seconds
20:25:43:520 dev /dev/ttyAMA0
20:25:43:520 COM: --dev: /dev/ttyACM0 (ConBee II)
20:25:44:482 wait reconnect 14 seconds
20:25:45:481 wait reconnect 13 seconds
20:25:46:481 wait reconnect 12 seconds
20:25:47:481 wait reconnect 11 seconds
20:25:48:481 wait reconnect 10 seconds
20:25:49:482 wait reconnect 9 seconds
20:25:50:481 wait reconnect 8 seconds
20:25:51:481 wait reconnect 7 seconds
20:25:52:481 wait reconnect 6 seconds
20:25:53:020 dev /dev/ttyAMA0
20:25:53:481 wait reconnect 5 seconds
20:25:54:481 wait reconnect 4 seconds
20:25:55:481 wait reconnect 3 seconds
20:25:56:481 wait reconnect 2 seconds
20:25:57:481 wait reconnect 1 seconds
20:25:57:527 dev /dev/ttyAMA0
20:25:57:527 COM: --dev: /dev/ttyACM0 (ConBee II)
20:25:58:482 wait reconnect 15 seconds
20:25:58:525 dev /dev/ttyAMA0
20:25:58:525 COM: --dev: /dev/ttyACM0 (ConBee II)
20:25:59:482 wait reconnect 14 seconds
20:26:00:481 wait reconnect 13 seconds
20:26:01:481 wait reconnect 12 seconds
20:26:02:481 wait reconnect 11 seconds
20:26:03:037 dev /dev/ttyAMA0
20:26:03:482 wait reconnect 10 seconds
20:26:04:481 wait reconnect 9 seconds
20:26:05:481 wait reconnect 8 seconds
20:26:05:883 scan skip host .252
20:26:05:986 scan finished
20:26:06:481 wait reconnect 7 seconds
20:26:07:481 wait reconnect 6 seconds
20:26:08:481 wait reconnect 5 seconds
20:26:09:481 wait reconnect 4 seconds
20:26:10:481 wait reconnect 3 seconds
20:26:11:481 wait reconnect 2 seconds
20:26:12:481 wait reconnect 1 seconds
20:26:12:524 dev /dev/ttyAMA0
20:26:12:524 COM: --dev: /dev/ttyACM0 (ConBee II)
20:26:13:020 dev /dev/ttyAMA0
20:26:13:481 wait reconnect 15 seconds
20:26:13:523 dev /dev/ttyAMA0
20:26:13:523 COM: --dev: /dev/ttyACM0 (ConBee II)
```

The device does show up in the Phoscon Web UI as a ConBee II device, where the version is displayed as 2.05.66 and the firmware version is displayed as "Not connected".

Any help/suggestions are appreciated.

manup, 2019-06-20:
>Firmware version 2.5.65 (as shown by home assistant integrations)

That's only the deCONZ version not the firmware version. Currently the ConBee II isn't connected at all, the 15 seconds are only the attempts to connect to it.

Please make sure that the user on the host system is in the group `dialout` as described here: https://phoscon.de/conbee2/install#docker

Can you provide the output of `lsusb` on the host system?

Skwaaier, 2019-06-20:
Thank you for your quick reponse.

I had not added to user to the group `dialout` indeed. Therefore, I've setup a SSH through port 22222 to execute terminal commands in the host system. The HassIO client does not recognize `sudo` or `gpasswd` as a command, thus executing the proposed code results in an error unfortunately. Is there another method for add the user to the `dailout`?

The output of `lsusb` is as following:
```
Bus 001 Device 001: ID 1d6b:0002
Bus 001 Device 003: ID 0424:2514
Bus 001 Device 002: ID 0424:2514
Bus 001 Device 004: ID 0424:7800
```

KonZe187, 2019-06-21:
Hi, I can confirm this issue.

I'm on a Proxmox host with x86 hardware

Ubuntu 18.10
Home Assistant 94.4
deCONZ AddOn 2.7
deCONZ 2.05.66

It worked flawless for a while but for a few hours now, I got the same problem as Skwaaier. The log says reconnecting and the Phoscon App says not connected.

Might be just an issue with Home Assistant... what I find weird is that this didn't happen right after updating the add-on or Home Assistant, it just happened randomly, though not too long after an update.

manup, 2019-06-22:
>The HassIO client does not recognize sudo or gpasswd as a command, thus executing the proposed code results in an error unfortunately. Is there another method for add the user to the dailout?

There is also another command:

    sudo groupadd dialout

I'm not sure if that works (or even is needed) on HassOS.

>The output of lsusb is as following:

>Bus 001 Device 001: ID 1d6b:0002
Bus 001 Device 003: ID 0424:2514
Bus 001 Device 002: ID 0424:2514
Bus 001 Device 004: ID 0424:7800

This looks strange ConBee II isn't in that list.

Skwaaier, 2019-06-22:
The suggested `usermod` command is not valid either unfortunately.

Additionally I've performed the installation of the ConBee II on my laptop (Ubuntu 18.04LTS)  according to the documentation. After a successful installation, the deCONZ software does not detect the stick here either. The drop-down menu in which you should select your device remains blank.

The output of lsusb on this system is as following:
```
Bus 002 Device 002: ID 8087:8000 Intel Corp. 
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:8008 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 8087:07da Intel Corp. 
Bus 003 Device 003: ID 5986:0536 Acer, Inc 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

sluquet, 2019-06-26:
I had the same issue on two sticks so far. Returned both to the store.

Skwaaier, 2019-06-26:
@sluquet Both sticks were ConBee II devices?

sluquet, 2019-06-26:
@Skwaaier Yes. On my third ConBee II now.

kiucheng, 2019-06-28:
I can report I have the same issue. lsusb does not show the device. Then I put it in my Windows 10 laptop. Installed Deconz. It asked for a firmware update. Clicked update and went 100%. Then ask for firmware update again.

sluquet, 2019-06-28:
@kiucheng exactly the same issue I had on both my previous sticks, super annoying. Because of this I am holding off migrating my lights from my Hue hub. Just running a few IKEA bulbs, two switches and three sensors on the ConBee II until I have a feeling it is more reliable.

manup, 2019-06-28:
>I can report I have the same issue. lsusb does not show the device. Then I put it in my Windows 10 laptop. Installed Deconz. It asked for a firmware update. Clicked update and went 100%. Then ask for firmware update again.

Is the ConBee II listed in the windows device manager?

Since the update from deCONZ doesn't work can you please try to update the firmware with the manual update instructions:

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually#update-in-windows

As a side note, we are currently working very hard to get the issues fixed, there will be updated deCONZ firmware and bootloader versions to fix some nasty usb side effects on some systems.

kiucheng, 2019-06-29:
Thanks @manup. It works! I have to try updating firmware 4-5 times before it succeed as the Conbee II keep disconnect and reconnecting. I did the command line update in Windows 10. Now running deconz in a docker in QNAP NAS. FIrmware version 0x26490700

manup, 2019-06-29:
Can you please provide which Windows 10 version/build you're running as shown in thw Windows system settings? We're going to do more testing with different versions.

Was it a native Windows installation or running in a VM?

kiucheng, 2019-06-29:
Native Windows installation. Windows 10 Home 10.0.17134 Build 17134

manup, 2019-06-29:
Thanks, I'll put on the list for testing.

Skwaaier, 2019-08-07:
An update from my side: I've returned the ConBee II device and purchased a new one recently. This time no problems arose during the setup and all devices were added smoothly. So there might have been an issue with the previous device. Anyway, thanks for the input and help all!

closed 2019-08-07 by Skwaaier