dresden-elektronik/deconz-rest-plugin #1619: Bosch Pro-Grade ISW-ZPR1-WP13  Motion Detector

hgelpke, 2019-06-21:
The device pairs properly and reports motion for awhile but seems to go into a sleep mode that stops reporting information. It can be woken up by pressing a button with the case open similar to the pairing procedure but otherwise goes dead. 

Device also fails to report temperature and based on the smartthings discussion that might be what keeps it alive.

bdickhaus, 2019-06-22:
I am seeing the same thing.

closed 2019-10-28 by stale[bot]

mamrai1, 2020-10-09:
[image]
[image]
[image]
Just paired 2 Bosch motion sensors.. Temp is shown but battery state shouldn't report persentage?

SwoopX, 2020-10-09:
Yes, probably. However, any information on what values the devices uses has never been provided. For that, we require a screenshot from the power configuration cluster of deconz GUI while the attributes have been read.