dresden-elektronik/deconz-rest-plugin #1620: deCONZ-homebridge-install.sh needs to be updated

ebaauw, 2019-06-21:
- As of homebridge-hue@0.11.28, homebridge-lib is no longer included in homebridge-hue, and needs to be installed separately.  Best do that in a single command, `npm -g install homebridge-lib homebridge-hue`; `npm` will issue a warning about the peer dependency when installing homebridge-hue separately.
- For some time now (don't know the exact version), homebridge no longer depends on C/C++ modules, so the `--unsafe-perm` option can be dropped when installing homebridge.

I'm happy to provide the changes in a PR, but I cannot test them, as I don't know how to make deCONZ manage the homebridge installation.  Is there a hidden page in Phoscon, or do I need to enable the deconz-homebridge-install service manually?

manup, 2019-06-21:
I've stumbled over this yesterday with a broken installation and created some internal bug report how it failed. A PR is highly welcome, since the colleague maintaining this code is on vacation until July.

The deCONZ Homebridge-hue integration is provided in form of a preconfigured sd-card image "Homebridge-hue + Raspbian Stretch". It has NodeJS, Homebridge and Homebridge-hue installed and the systemd `deconz-homebridge.service` running.

https://phoscon.de/raspbee/sdcard#downloads

When the respective services are running the *Phoscon App > Settings > Gateway* page will show  Homebridge-hue specific UI elements with the HomeKit pin, a Homebridge reset button and auto update option. Unfortunately this isn't covered in the Phoscon App documentation yet but will be added soon. https://phoscon.de/en/app/doc#settings-en

ebaauw, 2019-07-23:
I'm sorry, this whole peer dependency setup proved way too complex.  As of homebridge-hue@0.11.30, homebridge-lib is once again listed as a regular dependency.

closed 2019-11-27 by stale[bot]