dresden-elektronik/deconz-rest-plugin #1622: Core dump when trying to load Zigbee configuration

lordkev, 2019-06-22:
Every time I try to load a Zigbee configuration from the "secret" settings page, I get a core dump. Version 2.05.66 as an add-on within Hassio running on a Pi. I've uninstalled and reinstalled and get the same thing.

```00:44:47:746 dev /dev/ttyAMA0
00:44:47:746 GW firmware version: 0x26490700
00:44:47:746 GW firmware version is up to date: 0x26490700
00:45:40:727 Current channel 11
00:45:40:736 Device TTL 2515 s flags: 0x7
00:46:40:727 Current channel 11
00:46:40:738 Device TTL 2455 s flags: 0x7
00:47:40:726 Current channel 11
00:47:40:737 Device TTL 2395 s flags: 0x7
00:48:40:385 Websocket disconnected 192.168.3.42:51536 (state: 0) 
00:48:40:728 Current channel 11
00:48:40:738 Device TTL 2335 s flags: 0x7
00:48:41:549 New websocket 192.168.3.42:51637 (state: 3) 
00:48:47:747 dev /dev/ttyAMA0
00:48:47:747 GW firmware version: 0x26490700
00:48:47:747 GW firmware version is up to date: 0x26490700
00:48:53:234 Websocket disconnected 192.168.3.42:51637 (state: 0) 
00:48:54:455 New websocket 192.168.3.42:51654 (state: 3) 
00:49:12:733 ZB rowid 1, conf: {"apsUseExtPanId":"0x0","curChannel":11,"deviceType":0,"extPanId":"0x212effff0425a4","fwversion":"0x26490700","macAddress":"0x212effff0425a4","networkKey":"37346262663739353232616233336335","nwkAddress":"0x0","nwkUpdateId":1,"panId":"0xa776","securityMode":3,"staticNwkAddress":false,"swversion":"2.05.66","tcAddress":"0x212effff0425a4"}
00:49:12:733 don't close database yet, keep open for 900 seconds
00:49:18:931 ZB rowid 1, conf: {"apsUseExtPanId":"0x0","curChannel":11,"deviceType":0,"extPanId":"0x212effff0425a4","fwversion":"0x26490700","macAddress":"0x212effff0425a4","networkKey":"37346262663739353232616233336335","nwkAddress":"0x0","nwkUpdateId":1,"panId":"0xa776","securityMode":3,"staticNwkAddress":false,"swversion":"2.05.66","tcAddress":"0x212effff0425a4"}
00:49:18:931 don't close database yet, keep open for 900 seconds
00:49:19:353 ZDP discovery done in 302 ms00:49:29:718 saved node state in 5 ms
00:49:29:743 sync() in 25 ms
00:49:34:725 Device protocol version: 0x010B
00:49:34:789 Current channel 11
00:49:34:811 start reconnect to network
00:49:34:814 CTRL ANT_CTRL 0x03
00:49:34:846 Device protocol version: 0x010B
00:49:34:904 Current channel 11
00:49:34:909 try to reconnect to network try=1
00:49:34:922 CTRL ANT_CTRL 0x03
00:49:34:966 Device protocol version: 0x010B
00:49:35:026 Current channel 11
00:49:35:046 CTRL ANT_CTRL 0x03
00:49:35:471 DEV config changed event
00:49:35:476 Device protocol version: 0x010B
00:49:35:535 Current channel 11
00:49:35:555 CTRL ANT_CTRL 0x03
00:49:35:585 Device protocol version: 0x010B
00:49:35:639 Current channel 11
00:49:35:657 CTRL ANT_CTRL 0x03
00:49:39:709 reconnect network done
00:49:39:772 saved node state in 1 ms
00:49:39:847 sync() in 74 ms
00:49:39:848 REST API plugin shutting down
00:49:39:867 sync() in 17 ms
QXcbConnection: Could not connect to display 
/run.sh: line 63:    36 Aborted                 (core dumped) deCONZ -platform minimal --auto-connect=1 --dbg-info=1 --dbg-aps=0 --dbg-zcl=0 --dbg-zdp=0 --dbg-otau=0 --http-port="${API_PORT}" --ws-port="${WEBSOCKET_PORT}" --upnp=0 --dev="${DECONZ_DEVICE}"```

lordkev, 2019-06-22:
This also happens when I try to load a backup file.

manup, 2019-06-22:
Can you please provide the coredump file, then we can have a look where exactly the code breaks.

closed 2019-10-27 by stale[bot]