dresden-elektronik/deconz-rest-plugin #1623: Losing ability to switch lights individually, brightness and colour temperature

DurrM, 2019-06-22:
Hey,

I discovered a problem with my Aqara lights.
After some time I cannot switch any of my lights individually. Even the settings for brightness and colour temperature are gone in the Phoscon web app.
I can however switch the whole group on/off without any problem. Also the Osram Smart Plug works flawless.
I already tried a complete new installation. Worked for some time, after that the problem occurred again. 
The strange thing is that whenever I switch a light on/off it appears without any error in the homebridge log. 

I'm using the RaspBee (FW: 26330500; 2.05.66)

Hope you guys have some ideas.

[image]
[image]

DurrM, 2019-06-24:
So something is pretty wrong here.

I deleted all my lights with `ph -H 192.168.178.20:90 -u XXXXXXXXXX delete /lights/XX` 

Then I added them again with the help of the Phoscon WebApp. Immediately all the settings (brightness, temperature) are there. Switching them individually on/off works like it is supposed to.

`ph -H 192.168.178.20:90 -u XXXXXXXXXX get /lights` gets me this:

[image]

After just two minutes I do the same command and I get this:

[image]

What's going on here? Is this a problem with 2.05.66? Can I downgrade to 2.05.065? Never had any problems with that version.

Hope somebody can help me.

ebaauw, 2019-06-24:
@manup, this smells like another manifestation of #1616.

I don't know this particular light, but it's whitelisted for special handling in `LightNode::setHaEndpoint()` in `light_node.cpp`.  Apparently, it exposes the wrong device type, causing deCONZ to expose only `state.on`.  The special handling is delaying initialisation, until the _Model Identifier_ has been read, so the device type can be overwritten.  Looking at the first screenshots, this succeeds.

I copied this logic for the iCasa wallswitch (to prevent the exposure of `state.bri`).  However, as a side effect, the REST API would no longer show `state.bri` for Hue dimmable lights (see #1616).  I fixed that by using a different strategy altogether to recognise the iCasa switch (see https://github.com/ebaauw/deconz-rest-plugin/commit/c12dae1f7d105a5b585f101b9901a6bd5e50bf49 - no PR yet), but I never understood why the Hue lights were affected.  I never checked what their resource look like shortly after restarting deCONZ, but I'll bet they do show `state.bri`.

My best guess: somehow the resource gets re-created later (when the light is discovered on the network?), calling `setHaEndPoint()` again.  Since `RStateColorMode` has already been set the first time, the initialisation is skipped, causing the type-dependent `state` attributes to be missing.

manup, 2019-06-25:
>@manup, this smells like another manifestation of #1616.

Yes it seems so, I guess it becomes increasingly challenging to support devices which don't implement the standard very well.

For the OSRAM/LEDVANCE lights I've added the `setLightNodeStaticCapabilities(LightNode *lightNode)` method to add, remove and init `ResourceItems` which aren't (and perhaps shouldn't?) be handled in  `setHAEndpoint()`. Might not be the best solution but is ok'ish maintainable, maybe should be later modified to describe the capabilities in the ZCLDB file.

https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/de_web_plugin.cpp#L1809

DurrM, 2019-06-25:
First of all thank you for your replies.
I gotta be honest with you. I understand only parts of what you wrote.

Just wanted to let you know. I deleted all of the lights again yesterday. Added them again and so far they seem to be working fine.
Will keep you posted if that should change.

ebaauw, 2019-06-28:
> so far they seem to be working fine.

Probably until you restart deCONZ.

closed 2019-11-02 by stale[bot]