dresden-elektronik/deconz-rest-plugin #1624: Rename Xiaomi Magic Cube problem

Lefey, 2019-06-23:
Xiaomi Magic Cube presents as two resource in rest api, first for discrete actions and second for analog turn angle and it has different names such as switch 1 & switch 2. So with home assistant its handy to use event_id "switch 2" for triggering atomation of brightness change for example.
But if i change name of Magic Cube in phoscon web ui, then both resource became same names. With that, i can not filter event_id in home assistant anymore.

iago1460, 2019-06-23:
I noticed something similar, I created an app daemon to help with automations,
[check it out here](https://community.home-assistant.io/t/improved-deconz-xiaomi-aqara-mi-cube-integration-with-appdaemon/122924) but a more permanent solution would be handy.

Lefey, 2019-06-23:
> but a more permanent solution would be handy.

Here temporary solution with manual renaming via deconz api 
https://github.com/home-assistant/home-assistant/issues/23269#issuecomment-486990529

Cheaterdev, 2019-06-23:
This is the same problem with other names:
I've added multisensor device to deconz - "aqara temperature and humidity sensor". It reports 3 values: Temperature, Humidity, Pressure.
I've renamed this device as "Bathroom_Sensor". 

In Home Assistant i see:
- sensor.bathroom_sensor
- sensor.bathroom_sensor_2
- sensor.bathroom_sensor_3

Just because all three created sensors were named as "bathroom_sensor". It's better to automatically generate name with type of sensor:
- sensor.bathroom_sensor_temperature
- sensor.bathroom_sensor_humidity
- sensor.bathroom_sensor_pressure

Possible solution - pack all variables in 1 device with atrtributes, like batery percent state.

Cheaterdev, 2019-06-23:
Also another problem with Xiaomi Magic Cube.
I have only 1 Xiaomi Magic Cube. But state and rotation are events only - there is no "current state" sensor in Home Assistant for cube.
For such devices Home Assistant creates just a "battery level" sensor. As a result - we have 2 "battery level" sensors for Xiaomi Magic Cube. One for "digital event" and one for "analog event". But values are same.

Maybe i need to duplicate this issue to Home Assistant issues?

chimpy, 2019-06-23:
Just FYI you can hide those through customize.yaml or Lovelace. You can always use the event to action an input_boolean giving you the last state.

Cheaterdev, 2019-06-23:
Last state is not a real problem(only impossible to determine current face after "shake" event"). I'm just saying that we have duplicated sensors for same value (Battery Level).

closed 2019-10-28 by stale[bot]