dresden-elektronik/deconz-rest-plugin #1626: Unknow device connected

WhistleMaster, 2019-06-23:
I have a device connected to my zigbee network but I cannot figure out what this device is. All the device I have are reported correctly and this one is an additional device I have no idea what it is. Even if I delete it, after a few moment, it comes back in the network.

[image]

[image]

ebaauw, 2019-06-23:
> Even if I delete it, after a few moment, it comes back in the network.

It's probably a stale entry in the neighbour table of one of your routers.  You may want to power cycle the router at the other end of the red line.

pergolafabio, 2019-06-24:
Yeah, I have the same , deleting doesn't help, just comes back... I have it only on one of my osrams power plug, always connected to that one...

Don't see it in phoscon, only in the mesh

Power cycle doesn't help either, even deleting the Osram, Brings that specific node back

WhistleMaster, 2019-06-24:
> It's probably a stale entry in the neighbour table of one of your routers. You may want to power cycle the router at the other end of the red line.

I tried your proposition, without success unfortunately.

pergolafabio, 2019-06-24:
see my screen here, i have them too , no idea what it is :)

i just ignore them

https://www.dropbox.com/s/i49dqpsx1x4im33/unknowdevices.JPG?dl=0

gpl1998, 2019-06-24:
same here. 

I deleted my five hue motion sensors in Phoscon. All 5 sensor disappeared in Phoscon permanently and for a while in Deconz. After a while, they came back in Deconz (but not in Phoscon). I deleted the 5 nodes in Deconz one more time - again, they disappeared for a while and came back after a little while. 

I sold the sensors and they left my house. Again, I deleted the nodes in Deconz and now the were gone. This weekend, I rebooted the Raspberry and - really strange - three of the five sensors appeared again. One more time, I deleted the sensor. Not restarted my Raspberry in the meanwhile, I'm really curious how it looks like after the next reboot.

In general: this is not a big issue as it doesn't have any impact on functionality or stability to the rest of the network.

pergolafabio, 2019-06-24:
in your case its a real hardware device (sensor)
in my screenshots , its  a unknown device, its not a hardware device :)

ebaauw, 2019-06-24:
> Power cycle doesn't help either, even deleting the Osram, Brings that specific node back
> I tried your proposition, without success unfortunately.

Maybe the router keeps the associated end devices in non-volatile memory.  I’m not too familiar with this part of the standard, and maybe different manufacturers handle this differently.  Only a factory reset of the router clears the non-volatile memory, but I’m not sure if that’s worth the trouble.

Smanar, 2019-06-24:
On my side, this kind of "ghost device" become deletable after somes days, just wait somes days before retrying to delete them.

pergolafabio, 2019-06-24:
I have them already like 4 months ;)

Smanar, 2019-06-24:
Hur, lol, ok, forget that I have said ^^.

WhistleMaster, 2019-07-21:
> Maybe the router keeps the associated end devices in non-volatile memory. I’m not too familiar with this part of the standard, and maybe different manufacturers handle this differently. Only a [...]

I tried, same result, it keeps coming back.

closed 2019-11-25 by stale[bot]

walshtj, 2020-11-19:
For me this turned out to be a Smarthings Motion sensor which was still supposed to be paired with the (soon to be decommissioned) Smartthings hub.  There is a chance it might have not been paired but I can't be sure.

The deConz app allowed me to find the hardware address and product name which was "centralite 3325-s".
Using the ST graph backend I found the two hardware addresses matched. 

I temporarily re-paired to ST and problem seemed to go away.