dresden-elektronik/deconz-rest-plugin #1627: Hue lights fail to reconnect after a firmware update

l-mb, 2019-06-24:
I'm having the worst luck this week.

So after my issue https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1614, I thought I'd give using a Hue bridge to update the firmware on the lights a try. And that appears to have worked - I used the dimmer switch to factory reset them, paired them with the Hue bridge, waited forever for the update to be installed (now at 1.46.13_r26312), tested whether the lights worked with the Hue bridge still.

And then I deleted them from the Hue bridge, used the dimmer switch to reset them again, and tried pairing them with the RaspBee again.

No amount of coercing has succeeded at the latter so far - resetting them and powering them off/on, or directly trying to re-add them after the reset w/o the power reset. While the network was open or only enabling the search later. I even powered off the Hue bridge to avoid any interference, and tried all permutations again. No luck at all; the light doesn't show up.

I also tried deleting them from deCONZ. And tried again. No progress.

However, I do get a router joining (with a changing NWK everytime I try, but a static IEEE address). Something is obviously going wrong there - Manufacturer Code is 0x0, Max Buffer/Incoming/Outgoing Size is 0, MAC Capabilities are 0x0e. 

I've tried pairing the light with the Hue bridge again, and that worked.

I'm at a loss. It's almost as if there's a piece of handshake missing that the Hue bridge does but deCONZ/RaspBee don't?

ebaauw, 2019-06-24:
When you remove a light from the Hue bridge, it sends a _Leave_ request to the light.  If the light was reachable, it should be ready to join another network, without any additional reset.  You probably still need to power-cycle the light while the network is open.

Are you sure you reset the light using the Hue dimmer, and didn’t accidentally create an ad-hoc network?

What’s the mac address of the router?  Hue devices start with 0x001788.

You might try and reset the light with deCONZ using touchlink from the old web app.

l-mb, 2019-06-24:
Hi @ebaauw, thanks!

Yes, that's what the MAC address starts with - it's consistenly the same, whenever I turn on the Hue lamp again, and nothing else shows up. (I'm experimenting only with that one now.)

Yes, I'm sure I didn't accidentally create an ad-hoc network. All other behaviour matches a reset.

I believe the device *does* join the deCONZ/RaspBee network. Because it does not try to rejoin the Hue hub unless I reset the device again (via the dimmer switch, at which point it can immediately be discovered by Hue without the need to powercycle, which deCONZ seems to need?), which to me suggests it believes itself to be linked to a network. (Which matches the network graph deCONZ shows: it connects to multiple other routers.)

It just, for reasons that I don't understand, is not functional on the DeCONZ network anymore. With 0 byte sized buffers, of course no IO can be send to the device at all. But looking at https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1014#issuecomment-445564737, I see your firmware levels match mine and obviously things are working for you.

FWIW, all Hue lights behave this way, I can't connect any of them to my network anymore. I'm just focusing on one light to keep the experiment repeatable. I've even tried playing with the network security settings, no luck. (All other devices - Dresden Elektronik, Osram - continue working.)

l-mb, 2019-06-24:
@ebaauw Would you kindly share your network settings, just so I can rule that out? Thanks!

l-mb, 2019-06-24:
Okay. This was annoying.

For reasons I no longer remember, I had my network settings set to "Standard - Network Key from TC".

The new Hue firmware does decidedly not approve and does not pair correctly with such settings. It needs that set to "High".

closed 2019-06-24 by l-mb