dresden-elektronik/deconz-rest-plugin #1628: Aqara wall switch does not update power consumption

cobirnm, 2019-06-24:
Deconz is exposing the power consumption of Aqara Wall Switch. The issue is that it only updates when I connect via VNC and manualy update the atributes pressing read button.

Smanar, 2019-06-24:
Hello,
Wich one application are you using to see it ? Home automation, websocket client, deCONZ ... ?

cobirnm, 2019-06-25:
Hello,

I use home assistant to connect to deconz. as soon as I press read the power consumption sensor gets updated in home assistant

cobirnm, 2019-06-29:
> Hello,
> Wich one application are you using to see it ? Home automation application, websocket client, deCONZ ... ?

Hello,

I use home assistant to connect to deconz. as soon as I press read the power consumption sensor gets updated in home assistant

Smanar, 2019-06-29:
I haven't Home assitant, but can see you see websocket log in it ? If you can see then only when you press the button, try to modify manualy the attribute  reporting in deconz.

cobirnm, 2019-06-30:
> I haven't Home assitant (and I haven't too this device to check if it works) , but can you see websocket log in it ? If you can see the good info but only when you press the button, try to modify manualy the attribute reporting in deconz.
> 
> You can try to remove completely the device, and re-pair it, deconz will reconfigure it for the network with good setting.

I removed and re-pair it. it didn't work. I still can only get it updated when I use VNC and refresh "Analog input".
 
When I try to change the attribute reporting deconz crashes. This happnes when I press the read button in "atribute info"

cobirnm, 2019-07-06:
Is there some parameter that I can change to read the output when I press the switch? Can someone help me?

cobirnm, 2019-10-28:
Can someone help me with this. Even with last updates installed and removing and adding again the swithc it still does not update unless I go to vnc and press read node.

Smanar, 2019-10-28:
The only way I know, is configuring yourself the attribute reporting.

Do you still have this bug

>When I try to change the attribute reporting deconz crashes. This happnes when I press the read button in "atribute info"

cobirnm, 2019-10-28:
> The only way I know, is configuring yourself the attribute reporting.
> 
> Do you still have this bug ?
> 
> > When I try to change the attribute reporting deconz crashes. This happens when I press the read button in "atribute info"

Yes I still have it. I dont know where to go to. The value exist, I can't change the report interval since deconz crashes! The idea would be for deconz to refresh this parameter every time the swich is pressed for example... I jut don't kbnow how to do this!

closed 2020-03-05 by stale[bot]