dresden-elektronik/deconz-rest-plugin #1630: Broken Gateway Site on Phoscon Webapp @ 2.05.66

brandungskieker, 2019-06-24:
Having Trouble to open Gateway-page on Phoscon Webapp since updating to 2.05.66. The page is interrupted by JS and HTTP Errors. 
Occurs on Safari and Chrome (latest) for Mac. 
See Screenshots for more details

[image]
[image]

ademobile, 2019-06-25:
See issue #1572 on a work round until they fix it!

manup, 2019-06-25:
Thanks for the report, that's a new one, will be fixed in the next version.

TilBumble, 2019-08-27:
I updated manually to 2.05.67 (openHABianPi headless) and retrieving the gateway page from a Windows 10 PC (Microsoft Edge browser) still leads to a more or less blank page.
(No text, only left and top frames)

The console shows the following warnings and Infos (no errors):
__________________________
HTML1300: Navigation wurde ausgeführt.
settings-gateway2.html (1,1)

2CSS3121: Die Medienabfrage -ms-viewport ist veraltet.

3HTML1500: Die Markierung kann nicht selbstschließend sein. Verwenden Sie eine explizite schließende Markierung.
settings-gateway2.html (82,5)

HTML1512: Endmarkierung ohne Entsprechung.
settings-gateway2.html (130,12)

HTML1500: Die Markierung kann nicht selbstschließend sein. Verwenden Sie eine explizite schließende Markierung.
settings-gateway2.html (215,3)
______________________

The workaround still works.

Can i be of further assistance to clear up this issue?

brandungskieker, 2019-08-27:
Had the same. Cleaning up the browser cache helps.

closed 2019-08-27 by brandungskieker