dresden-elektronik/deconz-rest-plugin #1631: Osram Smart+ Plugs cannot be switched on/off in V2_05_66

Markkuuss, 2019-06-25:
Yesterday I updated to the latest version V2_05_66. Since then, my two Osram Smart+ Plugs no longer work correctly. A timeout occurs when switching via the on/off cluster. That goes so far that the devices lose their zigbee connection afterwards. I can only switch on the devices by switching on the group they are in.

I have read on the Internet that newer firmware versions of Osram Plugs can cause problems. My plugs have versions 1.04.12 and 1.04.90.

Can anybody reproduce a similar problem with the Osram Smart+ Plugs?

manup, 2019-06-25:
Which firmware version are you using?

When using the On/Off cluster directly it should always work, since the commands will be send directly to the plug. I'll check later if I can reproduce this with my OSRAM Smart+ plug.

Markkuuss, 2019-06-25:
I am using firmware version 26330500.

It is also not possible to read the attributes of the OTAU cluster.

Can you please tell me which firmware your Smart+ Plugs have?

manup, 2019-06-25:
It's the European version V1.04.12.
I can currently just control it remotely from the office but this seems to work.

manup, 2019-06-25:
Have you tried to power-cycle the plugs?
Do the other lights in the network work?

Markkuuss, 2019-06-25:
> Have you tried to power-cycle the plugs?

Yes, without success.

> Do the other lights in the network work?

Yes, several of dresden elektronik, ubisys, philips, ... They all work. Only the two Osram plugs cause problems.

I just reset one of the two nodes with F5 in deconz. Normally the node acts as a ZLL. Now it is displayed as follows:
[image]

Markkuuss, 2019-06-25:
> It's the European version V1.04.12.

So the same version as me.

What about deconz? Do you also use the latest version and the latest firmware of the Raspbee?

Philje123, 2019-06-25:
I also had a problem with my Osram Plug. I couldn't control it anymore after updated deconz to 2.05.66.

My plug has firmware version V1.05.11.

I deleted it and re-added and it's been fine since.

Markkuuss, 2019-06-25:
I'm now back at 2.05.65. That didn't help at first. Then I reloaded the firmware (according to the [description](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually#update-in-raspbian)), now the Osram plugs are reachable as usual.

@Philje123: What do you mean by "deleted"?
This one: 
[image]
Or did you set the plug to factory settings?

pergolafabio, 2019-06-26:
i am also on  1.04.12 (eu)
i see newer versions here like 1.04.90 1.05.11

how to update those osrams? i have issues with it if my xiaomi sensors are connected to it

Philje123, 2019-06-26:
I deleted the plug via the phoscon web page.

I have an Osram lightify Hub so updated my osram products via that before pairing with deconz.

pergolafabio, 2019-06-26:
ah ok , you saw some release notes in there ? , so 1.05.11 is latest one?

Markkuuss, 2019-06-26:
1.04.90 is on my Osram Smart+ Outdoor Plug. But I think that is the same hardware as the indoor only in a different case.

An update is probably only possible via the Osram Hub, since the firmware files are only available for download as encrypted .bin files. Or is anyone familiar with another solution?

pergolafabio, 2019-06-26:
i hope for another solution :)
dont want to buy a hub, only for upgrading :)

pergolafabio, 2019-06-26:
i dropped osram a support mail, but not hoping for a good reply anyway :)

Philje123, 2019-06-26:
Yeah I only got the hub as my local supermarket were selling a load of Osram stuff off cheap and only paid £12 for it. It's come in handy as I have quite a few Osram bulbs etc.

pergolafabio, 2019-06-26:
you also have xiaomi motion sensors? seems there are issues with it in combination with osram , so maybe firmwares can help

Markkuuss, 2019-06-26:
But isn't the update procedure very complicated? Could you explain it shortly, @Philje123 ? Do you have to set the Osram devices to factory defaults? You'll have to somehow remove them from the deconz network and then reinsert them? Do the Osram devices get a new ID after they have been reinserted into the deconz network after the update?

Markkuuss, 2019-06-26:
> i dropped osram a support mail, but not hoping for a good reply anyway :)

Please tell us the answer here.

pergolafabio, 2019-06-26:
Yeah, once I receive a reply ;)

pergolafabio, 2019-06-26:
with googling i was able to find the directory to those encrypted bin files :)
http://s3-eu-west-1.amazonaws.com/lightify-osram-firmware/
and here is the link to the release notes for that 1.04.90 and 1.05.10

http://s3-eu-west-1.amazonaws.com/lightify-osram-firmware/firmware/release/Cloud/Release_notes/Plug%2001.txt

pergolafabio, 2019-06-26:
also, got response from lightify, already asked if possible to send me the unecrypted files :)

**_thank you for your request. Unfortunately the LIGHTIFY devices can be updated only via the LIGHTIFY Gateway with the LIGHTIFY App._**

pergolafabio, 2019-06-26:
and another reply from another osram team :-)

**_**The firmware of SMART + products can also be updated via the third-party systems from Qivicon and Samsung Smart Things. Please ask for the possibility and the necessary steps at the respective manufacture**_**r.

Markkuuss, 2019-06-26:
These are exactly the answers that can also be found in the Osram FAQs. Maybe you should ask Qivicon and Samsung Smart Things? ;-)

pergolafabio, 2019-06-26:
:-)

i dont have that hardware , so its a nogo anyway

timbru31, 2019-06-27:
I've sent a support request to SmartThings. 🤞

manup, 2019-06-27:
By looking at the firmware files I figure these just need to be extended with the proper Zigbee OTA header. Normally it doesn't matter if OTA files are encrypted since the data is only passed through and handled by the receiving device.

pergolafabio, 2019-06-27:
ok, i can try to update one of my osrams, but never done this before
ist there a guide? can it be done with then GUI application or command line like upgrading conbee stick ?
i am using marthoc addon

timbru31, 2019-06-27:
> By looking at the firmware files I figure these just need to be extended with the proper Zigbee OTA header. Normally it doesn't matter if OTA files are encrypted since the data is only passed through and handled by the receiving device.

I've tried to load the file in the OTAU Plugin, but it did not load correctly.
Would you shed some light on the `proper Zigbee OTA header`? (I'm too lazy to dig up the Zigbee spec again)

Markkuuss, 2019-07-02:
So I've been testing it again for the last few days. With the last stable version (V2.05.65) and the corresponding firmware (0x26330500) it doesn't work. I have disconnections and the Osram plugs are often shown as offline. If they are online, I often have timeouts. I restarted all devices, deleted the plugs from the network and added them again. Then it runs for a while until there are problems again.

I have now switched to firmware 0x262f0500. It has been running for two days without any problems, which was never the case with the 0x263... firmware versions.

pergolafabio, 2019-07-04:
some good news, received a reply from osram, they are going to make a platform to download the firmwares

 Many thanks for contacting us.

Our current company policy doesn´t permitt publishing of the firmware.
We are working on a platform which will offer the firmware and 
download options.This one will be available in 2 months.

 Best regards
Ledvance Smarthome Support Team
mailto:smarthome-support@ledvance.com
www.ledvance.com

rehnmarkse, 2019-07-05:
> So I've been testing it again for the last few days. With the last stable version (V2.05.65) and the corresponding firmware (0x26330500) it doesn't work. I have disconnections and the Osram plugs [...]

Allright, så running deconz 2.05.66, with firmware 0x262f0500, makes everything work again?

jo-me, 2019-07-10:
Same issue here with the Osram plugs.
Running 2.05.66 with firmware 0x26330500.
They work after replugging them, but will eventually be unavailable after a while.

pergolafabio, 2019-07-10:
what is a while? mine is stable for 2 weeks with osram? i must say, i dont use  the power on/off on osram, its just sitting in the wall act as a router

jo-me, 2019-07-10:
@pergolafabio maybe after a day or so. It is quite frequently since I updated to the versions above. Before I did not have trouble with them. I have a light connected to one of them that I use very evening and it happens now every other day that I have to unplug the plug so that it responds again. I typically interact with it from Home Assistant where it is listed as unavailable.

Strangely, in phoscon, I can toggle the light back and forth but the plug does not do anything.

pergolafabio, 2019-07-10:
well, in my case i have no issues anymore, fingers crossed tough, still 2 weeks passed
before on older firmware and 64 and .65 , my xiaomi  sensors loose connection to osram, i always needed to repair after 2-3 days

also, must say, i did some wifi changes, i had 3 accesspoints, each on one floor, i had them on signal 1 , 6 , 11  => so changed them to 1 , 6 , 1 , so still no overlapping because of 3 floors, and changed conbee to channel 25 , thats actually in 11 range for wifi2.4 
so no interence normally, this maybe helped too

rehnmarkse, 2019-07-16:
deconz 2.05.66, with firmware 0x262f0500 worked for around 2 weeks, then it got unresponsive again.
Only solution was to remove the power to the device, to get it to work again.
Ended up returning the device and replaced it with an Nexa Outdoor switch(433mhz) oldschool device instead.

timbru31, 2019-10-16:
> some good news, received a reply from osram, they are going to make a platform to download the firmwares
> 
> Many thanks for contacting us.
> 
> Our current company policy doesn´t permitt publishing of the firmware.
> We are working on a platform which will offer the firmware and
> download options.This one will be available in 2 months.
> 
> Best regards
> Ledvance Smarthome Support Team
> mailto:[smarthome-support@ledvance.com](mailto:smarthome-support@ledvance.com)
> [www.ledvance.com](http://www.ledvance.com)

Did you ever heard back from them that the platform is ready? (If not, could you possibly ping them again with the already started support request/ticket?)

pergolafabio, 2019-10-17:
No, sended some more mails, no replys anymore... I ditched my osrams, and bought Ikea, much better...

Adminiuga, 2019-10-17:
@timbru31 better write them directly. More people request the firwmare, more chances it would actually happen

pergolafabio, 2019-10-17:
here, i just share my last email
received from: SERVICES@SERVICES.LEDVANCE.COM

Odoslané: | 27.06.19 14:40:05
-- | --
Vec: | RE: RE: (8501723528) Firmware update using Deconz

Many thanks for contacting us.

Our current company policy doesn´t permitt publishing of the firmware.
We are working on a platform which will offer the firmware and 
download options.This one will be available in 2 months.

 Best regards
 Daniela Bakytova
Ledvance Smarthome Support Team
mailto:smarthome-support@ledvance.com
www.ledvance.com

Adminiuga, 2019-10-18:
Wrote them as well to smarthome-support@ledvance.com got a similar response

>many thanks for contacting us.

>Unfortunately for now are Firmware files not free for download, but we are already working on an solution. 
To install new firmware, your SMART+ Zigbee lamps need to be connected to a partner smart home system that is authorized to install our firmware. For firmware updates currently authorized systems are OSRAM LIGHTIFY, Samsung SmartThings, Schwaiger Home4You and the Qivicon/Telekom Magenta SmartHome system. 

I suggest writing them as well, as more people keep asking about this, then more reasons for them to do it.

pergolafabio, 2019-10-18:
It will come, maybe a Christmas present

CherAlban, 2019-10-30:
Is this what you are waiting for?
https://update.ledvance.com/firmware-overview

Adminiuga, 2019-10-30:
Yippee!

pergolafabio, 2019-10-30:
cool , but i sold my osrams :-)
strange they dont answer on mails

pergolafabio, 2019-10-30:
this i needed for my osrams plus :> 2. Improve network performance.

they had indeed weak perforcmace, thats why i sold them :-)

Adminiuga, 2019-10-30:
I can't find firmware for `LEDVANCE PLUG`, nor for `OSRAM LIGHTIFY Flex RGBW`

kugelkopf123, 2019-10-31:
> I successfully updated my OSRAM SMART+ devices by downloading the corresponding images below (renaming them from .ota -> .zigbee):
> Classic A60 W clear - LIGHTIFY -> CLA60 W CLEAR
> SMART+ Plug 01 -> Plug01 OnOff MK

Works perfectly! Thanks!

andriej, 2019-10-31:
> > I successfully updated my OSRAM SMART+ devices by downloading the corresponding images below (renaming them from .ota -> .zigbee):
> > Classic A60 W clear - LIGHTIFY -> CLA60 W CLEAR
> > SMART+ Plug 01 -> Plug01 OnOff MK
> 
> Works perfectly! Thanks!

Could you please point direction how to update with phoscon/deconz?

kugelkopf123, 2019-10-31:
Thats not an official HowTo. Its on your own Risk. 

I'm using a Raspberry Pi with a Raspbee as Gateway. 
- Download the right Firmware for your Device from the Firmware Website.
- Rename this File from .OTA to .zigbee (dot)
- Start the Deconz Software in the GUI from your Raspi
- From the dropDown Menu "Plugins" choose STD OTAU Plugin
- In this upcomming window, click on "..." OTAU File. En choose your renamed File.
- From the list, select the Device you want to update. It has to be the correct device for your FW File. 
- Now click on Query, the Image Type of the selected Device should change. 
- This Image HEX Code should be the same as the Code above > "Image Type". If it isn't, than this is not the right FW.
- Be shure that your device stays online for the time, you update.
- Click on Update. 
- Update process takes 10- 70min. 

Its faster if your Device is near your gateway.

andriej, 2019-11-17:
Thank you for the instructions. At least on my setup these are still loosing some of the messages that i.e. Xiaomi door or motion sensors are willing to send to my conbee :-(

alp-by, 2019-12-09:
> Thats not an official HowTo. Its on your own Risk.
>
> I'm using a Raspberry Pi with a Raspbee as Gateway.
>
> * Download the right Firmware for your Device from the Firmware Website.
> * Rename this File [...]

well... I am trying the following: when I press "Query" button...no changes.
Do you easily press "Update" any way?
[image]

kugelkopf123, 2019-12-09:
>> Thats not an official HowTo. Its on your own Risk. ...

@alp-by 

https://phoscon.de/en/support#ota-update-osram-devices

Check this link!
Don’t know why it’s like this on your gui. 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

alp-by, 2019-12-09:
> > > Thats not an official HowTo. Its on your own Risk. ...
> 
> @alp-by
> 
> https://phoscon.de/en/support#ota-update-osram-devices
> 
> Check this link!
> Don’t know why it’s like this on your gui.
> 
> Sent with [GitHawk](http://githawk.com)

thx. Yes, strange. I use both: 
1. linux docker with VNC and connect to VNC remotelly via VNC viewer. There button "select file" is not active it all. When I press it no reacting
2. windows -> screenshot was attached in the prev. message

kugelkopf123, 2019-12-09:
> thx. Yes, strange. I use both:
>
> 1. linux docker with VNC and connect to VNC remotelly via VNC viewer. There button "select file" is not active it all. When I press it no reacting
> 2. windows -> [...]

Perhaps the Plug is allready on the latest firmware? 
Its strange that you dont have any names on the devices.
Check it via the cluster info. Click on the (i think) right grey button from the two on your Osram Node. in the GUI. 
Change the view to Cluster Info (Look at the picture of from the manual on Step 2 and 3) choose cluster Basic or OATU and click on Read in the Cluster Info Panel. That should give you the acutal FW Verison.

alp-by, 2019-12-09:
> Perhaps the Plug is allready on the latest firmware?
> Its strange that you dont have any names on the devices.
> Check it via the cluster info. Click on the (i think) right grey button from the two on [...]

thx. I managed to update it finally. But my Phoscon still shows the old FW:  V1.04.12. Is it normal?
And btw: with a new software osram smart plug is still "light" or is it possible to have it as "switch"?

kugelkopf123, 2019-12-09:
> > > > > > Thats not an official HowTo. Its on your own Risk. ...
> > > > 
> > > > 
> > > > @alp-by
> > > > https://phoscon.de/en/support#ota-update-osram-devices
> > > > Check this link!
> > > > Don’t know why it’s like this on your gui.
> > > > Sent with [GitHawk](http://githawk.com)
> > > 
> > > 
> > > thx. Yes, strange. I use both:
> > > 
> > > 1. linux docker with VNC and connect to VNC remotelly via VNC viewer. There button "select file" is not active it all. When I press it no reacting
> > > 2. windows -> screenshot was attached in the prev. message
> > 
> > 
> > Perhaps the Plug is allready on the latest firmware?
> > Its strange that you dont have any names on the devices.
> > Check it via the cluster info. Click on the (i think) right grey button from the two on your Osram Node. in the GUI.
> > Change the view to Cluster Info (Look at the picture of from the manual on Step 2 and 3) choose cluster Basic or OATU and click on Read in the Cluster Info Panel. That should give you the acutal FW Verison.
> 
> thx. I managed to update it finally (I press the same buttons and it work from 2 or 3 times ... probably because of the VNC viewer software..don't know). But my Phoscon still shows as light with the "old" FW: V1.04.12. Is it normal?
> 
> [image]

The updated was 100%? And had you heard the Plug Switch off/on on the end of the update? Give it a bit time to reconnect to your network and than click once again on read. Perhaps you have to restart the system. Or unplug and replug the osram device. But Please, dont do this while updateing! ;)

alp-by, 2019-12-09:
> The updated was 100%? And had you heard the Plug Switch off/on on the end of the update? Give it a bit time to reconnect to your network and than click once again on read. Perhaps you have to [...]

yes update was 100%! As for "Plug Switch off/on on the end" - i don't remember, I didn't pay attention on it.... but you are right:I had to wait a little bit more....now it is V1.05.09 :)

kugelkopf123, 2019-12-10:
> > > > > > > > Thats not an official HowTo. Its on your own Risk. ...
> > > > > > 
> > > > > > 
> > > > > > @alp-by
> > > > > > https://phoscon.de/en/support#ota-update-osram-devices
> > > > > > Check this link!
> > > > > > Don’t know why it’s like this on your gui.
> > > > > > Sent with [GitHawk](http://githawk.com)
> > > > > 
> > > > > 
> > > > > thx. Yes, strange. I use both:
> > > > > 
> > > > > 1. linux docker with VNC and connect to VNC remotelly via VNC viewer. There button "select file" is not active it all. When I press it no reacting
> > > > > 2. windows -> screenshot was attached in the prev. message
> > > > 
> > > > 
> > > > Perhaps the Plug is allready on the latest firmware?
> > > > Its strange that you dont have any names on the devices.
> > > > Check it via the cluster info. Click on the (i think) right grey button from the two on your Osram Node. in the GUI.
> > > > Change the view to Cluster Info (Look at the picture of from the manual on Step 2 and 3) choose cluster Basic or OATU and click on Read in the Cluster Info Panel. That should give you the acutal FW Verison.
> > > 
> > > 
> > > thx. I managed to update it finally (I press the same buttons and it work from 2 or 3 times ... probably because of the VNC viewer software..don't know). But my Phoscon still shows as light with the "old" FW: V1.04.12. Is it normal?
> > > [image]
> > 
> > 
> > The updated was 100%? And had you heard the Plug Switch off/on on the end of the update? Give it a bit time to reconnect to your network and than click once again on read. Perhaps you have to restart the system. Or unplug and replug the osram device. But Please, dont do this while updateing! ;)
> 
> yes update was 100%! As for "Plug Switch off/on on the end" - i don't remember, I didn't pay attention on it.... but you are right:I had to wait a little bit more....now it is V1.05.09 :)
> 
> But it should work as light with new FW, shouldn't it? Because I still see no chance how to change the status in Phoscon (it works ON\OFF only from Domoticz)

Great!

Yes, you will find them in Phoscon on the group "Devices/Lights":
[image]
As you click on one of them, the green LED on the Button of the Plug should blink to recognize them.
To switch them on/off via Phoscon, you have to create a Group in the Mainmenu, add one Plug and from there you can switch it on and off. As Group or as individual device.

[image]

alp-by, 2019-12-10:
> Great!
>
> Yes, you will find them in Phoscon on the group "Devices/Lights":
> [image]
> As you click on one of them, the green LED on the Button of the Plug should blink to recognize them.
> To switch them [...]

thx a lot :)

[28 more replies]

closed 2020-05-02 by stale[bot]