dresden-elektronik/deconz-rest-plugin #1632: Phoscon didn't find the gateway on remote connection (non-standard port)

renegade2k, 2019-06-25:
Hi.
I'm working with ConBee on a RaspberryPi and besides deconz/phoscon there are also several services running on the RPi, so i had to split the ports for all of them.
My deconz got the http port 9595 (ws port 9696) now and is running well so far.
I can bring up the web-app and connect to the gateway.

So i forwarded the external port 5432 to 9595 on my router and i am able to access the first "/pwa" site (by <ext.ip>:5432), which then starts to search for gateways and did not find anything.
As mentioned it works well on my LAN and also by VPN but not over remote connection, even the port is forwarded.

Can you give me a hint, how to make it accessable?

manup, 2019-06-25:
>So i forwarded the external port 5432 to 9595 on my router and i am able to access the first "/pwa" site (by <ext.ip>:5432), which then starts to search for gateways and did not find anything.

Please don't do expose the deCONZ HTTP server to the public Internet, the security capabilities aren't suitable for that purpose.

>Can you give me a hint, how to make it accessable?

I would recommend either VPN or Tor hidden services.

renegade2k, 2019-06-25:
It wasn't meant to be public all the time. Just for initial configuration, which takes a bit longer, for that i have like 40 devices, which are connected and must be sorted, given rules and so on ^^

So it's not a misconfiguration by me, but it's still not meant to be accessed from the internet.
Thx for the quick reply.

closed 2019-06-25 by renegade2k