dresden-elektronik/deconz-rest-plugin #1633: please unleash the repo for buster

00lex, 2019-06-25:
upgrade from stretch to buster on my rpi3 runs well and phoscon runs without issues. 
now I want to make fresh install on a rpi4 and can't install deconz because of missing buster repo. 

thx

manup, 2019-06-26:
The Buster package repo will be available with the next version 2.05.67.
Does Raspberry Pi 4 with Buster still use 32-bit armhf packages or did they switch to arm64?

SwoopX, 2019-06-26:
I read they'll stick to 32-bit for the time being. However, they also said like 2 months ago, there will be no Pi update until beginning of next year :)

00lex, 2019-06-30:
ahh sry but when could we expect the buster repo? 
deconz seem to me buster ready. as I write deconz works after a upgrade. now i need the repo after fresh install. 

greetings!

00lex, 2019-07-01:
piep.. :)

00lex, 2019-07-01:
okay i was a litte impatiently and install 2.05.65 manual via wget on buster@rpi4. just works!
2.05.65 because 2.05.66 produce some errors in the hue pimatic plugin :(

https://github.com/markbergsma/pimatic-hue-zll/issues/21

00lex, 2019-07-17:
it seems something happens :)

```
Get:1 http://phoscon.de/apt/deconz buster InRelease [2,845 B]
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease                                                             
Hit:3 http://archive.raspberrypi.org/debian buster InRelease                                   
Ign:4 https://downloads.plex.tv/repo/deb public InRelease                                      
Hit:5 https://downloads.plex.tv/repo/deb public Release
Err:1 http://phoscon.de/apt/deconz buster InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY C9A78F8C43D8720A
Reading package lists... Done
W: GPG error: http://phoscon.de/apt/deconz buster InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY C9A78F8C43D8720A
E: The repository 'http://phoscon.de/apt/deconz buster InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

closed 2019-07-17 by 00lex